kizniche/Mycodo #422: Sensor support request: Xiaomi Mi Flora / Mi Plant

samsixtysix, 2018-03-09:
I don't know if it would be possible to support this as it is a Bluetooth based sensor. 

It is low cost and gives readings for light, temperature, moisture and soil conductivity.

There are a couple of python libraries available for it:

 https://github.com/kipe/miplant

https://github.com/open-homeautomation/miflora

and a bit more info on the comms it sends:

https://www.open-homeautomation.com/2016/08/23/reverse-engineering-the-mi-plant-sensor/

kizniche, 2018-03-09:
If you have this sensor, and are willing to test, I will work with you to get an input module added to Mycodo to enable its use.

samsixtysix, 2018-03-09:
I have one on the slow boat from China and would be happy to test once it arrives.  Assuming we can get a single one working do you think it will be possible to support several of these at once?

kizniche, 2018-03-09:
According to https://github.com/open-homeautomation/miflora the device is queried by supplying the sensor's BT MAC address, so you should theoretically be able to communicate with an unlimited number of sensors. Do you have more than one sensor on the way so we could test multiple at the same time?

samsixtysix, 2018-03-09:
Only 1 on the way but I can order a second one - delivery may take a while but could be here in 1 week if I'm lucky

kizniche, 2018-03-09:
We can work on getting one working, and if you want to get another to see if we can get multiple working, then do that. I'd suggest going with one for now, unless you're feeling particularly confident we can get multiple working.

samsixtysix, 2018-04-04:
I finally received the sensor so am happy to test whenever you have time to look at this.

frodus17, 2018-04-04:
These look really cool, and cheap on ebay as well!

kizniche, 2018-04-04:
Don't worry about the commit I just made adding support to Mycodo. The testing script is what I'd like you to test when you have time. This commit is merely getting the framework in place (I'm fairly confident we'll have this working quickly).

samsixtysix, 2018-04-04:
Having a bit of trouble getting the test script to work but it's probably
my fault as I'm new to the pi/Linux (though not programming in general).

It's giving an Importer for : No module named 'bluepy'.

I tried installing bluepy using:
Sudo pip install bluepy
And it is installed and seems to work but the test script still gives the
same error.

Any ideas?  I can see the sensor Mac address using blescan command

Sam

samsixtysix, 2018-04-04:
Sorry my phone corrected 'ImportError' to something else.

Sam

kizniche, 2018-04-04:
It looks like you're not using the exact commands I provided. Use copy and paste to make sure you execute one directly after the other, exactly as I wrote them.

samsixtysix, 2018-04-04:
I didn't copy/paste as my email wasn't on the pi(it is now).  Everything
seemed to install ok except for the following two lines:

~/mycodo_test/Mycodo/env/bin/pip install miflora
~/mycodo_test/Mycodo/env/bin/pip install btlewrap

These gave lots of red text so I put 'sudo' in front of them and they then
installed ok.

See below for what happens when I try copy/paste with the final command.

Sam

pi@raspberrypi:~ $ sudo ~/mycodo_test/Mycodo/env/bin/python3
~/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py -v scan
Scanning for 10 seconds...
Traceback (most recent call last):
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 73, in <module>
    main()
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 69, in main
    args.func(args)
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 34, in scan
    devices = miflora_scanner.scan(BluepyBackend, 10)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/miflora/miflora_scanner.py",
line 16, in scan
    for (mac, name) in backend.scan_for_devices(timeout):
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/btlewrap/bluepy.py",
line 107, in scan_for_devices
    from bluepy.btle import Scanner
ImportError: No module named 'bluepy'

kizniche, 2018-04-04:
You must execute the commands as they're written for it to work. The errors you are seeing is a result of not installing bluepy in the virtualenv I specified. If you need to, delete the ```mycodo_test``` directory and start over. It is crucial for you to use the commands I provided in order to test this.

kizniche, 2018-04-04:
If you receive any errors for a command I provided, please paste the entire error message here.

samsixtysix, 2018-04-04:
Trying from scratch again - only point noted is that it says my version of
pip is old

Get the following:

pi@raspberrypi:~/mycodo_test $ ~/mycodo_test/Mycodo/env/bin/pip install
miflora
Collecting miflora
  Using cached
https://www.piwheels.hostedpi.com/simple/miflora/miflora-0.3-py3-none-any.whl
Installing collected packages: miflora
Exception:
Traceback (most recent call last):
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/basecommand.py",
line 215, in main
    status = self.run(options, args)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/commands/install.py",
line 342, in run
    prefix=options.prefix_path,
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_set.py",
line 784, in install
    **kwargs
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_install.py",
line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_install.py",
line 1064, in move_wheel_files
    isolated=self.isolated,
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/wheel.py",
line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/wheel.py",
line 316, in clobber
    ensure_dir(destdir)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/utils/__init__.py",
line 83, in ensure_dir
    os.makedirs(path)
  File "/home/pi/mycodo_test/Mycodo/env/lib/python3.5/os.py", line 241, in
makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied:
'/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/test'
pi@raspberrypi:~/mycodo_test $

kizniche, 2018-04-04:
Oops. Missed a command. Execute the following command, then resume the test at the above command you received an error with.

```
sudo chown pi.pi ~/mycodo_test
```

samsixtysix, 2018-04-04:
pi@raspberrypi:~ $ sudo chown pi.pi ~/mycodo_test
pi@raspberrypi:~ $ ~/mycodo_test/Mycodo/env/bin/pip install miflora
Collecting miflora
  Using cached
https://www.piwheels.hostedpi.com/simple/miflora/miflora-0.3-py3-none-any.whl
Installing collected packages: miflora
Exception:
Traceback (most recent call last):
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/basecommand.py",
line 215, in main
    status = self.run(options, args)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/commands/install.py",
line 342, in run
    prefix=options.prefix_path,
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_set.py",
line 784, in install
    **kwargs
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_install.py",
line 851, in install
    self.move_wheel_files(self.source_dir, root=root, prefix=prefix)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/req/req_install.py",
line 1064, in move_wheel_files
    isolated=self.isolated,
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/wheel.py",
line 345, in move_wheel_files
    clobber(source, lib_dir, True)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/wheel.py",
line 316, in clobber
    ensure_dir(destdir)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/pip/utils/__init__.py",
line 83, in ensure_dir
    os.makedirs(path)
  File "/home/pi/mycodo_test/Mycodo/env/lib/python3.5/os.py", line 241, in
makedirs
    mkdir(name, mode)
PermissionError: [Errno 13] Permission denied:
'/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/test'

:( Sam

kizniche, 2018-04-04:
It's missing the '-R' in the command. I amended the comment moments after I sent it.

samsixtysix, 2018-04-04:
Tried a fresh install and still getting the same error on final scan
command:

pi@raspberrypi:~/mycodo_test $ ~/mycodo_test/Mycodo/env/bin/pip install
miflora
Collecting miflora
  Using cached
https://www.piwheels.hostedpi.com/simple/miflora/miflora-0.3-py3-none-any.whl
Installing collected packages: miflora
Successfully installed miflora-0.3
pi@raspberrypi:~/mycodo_test $ ~/mycodo_test/Mycodo/env/bin/pip install
btlewrapCollecting btlewrap
  Using cached
https://www.piwheels.hostedpi.com/simple/btlewrap/btlewrap-0.0.2-py3-none-any.whl
Installing collected packages: btlewrap
Successfully installed btlewrap-0.0.2
pi@raspberrypi:~/mycodo_test $ sudo ~/mycodo_test/Mycodo/env/bin/python3
~/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py -v scan

Scanning for 10 seconds...
Traceback (most recent call last):
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 73, in <module>
    main()
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 69, in main
    args.func(args)
  File
"/home/pi/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py",
line 34, in scan
    devices = miflora_scanner.scan(BluepyBackend, 10)
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/miflora/miflora_scanner.py",
line 16, in scan
    for (mac, name) in backend.scan_for_devices(timeout):
  File
"/home/pi/mycodo_test/Mycodo/env/lib/python3.5/site-packages/btlewrap/bluepy.py",
line 107, in scan_for_devices
    from bluepy.btle import Scanner
ImportError: No module named 'bluepy'

kizniche, 2018-04-04:
You didn't install bluepy

samsixtysix, 2018-04-04:
I see the problem now; I've been working from the script I got on my email that doesn't include the bluepy line.  I just checked the comments and it has an  extra command, maybe you edited it earlier and it didn't email the change.

samsixtysix, 2018-04-04:
All working fine now, reading correctly:

pi@raspberrypi:~/mycodo_test $ sudo ~/mycodo_test/Mycodo/env/bin/python3 ~/mycodo_test/Mycodo/mycodo/tests/manual_tests/miflora_test.py -v poll C4:7C:8D:66:1F:78Getting data from Mi Flora
DEBUG:miflora.miflora_poller:Received result for handle 56: 64 27 33 2E 31 2E 38
FW: 3.1.8
Name: Flower care
DEBUG:miflora.miflora_poller:Filling cache with new sensor data.
DEBUG:miflora.miflora_poller:Received result for handle 53: C8 00 00 AC 00 00 00 1F 3B 01 02 3C 00 FB 34 9B
Temperature: 20.0
DEBUG:miflora.miflora_poller:Using cache (0:00:00.003464 < 0:10:00)
Moisture: 31
DEBUG:miflora.miflora_poller:Using cache (0:00:00.004700 < 0:10:00)
Light: 172
DEBUG:miflora.miflora_poller:Using cache (0:00:00.005810 < 0:10:00)
Conductivity: 315
Battery: 100

kizniche, 2018-04-04:
Nice! The module I added should work as is. I'll go ahead and make a Mycodo release so you can upgrade and test it out.

samsixtysix, 2018-04-04:
Thanks for your work on this!

For anyone else interested in this sensor I'll report back after a few days
use.  Initial impressions are that it seems well made and pretty accurate
but the BT range is not great.

kizniche, 2018-04-04:
No Problem. Thanks for testing. I just made the release, so you can upgrade through the web interface.

fry42, 2018-04-13:
Hi!

I tried adding the miflora sensor today. Updated to 5.7.2, installation of dependencies went fine, used the test script to get the mac address, added sensor - but when I try to activate the input I get the following error:

Error: Could not activate Input controller with ID 10: invalid literal for int() with base 10: 'c4:7c:8d:66:48:03'

Looks like some wrong variable declaration?

samsixtysix, 2018-04-13:
Same problem for me (sorry didn't have a chance to test previously due to a dead SD card).  Also got the following after installing dependencies (although they seemed to install fine):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 218, in admin_dependencies
    dep = int(f.read(1))
ValueError: invalid literal for int() with base 10: ''

kizniche, 2018-04-13:
It's a little chaotic right now, as I'm working on 6.0.0, which will require a database deletion if coming from 5.x. So, I may not get to address this issue in the 5.x branch. I should have 6.0 released soon, though.

In the meantime, I can sit down later and tell you what you can edit to get this working.

samsixtysix, 2018-04-13:
No problem, I'm happy to wait for 6.0

kizniche, 2018-04-13:
@samsixtysix's error is different from @fry42 

@samsixtysix: did you cut off the end of your error? It looks like there's an open quote without a closing quote.

kizniche, 2018-04-13:
@fry42, to fix your issue, change this line:

https://github.com/kizniche/Mycodo/blob/7f6df4aa68c55cccf437409c651f45aa6b83ec16/mycodo/controller_input.py#L416

to this:

```python
            self.measure_input = MifloraSensor(self.location,
```

kizniche, 2018-04-13:
@samsixtysix, actually, that's not a double-quote, but two single quotes. That's an odd error. It has to do with the dependency system. I'll investigate it later.

RynFlutsch, 2018-04-20:
> Error: Could not activate Input controller with ID 10: invalid literal for int() with base 10: 'c4:7c:8d:66:48:03'

> 
Got the same

Theoi-Meteoroi, 2018-04-20:
You guys seemed to be having so much fun with this, I ordered 6 so I could play along. I’ll get them in early May.

samsixtysix, 2018-04-21:
I get the same (dependency) error as before using  6.0.4  (I deleted the old Mycodo directory before install but have not tried a full fresh Raspbian install)

Also I can add the miflora as a data source successfully but when I click to activate it I get the following message:
'Error: Cannot activate Input without the GPIO/I2C Address/Port to communicate with it set'

kizniche, 2018-04-21:
Do you have a MAC address saved?

samsixtysix, 2018-04-21:
Sorry, just realised I hadn't put the MAC address in - now working ok, thanks :)

kizniche, 2018-04-21:
You're the first to test... So the module is working as expected?

kizniche, 2018-04-21:
All measurements are reporting values (check the live page)?

samsixtysix, 2018-04-21:
If I click on dependencies I get an Error 500: internal server error with the traceback I posted before the only change being a line number 218 now 261

kizniche, 2018-04-21:
Can you paste the full traceback, please?

samsixtysix, 2018-04-21:
I was getting a zero value for moisture and conductivity for a few minutes but then I moved the plant pot into brighter light and on the next reading I got correct values for all fields.

samsixtysix, 2018-04-21:
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py",
line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py",
line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py",
line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line
220, in admin_dependencies
    dep = int(f.read(1))
ValueError: invalid literal for int() with base 10: ''

samsixtysix, 2018-04-21:
I also can't get Chromium to add the generated SSL certs - when added the browser says 'The private key for this certificate is missing or invalid'   I didn't see this problem with my previous install.

kizniche, 2018-04-21:
Can you manually edit your routes_admin.py and test the fix I just committed?

kizniche, 2018-04-21:
Perhaps you need to delete the previously-saved certificate from your browser.

samsixtysix, 2018-04-21:
Getting rid of the old cert fixed the SSL issue and your commit fixed the dependency error I was getting.  Thanks!

studi0z, 2018-04-26:
hello , i received today a RPI3b+ and  a Xiaomi Mi Flora and i'm having little issues with it .
Fresh install of  Mycodo 6.0.6 .Using the integrated BT on RPI .
all i have to do is to setup the MAC Address on Inputs ?
or do i need to set it up with : bluetoothctl ?
thx for the help

kizniche, 2018-04-26:
Try executing the following command in a terminal to scan for Bluetooth devices. Make sure your MiFlora is within range and on.

```
sudo hcitool lescan
```

kizniche, 2018-04-26:
If you detect your MiFloria, then copy the MAC address into the Input setting box in Mycodo, Save, then activate the Input. The output of the hcitool should look like this:

```
pi@pi0w-miflora:~ $ sudo hcitool lescan
LE Scan ...
C4:7C:8D:61:57:A0 (unknown)
C4:7C:8D:61:57:A0 Flower care
C4:7C:8D:62:41:BE (unknown)
C4:7C:8D:62:41:BE Flower care
```

closed 2018-04-28 by kizniche

studi0z, 2018-04-28:
kizniche sorry for late reply 
finally it was the sensor that wasn't working properly  .
i ordered 2 , so i end up testing the second one and worked perfectly .
i will be ordering more soon to test with the same RPI , i will have multiple zones to control the irrigation on my greenhouse .
thank you 

PS: i have this sensor arriving in the next days , can you please add it 

https://www.amazon.ca/Temperature-Humidity-Xiaomi-Bluetooth-Screen-Remote/dp/B079L6N6PC/ref=sr_1_3?ie=UTF8&qid=1524928075&sr=8-3&keywords=xiaomi+humidity

i would like to make a donation in Dash (crypto) ! your application is really good ,doin a fantastic job. super happy ! Huge Thank You

kizniche, 2018-04-29:
When you get the device, start a new issue and let me know if you can find code that can successfully return measurement values.