kizniche/Mycodo #423: Sensor support request:  Thermocouple

samsixtysix, 2018-03-09:
I may have just missed it but I couldn't find any reference to Mycodo supporting a thermocouple input.

If it could support one of the common/cheap k-type interface chips/boards like the MAX31855 then it would be useful for higher temperature applications such as heat treat kilns, BBQs, smokers, coffee roasters, reflow ovens etc

Most of the breakout boards I have seen use SPI and there is a python library available from adafruit:

https://github.com/adafruit/Adafruit_Python_MAX31855

kizniche, 2018-03-09:
I actually just added support for the MAX31855 and k-type thermocouples yesterday (see the end of #345). I haven't made a release yet, but plan to do so by the end of today.

samsixtysix, 2018-03-09:
Great news, now I'm just hoping for programmable 'rate of rise' / 'ramp rate'.  

I think I saw that this was something you were working on?

kizniche, 2018-03-09:
[Methods](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.md#methods) allow you to set any rise/fall rate you desire a PID to operate at. Let me know if that's what you're looking for.

samsixtysix, 2018-03-09:
Can you set the rate of rise
e.g. temp increases 2 degrees every 10 seconds until it reaches a set temp;
and have the pid regulate the heater to maintain this?  I looked at Methods but it wasn't immediately clear if this was possible and I remember a previous Issue asking about it that I don't think was closed.

closed 2018-04-06 by kizniche