kizniche/Mycodo #424: Preserve data from 5.4

eap904, 2018-03-13:
I am wanting to upgrade and realized the jump from 5.4 to 5.5 is big. 
Trying to properly export / import the sensor data I have gathered, for use on the new system. 

Is there a way to do this?

kizniche, 2018-03-13:
The sensor data is stored in influxdb. This doesn't change during a Mycodo update. There is a Mycodo database that is updated to the new database schema during an update, however. The Mycodo database is what identifies the data in the influxdb as being associated with a particular Mycodo controller (Input, Output, PID, etc.).

If your version of Mycodo doesn't have the import/export page (introduced sometime in v5) that allows influxdb data to be backed up/restored, then you'll just have to press upgrade and hope for the best.

There's a backup made of the Mycodo directory before every update is attempted, so you can always restore an older version if the update fails.

Other than that, just know this is experimental software and I don't guarantee anything to work.

eap904, 2018-03-14:
Where is this import export page? One of the main menus? (data imput math output)

kizniche, 2018-03-14:
[image]

kizniche, 2018-03-22:
The Mycodo directory doesn't contain any measurement data, only settings data.

kizniche, 2018-03-23:
After you upgrade, the humidity measurement will be available to export.

closed 2018-04-06 by kizniche