kizniche/Mycodo #425: error on Begin/Clear Calibration with Atlas PH ( I2C Mode )

studi0z, 2018-03-14:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 
#### Problem Description
Please list:

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-03-14:
Thanks for the bug report. I'll push a fix later today and make a release for you to upgrade and try the new code.

studi0z, 2018-03-14:
thx for fast reply !!! and thanks for sharing this great project !!

kizniche, 2018-03-14:
What version Mycodo?

studi0z, 2018-03-15:
Mycodo Version: 5.6.4 
Python Version: 3.5.3

studi0z, 2018-03-15:
got a little error on status bar but worked !!!! huge thx 

Success: Calibration: high (10.0 pH): ('error', OSError(121, 'Remote I/O error'))
Success: Command: end: ('error', OSError(121, 'Remote I/O error'))
5. Calibration: pH: Atlas Scientific (I2C) 4 ( Atlas pH (I2C)): Calibration Succeeded

Calibration Complete

kizniche, 2018-03-15:
Did that error only appear at the last part?

studi0z, 2018-03-15:
no it also appeared while i was calibrating with the solutions

kizniche, 2018-03-15:
If that's the case, I don't think calibration was successful. Can you verify it's actually been calibrated?

kizniche, 2018-03-15:
I think I may have a fix. Can you change this:

https://github.com/kizniche/Mycodo/blob/226bce6c955ac79289923b92bd82a4ee73669028/mycodo/utils/calibration.py#L26-L28

to this:

```python
        elif self.interface == 'I2C':
            self.ph_sensor_i2c = AtlasScientificI2C(
                i2c_address=int(str(sensor_sel.location), 16),
                i2c_bus=sensor_sel.i2c_bus)
```

save, then restart the frontend with ```sudo service mycodoflask```

then reload the page and try calibration again. That should fix the IO errors and allow calibration to work. If not, can you paste the new error?

studi0z, 2018-03-15:
it is calibrated , i verified with the ph4/7/10 solutions . i also have a 2nd PH probe to verify .

kizniche, 2018-03-15:
The error you showed indicates the calibration command was never sent to the board. Did you see lights flashing on th board to indicate communication?

kizniche, 2018-03-15:
Did you clear the calibration before attempting the calibration?

studi0z, 2018-03-15:
no , but my Atlas probe is brand new and wasn't calibrated when i first plugged  
i worked for sure  , time to order the dripper from Atlas  now  , is it compatible to Mycodo ?

kizniche, 2018-03-15:
I'm skeptical your calibration was successful. If you can confirm the above code change causes no errors, then the fix will have allowed the calibration commands to be communicated with the pH circuit. If the errors go away, then it most likely means the initial calibration with the errors did not work.

Did you test the probe before calibration to determine if the calibration was successful? If not, there's not a good way to know if the calibration procedure worked or not. You will have to trust me that the error being returned indicates there were no commands successfully reaching the pH circuit.

studi0z, 2018-03-15:
[image]

kizniche, 2018-03-15:
Does that indicate it's been calibrated? I don't think it does.

kizniche, 2018-03-15:
I would actually think the pH would read nearly exactly 10 if it were properly calibrated.

kizniche, 2018-03-15:
I have to go to sleep. If you make the above code edit and restart the frontend and retry the calibration, let me know if the errors have been resolved.

kizniche, 2018-03-15:
Just thought of something... Did you see the actual pH returned on the page during the calibration? There should have been a pH displayed next to the countdown clock, slowing your measured pH. If there wasn't a live updated pH measurement displayed, then it also suggests the calibration wouldn't work as well.

kizniche, 2018-03-15:
After reviewing the code again after a good night's sleep, I've strengthened my position that none of the calibration procedure was working. I believe my fix is the solution to the issue. Your measurement in a pH 10 solution strengthens the argument because of your inaccurate measurement following calibration. Let me know if you're willing to test the code change, otherwise I'm going to implement it in the next software release, as the change brings the calibration code to the same operation as the measurement code (why measurements worked and calibration didn't).

closed 2018-11-30 by kizniche