kizniche/Mycodo #426: LCD does not activate when attempt to display relay state

drgrumpy, 2018-03-15:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.1
 
#### Problem Description
 - Attempt to display relay state on LCD. e.g. Output 4 (Lights) State (on/off)
 - Activate LCD
 - nothing happens, LCD continues to show Mycodo version (see Daemon errors below)
 
### Errors
2018-03-15 00:34:38,819 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 'LCDController' object has no attribute 'LCD_CMD'
2018-03-15 00:39:45,509 - mycodo.lcd_1 - ERROR - Error: 'relay_state'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 171, in __init__
    each_lcd_display.line_2_measurement)
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 440, in setup_lcd_line
    self.lcd_line[display_id][line]['unit'] = self.list_inputs[measurement]['unit']
KeyError: 'relay_state'
2018-03-15 00:39:45,540 - mycodo.lcd_1 - INFO - Activated in 24436.7 ms
2018-03-15 00:39:45,547 - mycodo.lcd_1 - ERROR - Exception: 'LCDController' object has no attribute 'LCD_CMD'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_lcd.py", line 268, in run
    self.lcd_byte(0x01, self.LCD_CMD, self.LCD_BACKLIGHT_OFF)
AttributeError: 'LCDController' object has no attribute 'LCD_CMD'
2018-03-15 00:39:45,599 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 'LCDController' object has no attribute 'LCD_CMD'
2018-03-15 00:51:16,468 - mycodo.lcd_1 - INFO - Activated in 9383.0 ms

kizniche, 2018-03-15:
Can you verify the issue exists with the latest version?

drgrumpy, 2018-03-15:
Ok, will do, hadn't noticed the new version, they are so frequent ! Just upgrading now... will report back...

drgrumpy, 2018-03-15:
Yes, it is fixed in the new version, sorry to have troubled you.

closed 2018-03-15 by drgrumpy

kizniche, 2018-03-15:
No problem. Let me know if you come across any other problems.