kizniche/Mycodo #427: Data from input command shows no data on LCD

drgrumpy, 2018-03-15:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.4
 
#### Problem Description
 - I have two commands that read values from a sensor and output a single numerical value
(small python scripts to read temp and rh from a 1-wire DS2438 Humidity sensor ).
- Both scripts give expected output from command line
- Both are used as data inputs and display expected values on graphs and live measurements
- When displayed on the 16x2 LCD the temp measurement displays correctly
- When displayed on the LCD the rh measurement consistently displays 'NO DATA'
- Also when selecting the rh measurement to display it shows the measurement as 'temperature' in the dropdown box instead of 'humidity' as specified in the Data input page, i.e. 

> Input 8 LinuxCommand (DS2438 RH) temperature

Live measurement:

> DS2438 RH (8)
> Input (LinuxCommand), 30.0 second interval
> Measurement | Timestamp
> 47.39 % humidity | 2018/3/15 2:21:34

LCD shows:

> DS2438 R NO DATA 

### Errors
No errors in the logs

kizniche, 2018-03-15:
I just released v5.6.5, which incorporates a potential fix for this issue. Let me know if the issue is resolved after you upgrade to the latest version.

drgrumpy, 2018-03-15:
Just upgraded (takes ages....presumably most of the files haven't changed in a minor upgrade, is there a way to speed this up ?)
Same result....
Also the web interface/font is now almost impossible to read following the upgrade to 5.6.5 (using FF 58 on Linux Mint 18.2) see separate issue report....

kizniche, 2018-03-15:
Clear your cache to use the new CSS file. Resave your LCD line settings for the new code to take effect.

drgrumpy, 2018-03-15:
Sorry I realised about clearing the cache just after I posted (I waste so much time on web design sometimes  because I forget that).
Yep I tried re-saving already. I also deleted the line set completely and re-created... all with same result... in the line drop down selection box it is still calling the RH 'temperature', is this a clue ?

kizniche, 2018-03-15:
Strange. It shouldn't be. That's what I supposedly fixed. I'll review the code and test again.

kizniche, 2018-03-15:
My test shows the two Linux Command Inputs are working properly with the LCD dropdown:

[image]

[image]

kizniche, 2018-03-15:
Can you post similar screenshots of your settings and LCD page (and preferrably the inspector of the dropdown, if possible).

kizniche, 2018-03-15:
Or just verify whether you have similar results, if creating screenshots will be difficult.

drgrumpy, 2018-03-15:
OK:
The data page for the relevant sensor:
[image]

And the LCD line data:
[image]

drgrumpy, 2018-03-15:
I just made an attempt to delete and recreate the sensor on the data page (thinking that this would recreate the sensor in the db).... I deleted it ok, but then when I came to add a new Linux Command I got the following error:

> Error (Full Traceback):
> 
> Traceback (most recent call last):
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
>     response = self.full_dispatch_request()
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
>     rv = self.handle_user_exception(e)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
>     reraise(exc_type, exc_value, tb)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
>     raise value
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
>     rv = self.dispatch_request()
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
>     return self.view_functions[rule.endpoint](**req.view_args)
>   File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
>     return func(*args, **kwargs)
>   File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1122, in page_data
>     unmet_dependencies = utils_input.input_add(form_add_input)
>   File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_input.py", line 60, in input_add
>     new_sensor.name = DEVICE_INFO[form_add.input_type.data]['name']
> KeyError: 'name'

kizniche, 2018-03-15:
This issue was supposedly fixed in the latest version. None of my systems report this error anymore. Perhaps you have some issue with your system. Are you in a position to reinstall Mycodo from scratch?

drgrumpy, 2018-03-15:
Yes, can do. I'll try a complete shutdown, powerdown, reboot first....

drgrumpy, 2018-03-15:
So a reboot has resolved it, and it seems to be working as expected, with the RH value from the Linux Command displayed on the LCD. 

However, the information shown in the dropdown selection for the line set is different for line 1 and line 2 (I can't keep the box dropped for a screen shot, so here is copy of the relevant bits of source code), see Input 9 has become 'humidity' in line 2:

Line 1 dropdown:
>                 <option value="30f89194-3188-4af8-9b6b-a572e8479aa9,lux" selected>Input 7 TSL2561 (TSL2561 Input): lux</option>
>                     
>                   
>                 <option value="522052f5-fd38-4b3a-a1ee-e3fae6e9c64f,sensor_time">Input 9 LinuxCommand (DS2438 Temp): Timestamp</option>
>                 
>                   
>                     <option value="522052f5-fd38-4b3a-a1ee-e3fae6e9c64f,temperature">Input 9 LinuxCommand (DS2438 Temp): temperature</option>
>                   
>                 <option value="be83a81b-be10-426f-babc-e99b8ab8b69b,sensor_time">Input 10 LinuxCommand (DS2438 RH): Timestamp</option>
>                 
>                   
>                     <option value="be83a81b-be10-426f-babc-e99b8ab8b69b,humidity">Input 10 LinuxCommand (DS2438 RH): humidity</option>

Line 2 dropdown:
 

>                     <option value="30f89194-3188-4af8-9b6b-a572e8479aa9,lux">Input 7 TSL2561 (TSL2561 Input): lux</option>
>                     
>                   
>                 <option value="522052f5-fd38-4b3a-a1ee-e3fae6e9c64f,sensor_time">Input 9 LinuxCommand (DS2438 Temp): Timestamp</option>
>                 
>                   
>                     
>                       <option value="522052f5-fd38-4b3a-a1ee-e3fae6e9c64f,humidity">Input 9 LinuxCommand (DS2438 Temp): humidity</option>
>                     
>                   
>                 <option value="be83a81b-be10-426f-babc-e99b8ab8b69b,sensor_time">Input 10 LinuxCommand (DS2438 RH): Timestamp</option>
>                 
>                   
>                     <option value="be83a81b-be10-426f-babc-e99b8ab8b69b,humidity">Input 10 LinuxCommand (DS2438 RH): humidity</option>
>                   
>                 <option value="fe739d6b-81ff-4540-8460-d2e54ee88d3b,pid_time">PID 1 (Heating PID): Timestamp</option>
>

kizniche, 2018-03-15:
Great! I think we finally got this resolved. I just pushed a fix for lines 2 through 4. If line 1 is working, the rest will after you get the fix. I'll try to make a release later today.

drgrumpy, 2018-03-15:
Brilliant....I've copied your changes to the lcd file and restarted the front end (a lot quicker than an upgrade on a pi zero) and it is now consistent. I have to say the speed by which you fix things is very impressive..... hopefully I won't find any more issues for a few days as I will be putting the hardware in a suitable box....

closed 2018-03-24 by kizniche