kizniche/Mycodo #428: Web interface unreadable after upgrade to 5.6.6

drgrumpy, 2018-03-15:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.5
 
#### Problem Description
Upgrade to 5.6.5 (from 5.6.4)
Text unreadable on web interface - all text is too small, and/or unreadable against background
Parts of boxes, text area are covered
Looks like the css is messed up on this new version

Using FF 58.01 on Linux Mint 18.2

See screenshot:
[image]

drgrumpy, 2018-03-15:
Having posted that I had a thought.... and deleted the browser cache... which improved things considerably....

kizniche, 2018-03-15:
I'm going to keep this open for a while because I suspect others will come here to resolve this. With firefox, clearing the CSS cache is as simple as clicking the refresh button. F5 to refresh doesn't do this, so physically clicking the refresh button is the easiest way to resolve this issue.

kizniche, 2018-03-15:
This was a result of moving from Bootstrap 3 to 4. Although this is an inconvenience to those unfamiliar with CSS caching (and those who may not realize what's happening), using the latest Bootstrap will enable better interface compatibility and features.

closed 2018-03-24 by kizniche