kizniche/Mycodo #429: Missing dependency on install: libffi-dev?

fry42, 2018-03-18:
Hi!

Just installed Mycodo on a RPi3B+ :)
OS: Raspbian Stretch Lite 2018-03-13
Updated all pkgs & FW as of 2018-03-18
Installed Mycodo 5.6.6 as documented

The error I found in the setup.log:
------------------------------------
Running setup.py bdist_wheel for bcrypt: finished with status 'error'
  Complete output from command /home/pi/Mycodo/env/bin/python3.5 -u -c "import setuptools, tokenize;__file__='/tmp/pip-build-f0i3j1sn/bcrypt/s
etup.py';f=getattr(tokenize, 'open', open)(__file__);code=f.read().replace('\r\n', '\n');f.close();exec(compile(code, __file__, 'exec'))" bdis
t_wheel -d /tmp/tmpsx85n7_npip-wheel- --python-tag cp35:
  Package libffi was not found in the pkg-config search path.
------------------------------------

I just tried "sudo apt-get install libffi-dev" and restarted the setup script.
This time it installed successfully...

I can post the full setup.log if it helps.
Also, if there's anything to test on the new RPi just tell me...

Greetings!
 Stephan

kizniche, 2018-03-18:
Thanks. I took that dependency out by accident. I just made a new release, 5.6.7, incorporating the fix.

closed 2018-03-18 by kizniche

kizniche, 2018-03-18:
As far as anything needing testing with the 3B+, just let me know if anything isn't working properly.

commit 60acc07a (2018-03-18): Release v5.6.7, Fix bcrypt dependency issue preventing install (closes #429)