kizniche/Mycodo #430: Can't open System Information

donpesce, 2018-03-21:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.8
 
#### Problem Description
Please list:

  I want to open the System Information but i get an error 500. I had the same problem with version 5.6.4

### Errors
   
Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 549, in page_info
    df_output = df_output.decode("utf-8")
UnicodeDecodeError: 'utf-8' codec can't decode byte 0xf6 in position 17: invalid start byte

### Steps to Reproduce the issue:

 
### Additional Notes

What can i do to solve this problem?

kizniche, 2018-03-24:
Can you paste the output of the ```df``` command from the command line?

donpesce, 2018-03-24:
pi@raspberrypi:~ $ df
Dateisystem    1K-Bl▒cke Benutzt Verf▒gbar Verw% Eingeh▒ngt auf
/dev/root       15373520 5032124   9653880   35% /
devtmpfs          470180       0    470180    0% /dev
tmpfs             474788       0    474788    0% /dev/shm
tmpfs             474788   47908    426880   11% /run
tmpfs               5120      12      5108    1% /run/lock
tmpfs             474788       0    474788    0% /sys/fs/cgroup
/dev/mmcblk0p1     41853   21329     20524   51% /boot
tmpfs              94956       0     94956    0% /run/user/1000

donpesce, 2018-03-24:
I hope this helps.

closed 2018-03-24 by kizniche

donpesce, 2018-03-25:
Thank you for the update. Now it works!

commit f4204a04 (2018-03-24): Fix encoding issue with df command output (closes #430)