kizniche/Mycodo #431: Remote access

eap904, 2018-03-22:
Curious to know what solutions people use for remote access to a mycodo server. 

I am generally using VNC, which works pretty reliably, although the android app on my phone refuses to connect, but it's much easier to read as it was intended to be accessed through a browser. 

The server is attached to a phone's wifi hotspot, which is difficult to traverse. Eventually I can get an android phone and tether it to a router which supports VPN connections, and when I want to access the computers I can use a VNC connection on my phone / computer, but I'm guessing there are simpler solutions out there.

kizniche, 2018-03-22:
I either forward port 443 to access the https:// address or host a VPN server on one of my locally-networked Pis and then connect to that to access my network when outside of it.

zsole2, 2018-03-24:
I have an OpenMediaVault file server (not a RasPi but BananaPI since it has SATA interface), and this provides OpenVPN. The router forwards the necessary ports and use no-ip.com dynamic DNS.

eap904, 2018-04-04:
Having a cellular hotspot makes this more difficult. As far as I can tell, I can't forward a port through the hotspot. A service with a middleman works, like chrome remote or vnc cloud. I suppose my router may have an option to be a wireless repeater, and then be a client to a VPN server from my home, but I haven't yet been able to traverse the cell network and access any services directly

Theoi-Meteoroi, 2018-04-10:
I use VNC however I have used another tool very much like it to access virtual machines and clusters in the recent past - called nomachine - [https://www.nomachine.com](url)

There may be features in that product that help with specific issues for access.

mrplow, 2018-04-26:
Personally I create a tunnel to port 443 vis SSH when I'm at work

closed 2018-06-28 by kizniche