kizniche/Mycodo #435: Not running after successful install

bergerac54, 2018-03-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: Latest
 
#### Problem Description
Please list:

Attempted to install Mycodo on a fresh install of Raspbian multiple times but daemon and client dont run despite log showing install successful.

 
### Errors
[setup.log](https://github.com/kizniche/Mycodo/files/1850630/setup.log)

### Steps to Reproduce the issue:
Multiple attempts at running install script and pip fix
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

fry42, 2018-03-27:
Hi!

Does your Pi have a reliable internet connection?
From the log it looks like it failed to get files from github.com and therefore ended the installation script.
It could be that with every retry you got more and more of the required files but never all of them.

Check your network connection and try again (and again).

If you still have problems post the latest setup.log so we can see if anything has changed.

bergerac54, 2018-03-28:
Connection is generally reliable. Did a few more installs and found it consistently failing at the following command

`git clone -q git://github.com/maxlklaxl/python-tsl2591.git `/home/pi/Mycodo/env/src/tsl2591``

Managed to clone tsl2591 by changing git:// to https:// and installed manually. Mycodo install was successful after that

kizniche, 2018-03-28:
```
#### Installing/updating tsl2591 with pip
Obtaining tsl2591 from git+git://github.com/maxlklaxl/python-tsl2591.git#egg=tsl2591
  Cloning git://github.com/maxlklaxl/python-tsl2591.git to ./env/src/tsl2591
fatal: unable to connect to github.com:
github.com[0: 192.30.255.113]: errno=Connection timed out
github.com[1: 192.30.255.112]: errno=Connection timed out

Command "git clone -q git://github.com/maxlklaxl/python-tsl2591.git /home/pi/Mycodo/env/src/tsl2591" failed with error code 128 in None
```

This indicates an inability to connect to github.com. Additionally, I verified installing the tsl2591 via the setup.sh script works as expected for me. You can even see in [the code](https://github.com/kizniche/Mycodo/blob/be603e2751412f887aea632a023dd141085c8733/mycodo/scripts/dependencies.sh#L18-L66) there are several other pip install commands with github repositories.

So, I'm not sure what happened on your system, but I don't believe it's a Mycodo issue.

closed 2018-03-31 by kizniche