kizniche/Mycodo #436: Race condition when enabling "Command (On/Off)" output for a duration

antoinechauveau, 2018-03-28:
## Mycodo Issue Report:

Mycodo Version: 5.6.9
Patched with fix for issue #432 (https://github.com/kizniche/Mycodo/commit/be603e2751412f887aea632a023dd141085c8733)
 
#### Problem Description

When enabling a "Command (On/Off)" output for a duration, a race condition in the code often causes it to get disabled right away. The issue may exist for other output types too, but I haven't tested this use case.

### Steps to reproduce the issue:

This is reproducible directly on the "Output" page:
- Add a "Command (On/Off)" output. I've been using the following: 
  - On: echo "$(date): on" >> <somepath>/output.txt
  - Off: echo "$(date): off" >> <somepath>/output.txt
- Type in a "Duration On" and click the test "Turn On" button.
- Observe the contents of file output.txt. Most of the time the "off" message will have been printed right after "on", without waiting for the requested duration.
 
### Root cause:

I think that this is caused by a race condition between the call to OutputController.output_on_off() and the loop that checks whether to turn off the output in OutputController.run().

In output_on_off(), the code goes as follows:
1. Set output_on_duration to "True"
2. Enable the output
3. Set output_on_until to "current time + duration"

If the run() loop wakes up between 1. and 3, it finds an old, expired value of "output_on_until" and turns the output off right away.

To fix this issue locally, I changed output_on_off() so that it always sets "output_on_until" before "output_on_duration". This fixes the repro case above, but I haven't performed more extensive testing.

kizniche, 2018-03-28:
Thanks for the report. I just pushed a fix that I tested to work. There were actually two parts of the output controller that were causing two separate issues. One was masking the second. Now it should all work properly (once a release is made, unless you want to manually edit the changes).

closed 2018-03-28 by kizniche

antoinechauveau, 2018-03-29:
Thanks!

I don't completely understand the choices that you made for this fix:
- Won't adding "self.output_on_duration[output_id] = False" in run() modify the code paths taken when the code then turns the output off in output_on_off()? It seems that we won't enter into "# Write output duration on to database" anymore, which was the path that used to reset "output_on_duration" among other bookkeeping.
- In the "# Output is on, but not for a duration" case, the code still sets "output_on_duration" before "output_on_until". There should be less risk of a race here, because we don't call the command to turn on in between, but maybe that is worth modifying too?

kizniche, 2018-03-29:
I'll look into it. I made the fix rather quickly and was going to review it again before making a release.

commit e49f65e8 (2018-03-28): Prevent race condition preventing output from remaining on for a duration (closes #436)