kizniche/Mycodo #437: problem with adding SHT75

kobrok, 2018-03-30:
Hello,
thank you for creating Mycodo, I find it very clean, precise and reliable.
For some days I'm trying to add SHT75 sensor, but it doesn't match with Mycodo.
I followed mk-fg installation instructions  - sensor is working with his script using the virtualenv on the  terminal.

After I try to activate STH75 in the Mycodo Input panel, I get the message: 
"Error: Could not activate Input controller with ID 11: name 'xrange' is not defined"

Here are some logs:
2018-03-30 23:44:35,720 - mycodo.daemon - ERROR - Could not activate Input controller with ID 11: name 'xrange' is not defined
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 486, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 378, in __init__
    self.sht_voltage)
  File "/var/mycodo-root/mycodo/inputs/sht1x_7x.py", line 35, in __init__
    self.sht_sensor = Sht(self.clock_pin, self.pin, voltage=self.voltage)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/sensor.py", line 297, in __init__
    super(Sht, self).__init__(pin_sck, pin_data, **sht_comms_kws)
  File "/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/sensor.py", line 131, in __init__
    self._init()
  File "/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/sensor.py", line 136, in _init
    self.gpio.set_pin_value(pin, k='direction', v='low')
  File "/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/gpio.py", line 66, in set_pin_value
    ft.partial(open, get_pin_path(n, k), 'wb', 0) ) as dst:
  File "/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/gpio.py", line 22, in gpio_access_wrap
    for n in xrange(checks, -1, -1):
NameError: name 'xrange' is not defined 

To my modest coding knowledge, problem has to do with python versions.
Where should I dig? What should I learn?

kizniche, 2018-03-30:
Try changing ```xrange``` to ```range``` in ```/var/mycodo-root/env/lib/python3.5/site-packages/sht_sensor/gpio.py```

Apparently the sht-sensor module isn't Python 3 compatible. I have this sensor, so I'll try to hook it up and discover if I can get it working.

kobrok, 2018-03-30:
Thank you for a fast response. After I edited file and restarted backend I get  "Error: Could not activate Input controller with ID 11: string argument without an encoding"

kizniche, 2018-03-30:
There's probably a lot of refactoring that will be necessary. You're welcome to attempt to tackle it if you're up to the task. I should be able to accomplish it when I get enough time. You're going to have to scrutinize the full exception message to discover what part of the code is causing the issue.

kobrok, 2018-03-30:
Well, I'm much more competent with the engenering side of the whole thing. (I just had to learn how to change permissions to overwrite that file, so that's total newbie) I'm keeping my fingers crossed. Thank you for your attention.

kizniche, 2018-03-31:
I just finished refactoring and successfully tested the code to work with both Python 2 and 3.

kizniche, 2018-03-31:
I made a [pull request](https://github.com/mk-fg/sht-sensor/pull/9), so let's see if the developer responds. Otherwise, I'll look into integrating the new code into Mycodo manually.

kizniche, 2018-03-31:
Well, this certainly worked out nicely. Not only did the developer of the sht-sensor project respond quickly, he transferred ownership of the project so I can maintain it. So, I just made version 18.3.3 release that adds Python 3 support, and Mycodo will update to this version with the next release. I'll make that release sometime this weekend, but in the meantime, you can update your virtualenv to this new version with the following command:

```bash
~/Mycodo/env/bin/pip3 install sht-sensor==18.3.3 --upgrade
```

Then restart the daemon with:

```bash
sudo service mycodo restart
```

Then your SHT75 should be working with Mycodo. Let me know if you have any issues.

kobrok, 2018-03-31:
now I get: Error: Could not activate Input controller with ID 11: (functools.partial(<built-in method write of _io.FileIO object at 0x72821c60>, b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 1.0)

[mycodo.log](https://github.com/kizniche/Mycodo/files/1865238/mycodo.log)

I will follow with a fresh copy of raspian and mycodo, to avoid a mess done in between and let you know than.

kizniche, 2018-03-31:
I had that error appearing until I was able to refactor for Python 3 support. My sensor has been storing measurements in Mycodo all night without a hiccup.

kobrok, 2018-03-31:
I’ll be gone for a couple days - went hometown for easter, forgot to bring my computer charger with me... Happy Easter to all of you!

kobrok, 2018-04-03:
I´ḿ back. I have made a fresh install of system and running 5.6.10 Mycodo, but I still get that Error: Could not activate Input controller with ID 4: (functools.partial(<built-in method write of _io.FileIO object at 0x6ede9ab0>, b'\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00\x00'), 1.0)
I would appreciate your help with actions I need to make it work.

kizniche, 2018-04-03:
try ```sudo apt-get install python3-setuptools```, then restarting the entire Pi.

kizniche, 2018-04-03:
I doubt the command I just gave will work, but worth a shot. Are you sure you have your sensor connected properly? To verify, can you retrieve measurements with other scripts?

kobrok, 2018-04-03:
That command doesn´t work :(  I double-checked the connectivity again. It worked both with mk-fg trom the terminal back than, and later I tested sensor also by connecting it with a propper library to arduino.

kizniche, 2018-04-03:
So it is still working with other scripts, currently?

kobrok, 2018-04-04:
Hardware (sensor + cables + pullup res) are working under adruino.
I cant make it work on raspbian anymore. I tried to install both with and without virtualenv.

without virtualenv:
**$ pip install 'git+https://github.com/kizniche/sht-sensor.git#egg=sht-sensor'**
Looking in indexes: https://pypi.python.org/simple, https://www.piwheels.org/simple
Requirement already satisfied: sht-sensor from git+https://github.com/kizniche/sht-sensor.git#egg=sht-sensor in ./.local/lib/python2.7/site-packages (18.3.6)
**$ sht -v -trd 20 21**
bash: sht: command not found

with virtualenv:
 **$ virtualenv venv**
**$ source venv/bin/activate**
(venv) pi@raspberrypi:~ $ pip install 'git+https://github.com/kizniche/sht-sensor.git#egg=sht-sensor'
Collecting sht-sensor from git+https://github.com/kizniche/sht-sensor.git#egg=sht-sensor
  Cloning https://github.com/kizniche/sht-sensor.git to /tmp/pip-build-8fj68_/sht-sensor
Installing collected packages: sht-sensor
  Running setup.py install for sht-sensor ... done
Successfully installed sht-sensor-18.3.6
**(venv) ~ $ sht -v -trd 20 21**
Traceback (most recent call last):
  File "/home/pi/venv/bin/sht", line 11, in <module>
    load_entry_point('sht-sensor==18.3.6', 'console_scripts', 'sht')()
  File "/home/pi/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 480, in load_entry_point
    return get_distribution(dist).load_entry_point(group, name)
  File "/home/pi/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2693, in load_entry_point
    return ep.load()
  File "/home/pi/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2324, in load
    return self.resolve()
  File "/home/pi/venv/local/lib/python2.7/site-packages/pkg_resources/__init__.py", line 2330, in resolve
    module = __import__(self.module_name, fromlist=['__name__'], level=0)
  File "/home/pi/venv/local/lib/python2.7/site-packages/sht_sensor/__init__.py", line 4, in <module>
    from sht_sensor.sensor import Sht, ShtVDDLevel
  File "/home/pi/venv/local/lib/python2.7/site-packages/sht_sensor/sensor.py", line 12, in <module>
    from builtins import range
ImportError: No module named builtins

kizniche, 2018-04-04:
builtins is from setuptools, if I'm not mistaken. I'll look more into the later.

kizniche, 2018-04-04:
When in your virtualenv, issue the following command, then try your sht command again:

```
pip install future
```

kobrok, 2018-04-04:
Heavens! It's working. Both in terminal and Mycodo. Though I need to activate that virtualenv after every system reboot. Is there a way to make it persistent?

kizniche, 2018-04-04:
That's interesting, because nothing about Mycodo changed (it uses it's own virtualenv), so what was it that caused it to start working?

kizniche, 2018-04-04:
It's only when the script is ran in python 2.7 it needs ```future``` to be installed, which is why the virtualenv you created needed ```future``` installed.

> Is there a way to make it persistent?

You have to activate your virtualenv or execute its python/pip executable, to use it. I'm not sure I understand what you're trying to accomplish.

kobrok, 2018-04-04:
Might be that this time I created virtualenv with a different name than before.

kobrok, 2018-04-04:
Thank you Kizniche, your help means a lot to me. Now I will learn more about those commands, to be more aware of what I'm doing.

kizniche, 2018-04-04:
There must be something else you did for it to begin working, because the virtualenv is different from Mycodo's. The virtual environment merely separates python modules, so anything installed in one environment doesn't impact scripts ran from another virtualenv. So, if Mycodo just became able to measure your SHT sensor, without changing anything in Mycodo, that leads me to believe there was something you did other than with the separate virtualenv that fixed it.

kobrok, 2018-04-04:
well, I try not to make more than one change at the time. I changed gpio pins some hours ago. But how would we explain that fact: that now when I restart raspbian, mycodo wouldn't see the sensor unless I start virtualenv in terminal?

kizniche, 2018-04-04:
> mycodo wouldn't see the sensor unless I start virtualenv in terminal?

I'm not sure what this means. You're saying upon bootup, Mycodo can't read the sensor. Then, once you acquire a measurement with the script, from the new virtualenv, Mycodo can then begin acquiring measurement without error?

kobrok, 2018-04-04:
to be more precise - mycodo wouldn't see it until I make first measurement within virtualenv

kobrok, 2018-04-04:
yes

kizniche, 2018-04-04:
It sounds like you may have something buggy about your sensor. My system doesn't have this issue with my SHT75 sensor and Mycodo.

kizniche, 2018-04-04:
What you can do is edit your crontab file with ```sudo crontab -e``` and add the line:

```
@reboot /home/pi/path-to-virtualenv/bin/python /home/pi/path-to-sht-sensor/sht -v -trd 20 21
```

So upon bootup, one measurement is acquired.

kobrok, 2018-04-04:
thanx. I will try

kizniche, 2018-04-04:
Though, I would recommend against such a bandaid to the issue. Are you able to execute the script using the global python3?

```
pyhon3 /home/pi/path-to-sht-sensor/sht -v -trd 20 21
```

kizniche, 2018-04-04:
Well, interestingly, I was able to reproduce your issue. Now I can look or a solution.

kizniche, 2018-04-04:
Okay. I was able to fix the issue. I just released [sht-sensor 18.4.1](https://pypi.org/project/sht-sensor/) with the fix, so you can get the latest version into Mycodo with:

```
~/Mycodo/env/bin/pip install sht-sensor --upgrade
```

Then restart your system or the daemon.

kizniche, 2018-04-06:
It appears this issue has been finally solved. Let me know if there are any other issues that appear. Thanks for the support.

closed 2018-04-06 by kizniche

kobrok, 2018-04-09:
No other isues so far. Thank you for your support.