kizniche/Mycodo #438: GPIO Output not responding properly after fresh install

rowkois, 2018-03-31:
## Mycodo Issue Report:
  - Specific Mycodo Version:Mycodo Version: 5.6.10
 uname - 
Linux raspberrypi 4.14.30-v7+ #1102 SMP Mon Mar 26 16:45:49 BST 2018 armv7l GNU/Linux

#### Problem Description

fresh new installation of Mycodo, install was successfull, webserver is running without error.
Using GPIO as Output Relay starts turns on the relay but turning off doesen't work. 
 changing to "on trigger: low" also doesn't turn of Relay.
### Errors
  No Error was displayed.
2018-03-31 20:57:41,856 - mycodo.input_1 - INFO - Activated in 650.9 ms
2018-03-31 20:57:41,859 - mycodo.daemon - INFO - All activated Input controllers started
2018-03-31 20:57:41,860 - mycodo.daemon - INFO - All activated Math controllers started
2018-03-31 20:57:41,861 - mycodo.daemon - INFO - All activated PID controllers started
2018-03-31 20:57:41,864 - mycodo.daemon - INFO - All activated LCD controllers started
2018-03-31 20:57:42,616 - mycodo.conditional - INFO - Conditional settings refreshed
2018-03-31 20:57:42,619 - mycodo.conditional - INFO - Conditional controller activated in 252.6 ms
2018-03-31 20:57:42,620 - mycodo.daemon - INFO - Mycodo daemon v5.6.10 started in 3.890 seconds
2018-03-31 20:57:42,631 - mycodo.daemon - INFO - 34.67 MB RAM in use
2018-03-31 21:03:58,943 - mycodo.output - INFO - Output 3 setup on pin 17 and turned OFF (OFF=LOW)

Turn Output 3 on doesn't appear in Daemon Log.

### Steps to Reproduce the issue:
1. Fresh install accroding to install manual
2. connect Relayboard to GPIO Pin(s)
3. Define a Connected pin as GPIO on/off
4. Try to turn Output on and then off again.
 
### Additional Notes

kizniche, 2018-03-31:
What is happening at the GPIO pin? Can you connect a volt meter and see if 3.3 volts is being output when it's expected to be high?

kizniche, 2018-04-01:
I've had plenty of moments with new relay boards not working, spending a long time in the software, only to find it was miswired.

So it's all working now?

rowkois, 2018-04-01:
yes, it's working, thanks for your adwise.

closed 2018-04-01 by kizniche