kizniche/Mycodo #439: DS18B20 Precision

drgrumpy, 2018-04-02:
Discussion in another post, just made me think about precision settings for DS18B20s....

So this is just for info in case it helps others, as I can't find anything in the Mycodo code that sets or interrogates these values, so just to make people aware....

The value can be stored in eprom, this means that sensors from different sources or used
in different systems, could have inconsistent default precision settings.....(as was the case for me - I have one that defaults to 11 bit and one that defaults to 9 bit - this can result in them apparently reading very different temperatures when they are actually quite close...)

It would be a good idea to set all sensors on the bus to the same precision at startup.... but which to use ?
Mode	Resol	Conversion time
9 bits	0.5°C	93.75 ms
10 bits	0.25°C	187.5 ms
11 bits	0.125°C	375 ms
12 bits 	0.0625°C	750 ms

A good compromise could be 11 bit as a default... or maybe 12 bit, but then more care is needed not to interrogate too freuquently

Perhaps it is does not need to be done in Mycodo as it easy to do from the terminal:

From: https://www.kernel.org/doc/Documentation/w1/slaves/w1_therm:

> Writing a value between 9 and 12 to the sysfs w1_slave file will change the
> precision of the sensor for the next readings. This value is in (volatile)
> SRAM, so it is reset when the sensor gets power-cycled.
> 
> To store the current precision configuration into EEPROM, the value 0
> has to be written to the sysfs w1_slave file. Since the EEPROM has a limited
> amount of writes (>50k), this command should be used wisely.

So to set temporarily:
`$ sudo sh -c "echo 11 >/sys/bus/w1/devices/28-000005f35388/w1_slave"`

To make it persistent:
`$ sudo sh -c "echo 0 >/sys/bus/w1/devices/28-000005f35388/w1_slave"`

Alternatively it could be done within mycodo using the python library:
```
sensor = W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000588806a")
sensor.set_precision(11)
```

If the persist argument is set to False this value is "only" stored in the volatile SRAM, so it is reset when the sensor gets power-cycled.

If the persist argument is set to True the current set precision is stored into the EEPROM. Since the EEPROM has a limited amount of writes (>50k), prob. should not be done in Mycodo

```
sensor = W1ThermSensor(W1ThermSensor.THERM_SENSOR_DS18B20, "00000588806a")
sensor.set_precision(11, persist=True)
```

kizniche, 2018-04-03:
How about this?

[image]

drgrumpy, 2018-04-03:
Wow, very impressive, so quick.... you clearly have too much time on your hands... I feel I should make the effort to understand the code be able to make direct contributions, but when you make this additions so quickly.... 

I guess this would be a 'temporary' setting... as it would get applied every time Mycodo starts up ? What would it default to if no selection is made ? of should there be a default option that sets no value and hence uses whatever is stored in the device

I have a general concern about how much this and other added features add to the footprint of Mycodo... (but of course you  have a much better view of this)... so for me being able to set within Mycodo is not that important as it easy to do from the cli (my approach has been to write permanently to eeprom once with 11-bit)

kizniche, 2018-04-03:
I added these inputs: DS18S20, DS1822, DS28EA00, DS1825, MAX31850K, then added the option to set the SRAM resolution in the Input options of DS18B20, DS1822, DS28EA00, DS1825. In the Calibration & Setup page, I added the ability to set the EEPROM resolution of DS18B20, DS1822, DS28EA00, DS1825. This should cover all bases. The Input option is nice because you can select from leaving it use the Chip Default (whatever is in EEPROM) or set the SRAM resolution when the Input controller is started, so all measurements after that are at the set resolution.

closed 2018-04-06 by kizniche