kizniche/Mycodo #440: How to use Sunrise and Sunset

drgrumpy, 2018-04-02:
I guess this might be a feature request,

I would like to ramp my temperature PID set point according to daylight....

i.e. at sunrise start ramping up the temp to day setpoint
at sunset (or say an hour before) start ramping down to night setpoint

I see there is a sunriseset.py utility, but I can't see any way to easily use it....

It would be good to be able to replace the time value in method tracking with the output from this...
if time > sunrise ramp up
if time > sunset ramp down

Or to start the tracking based on a measured light level, conditional value.....
e.g. if lux < 200 ramp down
lux > 200 ramp up

frodus17, 2018-04-02:
This would be something you can do under Methods. Methods allows you to set a daily (time based) start time and end time. Then, under your PID, under Setpoint Tracking Method, you can link the two. This allows time based ramp/soak like you're asking.

kizniche, 2018-04-02:
> I see there is a sunriseset.py utility, but I can't see any way to easily use it....

Yes, I made a working class for determining the sunrise/sunset based on time and geolocation, however I haven't attempted to integrate it into Mycodo.

drgrumpy, 2018-04-02:
@frodus17 thanks, but yes I know about that and that is what I am already doing.... but the time of sunrise and sunset changes daily ! 
@kizniche that would explain why I couldn't find any way to use it !

frodus17, 2018-04-02:
@drgrumpy Oh, I get it now. It's a dynamically changing Method, based on the sunrise/sunset in your area code, correct? Seems like a cool idea.

drgrumpy, 2018-04-02:
Exactly.....
For plant growth e.g. in a greenhouse you want the temperature to ramp up with the light in the morning and ramp down in the evening....
A glasshouse controller I used to use tried to figure out the sunrise and set times based on light levels the previous day... but after a dull day in winter it would draw the curtains and put the plants to bed too early !
I suppose basically what would be needed is for the method to be triggered by a conditional ... that way it has more flexibility...and the python sunsetrise script could output a simple 1 or 0 for day/night.

kizniche, 2018-04-02:
I was in a particularly productive mood today...

[image]

kizniche, 2018-04-02:
I just tested and the sunrise/sunset conditional appears to be working. It'll go out in the next release. Please test and report back here if any issues occur.

[image]

closed 2018-04-02 by kizniche

drgrumpy, 2018-04-07:
I may be being dense, but I can't see a way to get the sunrise / sunset times or day/night status to the input of a method time ?
I.e. in place of the time value in a daily method tracking to be able to use the sunrise, sunset values to start ramping the pid up/down....
Just using the sunrise condition to change he PID setpoint won't allow me to ramp up/down, although it could be done in steps......

kizniche, 2018-04-07:
I would set up a sunrise PID and a sunset PID, with the appropriate method for each (sunrise ramp up, sunset ramp down). Then create two sunrise/sunset conditionals. Use the sunrise trigger to start the sunrise PID and sunset trigger to start the sunset PID. Have the sunset conditional also trigger turning off the sunrise PID, and the sunrise trigger turn off the sunset PID.

kizniche, 2018-04-07:
Does that setup accomplish what you're trying to do?

kizniche, 2018-04-08:
I updated the original instructions with more details.

kizniche, 2018-04-08:
Actually, the way I described will not work. Give me a moment to develop another method.

kizniche, 2018-04-08:
I just realized there's an important piece of this setup that's missing. I'll describe the setup first, then discuss me adding support for the missing piece:

 - Create two Duration methods, one for sunrise and the other for sunset. The sunrise will ramp up from 0 to 100 (% duty cycle) and the other will do the opposite.
 - Make the end of the sunrise method (when it;s at 100) continue for a sufficient duration (86400 seconds) to be able to reach sunset. The same for the sunset method, but it will continue for 86400 seconds at 0.
 - Set up two PWM Method Timers, one for sunrise and the other for sunset, and select the appropriate method for each.
 - Create two sunrise/sunset conditionals.
 - Set the sunrise conditional to start the sunrise Timer and stop the sunset Timer.
 - Set the sunset conditional to start the sunset Timer and stop the sunrise Timer.

Now, the issue is there isn't a conditional action to turn timers on and off. This will be easy to add, so I'll work on adding support so this setup will work.

kizniche, 2018-04-08:
Actually, my first set of instructions were the correct way to accomplish changing the temperature. I had it in my mind you were using PWM for lighting. Looking back at your first post, it's temperature you want to affect. In that case, you will want to use PIDs, not Timers. Follow [these instructions](https://github.com/kizniche/Mycodo/issues/440#issuecomment-379507368) but instead of your ramp up and down being PWM, make it whatever temperature you want.

drgrumpy, 2018-04-08:
Ah ok, I think I got it now, yes I am controlling temperature, heater output duration.... what I hadn't got was that I need to have two PIDs to control one output and switch from one to the other.....
.... but am I correct that that will mean that I will loose the i, and d values when the switch over occurs? The i value in particular is quite important for ensuring stability in my set up....
I won't be able to test for a couple of days...

kizniche, 2018-04-08:
It seems the best way to solve this issue is to be able to merely change the PID method while the PID is running. A PID method changing conditional action seems appropriate.

kizniche, 2018-04-08:
I just added the conditional action to change a PID method (while running or stopped). This should preserve all controller values and enable a smooth transition from your rise method to your fall method. This is a great feature addition.

drgrumpy, 2018-04-13:
A question:
I have set up the sunrise and sunset conditionals to start the two duration methods ramping the temp up and down....(we will have to wait until the next sunrise to see if it takes effect), does it matter what method is set in the pid ? Presumably whatever is there will get over ridden when the sunrise triggers the ramp up method ? Or should it be set as disabled ?

drgrumpy, 2018-04-14:
Brilliant, thanks, I can confirm that so far (sunrise) it has worked as expected.

drgrumpy, 2018-04-14:
Unfortunately I spoke to soon, the sunset changeover did not happen, and it failed to set any set point:
from the log at the point the sunset method should have kicked in:

> 2018-04-14 18:27:05,712 - mycodo.pid_1 - ERROR - Run Error: time data 'Ready' does not match format '%Y-%m-%d %H:%M:%S.%f'
> Traceback (most recent call last):
> 
>   File "/var/mycodo-root/mycodo/controller_pid.py", line 199, in run
>     self.logger)
>   File "/var/mycodo-root/mycodo/utils/method.py", line 194, in calculate_method_setpoint
>     str(controller.method_start_time), '%Y-%m-%d %H:%M:%S.%f')
>   File "/usr/lib/python3.5/_strptime.py", line 510, in _strptime_datetime
>     tt, fraction = _strptime(data_string, format)
>   File "/usr/lib/python3.5/_strptime.py", line 343, in _strptime
>     (data_string, format))
> ValueError: time data 'Ready' does not match format '%Y-%m-%d %H:%M:%S.%f'

kizniche, 2018-04-14:
Here is the likely solution to this issue. Note that line numbers may not be the same because this file has undergone edits since your upgrade.

Edit this:

https://github.com/kizniche/Mycodo/blob/add_unique_id/mycodo/controller_pid.py#L752-L767

So it looks like this:

```python
    def set_method(self, method_id):
        """ Set the method of PID """
        self.method_id = method_id

        with session_scope(MYCODO_DB_PATH) as db_session:
            mod_pid = db_session.query(PID).filter(
                PID.unique_id == self.pid_id).first()
            mod_pid.method_id = method_id
            mod_pid.method_start_time = 'Ready'
            mod_pid.method_end_time = None
            db_session.commit()

        self.initialize_values()

        if self.method_id != '':
            self.setup_method(method_id)

        return "Method set to {me} and started".format(me=method_id)
```

kizniche, 2018-04-14:
Actually, that may not be the fix. I'm continuing to investigate.

kizniche, 2018-04-14:
Try this change instead.

Edit this:

https://github.com/kizniche/Mycodo/blob/14f8bfe460f918e911db281d12d9e6d3e82e8259/mycodo/controller_pid.py#L752-L767

To look like this:

```python
    def set_method(self, method_id):
        """ Set the method of PID """
        with session_scope(MYCODO_DB_PATH) as db_session:
            mod_pid = db_session.query(PID).filter(
                PID.unique_id == self.pid_id).first()
            mod_pid.method_id = method_id
            mod_pid.method_start_time = 'Ready'
            mod_pid.method_end_time = None
            db_session.commit()

        if method_id != '':
            self.setup_method(method_id)

        return "Method set to {me} and started".format(me=method_id)
```

Then, at the end of setup_metho():

https://github.com/kizniche/Mycodo/blob/14f8bfe460f918e911db281d12d9e6d3e82e8259/mycodo/controller_pid.py#L307-L356

Add this to the end:

```python
            if method_id != '':
                self.method_id = method_id
```

So the end looks like this:

```python
            else:
                # Method neither instructed to begin or not to
                # Likely there was a daemon restart ot power failure
                # Resume method with saved start_time
                self.method_start_time = datetime.datetime.strptime(
                    str(pid.method_start_time), '%Y-%m-%d %H:%M:%S.%f')
                if method_data_repeat and method_data_repeat.duration_end:
                    self.method_end_time = datetime.datetime.strptime(
                        str(pid.method_end_time), '%Y-%m-%d %H:%M:%S.%f')
                    if self.method_end_time > datetime.datetime.now():
                        self.logger.warning(
                            "Resuming method {id}: started {start}, "
                            "ends {end}".format(
                                id=method_id,
                                start=self.method_start_time,
                                end=self.method_end_time))
                    else:
                        self.method_start_act = 'Ended'
                else:
                    self.method_start_act = 'Ended'

            if method_id != '':
                self.method_id = method_id
```

drgrumpy, 2018-04-15:
changes made and restarted daemon...

kizniche, 2018-04-15:
I just noticed the last code change I mentioned won't be able to disable the method. This change should fix that, at the end of setup_method() in Mycodo/mycodo/controller_pid.py:

```python
            if method_id != '':
                self.method_id = method_id
```

use just this:

```python
            self.method_id = method_id
```

drgrumpy, 2018-04-16:
After the last changes, sunrise ramp up happened ok, but got the following:

2018-04-16 06:46:15,849 - mycodo.conditional - ERROR - Run Error: unorderable types: NoneType() < float()
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_conditional.py", line 123, in run
    self.timer_period[each_id] < time.time() and
TypeError: unorderable types: NoneType() < float()

kizniche, 2018-04-16:
At this point I don't think I'm going to put too much time in diagnosing this. Rather I'm going to focus it in getting 6.0 ready for release. I don't want to distract myself with fixing something I likely already fixed in the current 6.0 build I'm making. In any case, I'll test in 6.0 what you're saying isn't working in 5.x

drgrumpy, 2018-04-16:
OK, I ended re-booting as the web server became unresponsive....
I think I will restore the original 5.7.2 version, it will be safer for my plants ....

kizniche, 2018-04-16:
Sorry. I thought it would be an easy fix. I didn't want to get too much into it, then I'd be fixing the same bugs in two different versions. If this was a normal version jump, no problem, but I've made extensive changes to get to 6.0

drgrumpy, 2018-04-16:
Makes sense