kizniche/Mycodo #441: Cooling PID and PWM not displaying, working

drgrumpy, 2018-04-02:
## Mycodo Issue Report:
  - Specific Mycodo Version: 5.6.10
(just upgraded from 5.6.6, in case the issue had been fixed...
 
#### Problem Description

I have PID cooling control set up:
[image]

The PWM is output to a fan...
It show as active on the output page.....

The fan is switched on by a relay when the PWM value is less than -15 (as it cooling)
The fan is switched off by a relay when PWM is greater than -10

### Errors
When testing a while ago (version 5.6.4 or 5 ?) this was working as expected..

No Cooling PWM value displays on the dashboard (this should happen irrespective of the relay)
Now the fan never turns on....
PID display only ever displays 'No Data', even when it should be active
[image]

My guess is that this means it is not saved in the db and so is not accessible to the conditionals, so they never trigger...

Nothing odd is recorded in the logs...

kizniche, 2018-04-02:
I have been able to replicate the issue with the PID control dashboard element. I just pushed a fix for that.

> Now the fan never turns on....

However, I have not been able to replicate your issue where the PWM isn't being generated. Have you gone to the Output page to verify the PWM isn't turning on? My PWM generates nicely when I set up a simple PID controller to lower using a PWM output.

kizniche, 2018-04-02:
Additionally, check the values stored in the database by adding a graph and selecting the Temperature Input and PWM Output as well as the PID Setpoint and PID Output Duty Cycle to verify if the PID controller is working properly.

drgrumpy, 2018-04-02:
Sorry may be I was not clear, I was in a panic when I noticed that the fans hadn't come on and the temperature was climbing (good job no plants put in yet !)...
So yes in the ouput page the expected PWM value was showing, i.e. the PWM is generated..
but nothing appeared on the main graph (it was supposed to be and used to when testing)... but the relay (set to turn on when pwm < -15) did not... (I manually over-rode the relay and the fan indeed turned on)...
If you recall all my testing worked because I raised the issue about negative PWM values for cooling....
and the fact that I had to think in reverse for the relay trigger values....so yes I already have the Temp input and PWM output set to display and as you saw also the PID values.
I'll apply the fix, but won't be able to report more until tomorrow and if the sun comes out...

Is there a recommended/quick way to apply the fix? it took 25 mins to do the upgrade from to 5.6.10 earlier...

kizniche, 2018-04-02:
Well, the fix I implemented only dealt with displaying the proper values on the PID dashboard element. It did not address any issues with outputs actually turning on/off because I could not replicate your output issue. You can always restore a previous backup (backups are created before any upgrade occurs).

kizniche, 2018-04-02:
Have you created a graph like I suggested to determine if any of the values I previously mentioned are being stored?

drgrumpy, 2018-04-03:
Yes (but I haven't updated with your fix):
And now there is some sunshine, so I can test....to summarize:
Output page: Cooling PWM value shows value of ~50%
Dashboard graphs:
Cooling PID P - displays
Cooling PID I - displays
Cooling PWM duty cycle - nothing
Cooling PID Output duty cycle - nothing
so I guess the 'nothing' means the data is not being saved and so are not available to the conditional that turns on the relay...

drgrumpy, 2018-04-03:
So I figured maybe if I deactivate the original cooling PID and create another identical one and activate that it will trigger stuff to start saving the data.....
But... even though the second cooling PID is identical when I try to activate it gives:

> Error: Actuate PID: Cannot activate PID controller if raise and/or lower relay IDs are not selected

Output Raise = disabled
Output Lower = Fan PWM

(The identical original activates without error)
BTW still trying to figure out how to apply the fix without doing an 'upgrade' (relatively new to git)

kizniche, 2018-04-03:
I'll investigate this issue today. Here is how to upgrade to the head of the repo https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch

kizniche, 2018-04-03:
Keep in mind you may ruin your Mycodo install if you do upgrade this way

kizniche, 2018-04-03:
I am constantly messing with the database schema and it is not finalized for release. If I change it after you upgrade to master, your database is toast come release

kizniche, 2018-04-03:
When you go to the "System Information" page, are all the text at the top colored green?

kizniche, 2018-04-03:
Quick question... Did you happen to leave the Direction set to Both?

kizniche, 2018-04-03:
I confirmed that error you received was most likely due to an improperly set PID Direction.

drgrumpy, 2018-04-03:
Yes all coloured green on system info page

drgrumpy, 2018-04-03:
Oops... yes I had it at 'raise', sorry, so now just deactivated the original PID and activated the now identical replacement without error and......

drgrumpy, 2018-04-03:
No error, but still no PWM output in the graphs...
[image]

(I overwrote the files with the latest using rsync and ran the alembic upgrade manually)
It was def. working as I expected... maybe back in version 5.6.4 (but not in 5.6.5 because that caused me to update to 5.6.10)

kizniche, 2018-04-03:
Just so I completely understand you, the PWM is being generated, but the value is just not being written to the database?

drgrumpy, 2018-04-03:
Output:
[image]

drgrumpy, 2018-04-03:
[image]

drgrumpy, 2018-04-03:
[image]

kizniche, 2018-04-03:
I am unable to replicate your issue. It has been working fine for me. I tested yesterday and just now I made a new PWM output and PID:

[image]

kizniche, 2018-04-03:
I should have asked sooner, but are there any errors in the daemon log?

drgrumpy, 2018-04-03:
No, at least not related (saturated errors for the TSL2561)

kizniche, 2018-04-03:
Do you have a different SD card to install Raspbian/Mycodo fresh to rule out confounding variables?

drgrumpy, 2018-04-03:
Yes, that was my next thought......
..... problem is I had all this working on test.... but now the pi is installed in a box, with relays and lcd...... not sure I will be able to swap sd cards (easily, bad design on my part)..... so it may take some time... or it may be quicker uninstall and start afresh.... I need to physically go the box....

drgrumpy, 2018-04-03:
Just looking at the data directly in influx db:

> SELECT * FROM duty_cycle  where time > now() - 1h
name: duty_cycle
time                           device_id                            value
----                           ---------                            -----
2018-04-03T16:37:23.6630854Z   d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T16:38:23.675030537Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T16:39:23.908419615Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T16:40:23.733133935Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T16:41:23.786149658Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:04:24.520736179Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:05:23.610352056Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:06:23.958061332Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:07:24.137315481Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:08:23.997291593Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:09:24.6330571Z   d44299ac-c43a-4042-b57c-28c0c0f161c4 0

Whereas the time now is 17:20 ! and cooling pwm is active, so it clearly points to the data not being saved in the db for some reason.....

Might it be quicker to delete/recreate the influx db ?
Is the issue that it is not saving -ve values to the db ?

drgrumpy, 2018-04-03:
Yes... and a further clue the time of that last measurement would have been when the cooling fan should have kicked in (i.e. the value would have gone negative...)

drgrumpy, 2018-04-03:
And hey presto, if I change the set point (at 17:44) so that the cooling switches off, the data is again saved:
> SELECT * FROM duty_cycle  where time > now() - 1h
name: duty_cycle
time                           device_id                            value
----                           ---------                            -----
2018-04-03T17:04:24.520736179Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:05:23.610352056Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:06:23.958061332Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:07:24.137315481Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:08:23.997291593Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:09:24.6330571Z   d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T17:44:27.154453459Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
>

kizniche, 2018-04-03:
This may be an issue I came across a while ago with influxdb, that if the first value for a measurement is an integer, any following non-integer (float) values are not able to be saved. You can try erasing your influxdb database and recreating it to test this.

kizniche, 2018-04-03:
I can't recall the exact command to drop the infludb database ("mycodo_db"), but the command to recreate it again is:

```
sudo ~/Mycodo/mycodo/scripts/upgrade_commands.sh update-influxdb-db-user
```

drgrumpy, 2018-04-03:
So I deleted and recreated the influxdb:

sudo service mycodo stop
sudo influxd backup /mycodo_db_snap
influx
> DROP DATABASE mycodo_db
exit
sudo /home/pi/Mycodo/mycodo/scripts/upgrade_commands.sh update-influxdb-db-user
sudo service mycodo start

all ok...
back in mycod web interface set the cooling setpoint so that it kicks in, same result data values cease in the database when the duty cycle becomes negative, and appear again when the value returns to zero (ie. from 18:10 to 18:16 !) 

> SELECT * FROM duty_cycle  where time > now() - 1h
name: duty_cycle
time                           device_id                            value
----                           ---------                            -----
2018-04-03T18:08:52.507706299Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T18:09:53.563739996Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T18:10:54.885373252Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T18:16:52.560623508Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T18:17:52.512374794Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
2018-04-03T18:18:52.725206163Z d44299ac-c43a-4042-b57c-28c0c0f161c4 0
>

drgrumpy, 2018-04-03:
I have been digging around influxdb and as you indicate:

```
> show field keys from duty_cycle
name: duty_cycle
fieldKey fieldType
-------- ---------
value    integer
```

it seems this may be 'by design' in influxdb and 

> Field value types cannot differ within a shard but they can differ across shards.

kizniche, 2018-04-03:
I may have to, upon the initial installation, add a dummy measurement for every possible measurement, to ensure the fieldType is initialized as float. Otherwise, have you come across any documentation about changing the fieldType after the fact?

drgrumpy, 2018-04-03:
From my random digging around and testing in influxdb:

```
> create database testdb
> use testdb
> insert test1 value=0
> insert test2 value=0.0
> show field keys from test1
name: test1
fieldKey fieldType
-------- ---------
value    float
> show field keys from test2
name: test2
fieldKey fieldType
-------- ---------
value    float
```
So it looks like influxdb is not the problem and will default a number to a float...

However, the python client library InfluxDBClient used in Mycodo uses json and if json is supplied with '0' it will set the type to int
the same will happen if the value is 1, 2, 3 etc.
So I guess this is where the problem may lie.....

The workaround we need to make json interpret all values as floats,
do we need to just wrap the values in float() in influx.py ?
(unless they have already been made into a float ?)

kizniche, 2018-04-03:
This is how Mycodo formats the data: https://github.com/kizniche/Mycodo/blob/7320d2dd44638ef5600d4b8e1169583dec86af7b/mycodo/utils/influx.py#L41-L83

And is subsequently entered into the influxdb here: https://github.com/kizniche/Mycodo/blob/7320d2dd44638ef5600d4b8e1169583dec86af7b/mycodo/utils/influx.py#L232-L268

I can't test anything at the moment, but that's how Mycodo interfaces with influxdb.

drgrumpy, 2018-04-03:
format_influx_db_data
-> value -> float(value)
delete and recreate database as above...
restart mycodo:

````
> SELECT * FROM duty_cycle  where time > now() - 1h
name: duty_cycle
time                device_id                            value
----                ---------                            -----
1522787867397045233 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
> show field keys from duty_cycle
name: duty_cycle
fieldKey fieldType
-------- ---------
value    float
> 
```
Change cooling set point to force fan to start:
```
> SELECT * FROM duty_cycle  where time > now() - 1h
name: duty_cycle
time                device_id                            value
----                ---------                            -----
1522787867397045233 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522787927333209356 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522787987520528788 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522788047227388205 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522788108157614511 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522788168319054099 d44299ac-c43a-4042-b57c-28c0c0f161c4 0
1522788227676173014 d44299ac-c43a-4042-b57c-28c0c0f161c4 -95
1522788228171603420 501c3e16-6d93-4c43-9041-a8d13bd804a4 -100
```
Graphs show pwm and relay switches on as expected
Hopefully wrapping the value in float will not have any other consequences....
> 
E

drgrumpy, 2018-04-03:
Just noticed your comment before I finished writing my post...

kizniche, 2018-04-03:
So changing this:

https://github.com/kizniche/Mycodo/blob/7320d2dd44638ef5600d4b8e1169583dec86af7b/mycodo/utils/influx.py#L255-L258

To this:

```python
    data = [format_influxdb_data(device_id,
                                 measure_type,
                                 float(value),
                                 timestamp)]
```

Solves the issue?

Pressure is stored in the influxdb as an integer, so if this fix is implemented, all users that use pressure sensors may cease to have values entered into the database. Though, I believe that if the influxdb fieldType is float, an integer will be entered, but not the opposite (as evinced just now).

drgrumpy, 2018-04-03:
Should have said changes are on lines 71 and 81

kizniche, 2018-04-03:
I think changes can go either place. My concern now, with implementing this fix, is it's most likely going to impact pressure measurement storage, unless the fieldType can be changed.

kizniche, 2018-04-03:
if fieldType can be changed, I can add a command to be executed during the user's database schema upgrade process. I'll have a look around later, unless you're feeling ambitious ;)

drgrumpy, 2018-04-03:
Again overlap in posts....
From my reading up on influxdb... it will by default interpret all numeric values as a float unless explicitly cast to an integer using the 'i' suffix:
```
> insert test3 value=0
> show field keys from test3
name: test3
fieldKey fieldType
-------- ---------
value    float
> insert test3 value=0i
ERR: {"error":"partial write: field type conflict: input field \"value\" on measurement \"test3\" is type integer, already exists as type float dropped=1"}
```

kizniche, 2018-04-03:
So, if the influxdb fieldType is integer, and a float is entered, the value will successfully be entered?

drgrumpy, 2018-04-03:
or I guess supplied as type integer by json...
the issue could also appear intermittently depending on the sharding of influxdb database - the data types can be different in different shards !

drgrumpy, 2018-04-03:
No, it will fail
```
> insert test4 value=1i
> show field keys from test4
name: test4
fieldKey fieldType
-------- ---------
value    integer
> insert test4 value=1.5
ERR: {"error":"partial write: field type conflict: input field \"value\" on measurement \"test4\" is type float, already exists as type integer dropped=1"}
```

drgrumpy, 2018-04-03:
According to the influx db faqs:

> **Can I change a field’s data type?**
> Write the Data to a Different Field
> 
> The simplest workaround is to begin writing the new data type to a different field in the same series.
> Work the Shard System
> 
> Field value types cannot differ within a shard but they can differ across shards.
> 
> Users looking to change a field’s data type can use the SHOW SHARDS query to identify the end_time of the current shard. InfluxDB will accept writes with a different data type to an existing field if the point has a timestamp that occurs after that end_time.
> 
> Note that this will not change the field’s data type on prior shards. For how this will affect your queries, please see How does InfluxDB handle field type discrepancies across shards.

I am still getting my head around influx db terminology but I think this could mean that you just change the different device_id that way all the old data will remain....
Or to use shards we would need to create a proper retention policy (something I was going to look into at some point) instead of relying on the autogen...(I couldn't find anything about this in mycodo and it seems that it is just defaulting to autogen)

kizniche, 2018-04-03:
Changing the device_id won't work, as that's just a tag. The measurement itself (i.e. pressure) is what is assigned the data type. Therefore, all pressure measurements (regardless of the ID tag) are currently integer. This is why several sensors (each with a unique ID) can add pressure measurements to the database.

drgrumpy, 2018-04-03:
A bit more testing, just changing the device_id won't work...
need to change the 'measurement' i.e.
pressure -> pressure_flt

Or....
I think the changes need to be in: format_influxdb_data()
and just test for measurement_type of pressure, something like:

```
     if measure_type == 'pressure' : 
         return { 
             "measurement": measure_type, 
             "tags": { 
                 "device_id": device_id 
             }, 
             "fields": { 
                 "value": int(value) 
             } 
    else:
        return {
            "measurement": measure_type,
            "tags": {
                "device_id": device_id
            },
            "fields": {
                "value": float(value)
            }
        }
} 
```
I am not used to python, so not always clear about what is going on in your code..may need another level of nesting to handle the if timestamp

kizniche, 2018-04-03:
This looks like a good solution, although hackish, it does get the job done with minimal obtrusion. Though, I will need to include a "# TODO: Delete this for version 7.0.0 and force all values to float" to remind myself to remove this at the next major version jump (when users cannot upgrade and must reinstall fresh).

drgrumpy, 2018-04-03:
My second guess at writing python !

kizniche, 2018-04-03:
```python
    # TODO: Remove this check at next major version, force pressure to be float
    if measure_type == 'pressure':
        checked_value = int(value)
    else:
        checked_value = float(value)

    if timestamp:
        return {
            "measurement": measure_type,
            "tags": {
                "device_id": device_id
            },
            "time": timestamp.strftime('%Y-%m-%dT%H:%M:%S.%fZ'),
            "fields": {
                "value": checked_value
            }
        }
    else:
        return {
            "measurement": measure_type,
            "tags": {
                "device_id": device_id
            },
            "fields": {
                "value": checked_value
            }
        }
```

drgrumpy, 2018-04-03:
Cool

kizniche, 2018-04-03:
Thanks for the help.

closed 2018-04-03 by kizniche