kizniche/Mycodo #442: install mycodo error???

bike2538, 2018-04-03:
what happen ?!
[image]

drgrumpy, 2018-04-03:
You haven't told us what you did, or your os
Are you using the Raspbian ?
Did you go to 'Releases', download the relevant zip or gz file, then extract the files and run the install script?

bike2538, 2018-04-03:
Sorry Guys for not giving enough details.  The screen showed no sign of the process running.  So I thought it crashed.  I just found that I was not patient enough to wait until the installation finish as the internet at home was so so slow.  Now the installation was successful.  I would suggest that it could be great if there is some kind of progress bar during the installation process. 
Thank you very much.

closed 2018-04-06 by kizniche