kizniche/Mycodo #443: Bug: Raspberry pi crashing after 2-3 days of uptime.

Kattihatt, 2018-04-03:
Hello again!

I were running 5.6.9 when this happened. I'm suspecting it's Mycodo for two reasons. It's almost the only thing I'm using it for, and the daemon.log is nearly full of Mycodo. Nothing else. 

Here is what the graphs look like: https://imgur.com/a/Ci8o7

It stops taking measurements, it stops being responsive and both the red and the green lights are constantly on, on the RPi. The CPU feels very warm, but now I can at least touch it after I put on a heatsink. This happened twice before but then it was untouchable. 

Mycodo log: http://paste.debian.net/1018310/
I don't know how to paste large log files so that I can include everything from this day. Log file would be around 40 MB. How can I do this? Most of these pastesites don't accept that large files it seems. But here's a portion of what fills up most of the log file, for the time that is seen in the images: https://pastebin.com/NT4dCf9X

Any ideas to what could be causing this?

kizniche, 2018-04-03:
What is the period of your measurements. You may be taking them too soon on the same ADC, causing locking issues.

Kattihatt, 2018-04-05:
I changed all the times so that they're 15 seconds apart. When I went to bed last night it had crashed again. Don't really know how to find the cause. There's nothing shown in the logs apart from what is seen there :/

kizniche, 2018-04-05:
Are measurements able to be acquired from channels 6 and 7 when the script is ran manually (with Mycodo not attempting to acquire measurements in the background)?

What I would do is deactivate everything in Mycodo, then activate one part at a time (one channel), until an issues arises. This way we can figure out at what point (and what part) causes the catastrophic issue you're having (daemon stops).

Also, check with ```ps aux | grep mycodo_daemon``` to see if the daemon has actually stopped, or it's just not responding.

I pushed a commit just now that will handle the errors with locking better (makes the messages smaller), so if you want to upgrade, use [these instructions](https://github.com/kizniche/Mycodo/wiki/Updating-to-Github-Master-Branch) and restart the daemon, then continue testing.

kizniche, 2018-04-05:
There are things which I cannot test because I don't have this ADC. The first thing I would test is whether or not the file locking for the MCP3008 is even necessary. If it's not, we can remove all that code and not even worry about it slowing down reads. Those locking lines are as follows:

Delete this:

https://github.com/kizniche/Mycodo/blob/21b2b153bcc13bba7a89d417646a05c11d516f7b/mycodo/devices/mcp3008.py#L26-L27

Change this:

https://github.com/kizniche/Mycodo/blob/21b2b153bcc13bba7a89d417646a05c11d516f7b/mycodo/devices/mcp3008.py#L30-L55

To this:

```python
    def read(self):
        """ Take a measurement """
        self._voltage = None
        lock_acquired = False
        try:
            self.adc = Adafruit_MCP3008.MCP3008(clk=self.clockpin,
                                                cs=self.cspin,
                                                miso=self.misopin,
                                                mosi=self.mosipin)
            self._voltage = (self.adc.read_adc(self.channel) / 1023.0) * self.volts_max
        except Exception as e:
            self.logger.exception(
                "{cls} raised exception during read(): "
                "{err}".format(cls=type(self).__name__, err=e))
            return 1
```

Then restart the daemon. If measurements are able to be consistently acquired without this locking, we can exclude this from causing the issue.

kizniche, 2018-04-05:
Perhaps it may also be an issue with the polling frequency, with too little time between measurements. You can also try adding a pause before the measurement by changing:

https://github.com/kizniche/Mycodo/blob/21b2b153bcc13bba7a89d417646a05c11d516f7b/mycodo/devices/mcp3008.py#L44-L50

To this:

```python
            if lock_acquired:
                import time
                time.sleep(0.5)
                self.adc = Adafruit_MCP3008.MCP3008(clk=self.clockpin,
                                                    cs=self.cspin,
                                                    miso=self.misopin,
                                                    mosi=self.mosipin)
                self._voltage = (self.adc.read_adc(self.channel) / 1023.0) * self.volts_max
                lock.release()
```

kizniche, 2018-04-06:
FYI, the latest commit, above, moves the initialization of the MCP3008 class out of the measurement function and also adds a 0.1 second delay before measurements (to prevent rapid back-to-back measurements).

Kattihatt, 2018-04-08:
> Are measurements able to be acquired from channels 6 and 7 when the script is ran manually (with Mycodo not attempting to acquire measurements in the background)?
>
> What I would do is deactivate [...]

I haven't had this issue with the readings since after the first crash. Just doesn't appear any more and they've all been fine since, so I'll have to wait for them to happen again.

I don't know if the daemon stops only, the whole Raspberry Pi becomes unaccessible, the green light is not blinking any more and it doesn't respond to my GPIO reboot or shutdown buttons. I have to hard reset it when this happens.

> Also, how long do you have the pre-output run, for each input?

Pre-output is set to 5 seconds. Pause in between readings is 15 seconds. 

I just did the upgrade. I shall change the code soon. 

Also after updating, my graph for the Linux command data input isn't working. I can make a new graph, I can change whatever and everything works fine until I put that CO2 sensor readings into it from the linux command input. Don't know what that can be about.

Kattihatt, 2018-04-08:
No crashes, but I did have this during the night though: https://i.imgur.com/Oi8vzx7.png Became unresponsive for a while, CPU temperature rose quickly and then it all seemed to go back to normal.

kizniche, 2018-04-08:
> I just did the upgrade. I shall change the code soon.

I incorporated the 0.1 second pause in the latest release, so if you don't experience any issues since the upgrade, this may be fixed.

kizniche, 2018-04-08:
However, if you want to test if the locking is necessary, the new changes are as follows:

Delete this:

https://github.com/kizniche/Mycodo/blob/cc11e65a78ce48f6b9568f6d32ba4d46d7a86fd9/mycodo/devices/mcp3008.py#L27-L28

Change this:

https://github.com/kizniche/Mycodo/blob/cc11e65a78ce48f6b9568f6d32ba4d46d7a86fd9/mycodo/devices/mcp3008.py#L27-L28

To this:

```python
    def read(self):
        """ Take a measurement """
        self._voltage = None
        try:
            sleep(0.1)
            self._voltage = (self.adc.read_adc(self.channel) / 1023.0) * self.volts_max
        except Exception as e:
            self.logger.exception(
                "{err}".format(cls=type(self).__name__, err=e))
            return 1
```

kizniche, 2018-04-10:
Also, if it appears to be the fault of Mycodo, turn off all Mycodo controllers and enable one-by-one to determine what exact part is failing.

Kattihatt, 2018-04-11:
No, it crashes anyway! Don't know what could be the cause. I have two Raspberry Pis and this happens to both of them. I have no idea what I'm doing wrong. Starting to feel a bit discouraged. But I can assume that we know that Mycodo isn't the cause of this problem.

Kattihatt, 2018-04-11:
I have this:
> Apr 11 00:03:17 GreenhousePi influxd[352]: ts=2018-04-10T22:03:17.317087Z lvl=info msg="Failed to open shard" log_id=07OPC3hG000 service=store trace_id=07OPC4G0000 op_name=tsdb_open db_shard_id=28 error="[shard 28] error opening memory map for file /var/lib/influxdb/data/mycodo_db/autogen/28/000000002-000000001.tsm: init: error reading magic number of file: EOF"
in my syslog file, even though I had Mycodo turned off with the above mentioned commands though. I don't know what this means.

kizniche, 2018-04-11:
Hmmm. Have you tried ```sudo service influxdb stop``` ?

kizniche, 2018-04-11:
Have you set up anything else after installing Raspbian? Are both of your SD cards the same? Do you have a different brand SD card to test?

kizniche, 2018-04-11:
Don't get too discouraged, because from what I've seen with Pis, this is rare (for both of your Pis to do this). So, it indicates to me that it's most likely something that can be fixed if we find the commonality between your systems. Perhaps it's something to do with the router they both connect to (if they do both connect to the same router). Perhaps try installing Raspbian fresh and see if there are issues at that point.

kizniche, 2018-04-11:
Are you connecting to wifi or ethernet? If wifi, how did you set up the connection?

Kattihatt, 2018-04-14:
Both the SD cards are the same but I'll focus on this Pi. I don't know about the other and if it's the same issue. I've had this one running for two days with Mycodo of and all peripherals disconnected. Then I turned Mycodo back on and no issues so far. I have a screen running with dmesg -w and one with tail -f /var/log/syslog (if that's the path, can't remember). It's been running for two more days so it's looking promising. Maybe it's the ADC after all, but I wouldn't have assumed that the hardware would somehow cause these issues. I'll connect everything but the ADC and run it for a while as well, then lastly connect that one and see what's going on.

kizniche, 2018-04-14:
Pis are sometimes sensitive to electrical disturbances, so if the issue appears when the ADC is connected, you may need to look into alternate wiring to prevent any potential electrical issues (too high current draw, voltage spikes, etc.).

Kattihatt, 2018-04-18:
So I was running with Mycodo on and it took longer than usual, but I think I found the issue. Had a screen running the whole time now. https://i.imgur.com/nh9W33T.jpg I don't know if it's related to Mycodo. I wouldn't now how to check it. I will have the exact time of the crash this time and can check the logs for errors. But even if I know roughly what it is that causes it, I don't know any details about how this chip works and what could be causing it to happen.

kizniche, 2018-04-18:
Is your Pi completely disconnected from all devices (i.e. all ports unused: USB, ethernet, HDMI, camera module, etc.)?

I take it because it's a screenshot of a monitor, you have a keyboard, mouse, and monitor connected. I've read some forums with the kernel panic that the keyboard is possibly causing the issue. That's not to say it's your keyboard, but it may be a peripheral you have connected.

Can you try running your Pi headless (only connected to network and SSH into the pi?

Kattihatt, 2018-04-19:
Yes, HDMI is used for the screen and a keyboard is used so that I could monitor everything. 

I've always ran it headless. Until I started looking for this error and people told me to connect a screen so that I could actually see the crash log screen. But then I can't see this error message and I'm not helped by it. So connecting a keyboard and a screen is a response to it crashing and a way to try to find the error. I don't know if I could do this without the screen since it apparently doesn't show up in logs.

kizniche, 2018-04-20:
If you've experienced this issue with no peripherals attached, I would strongly suggest trying a different SD card. That's the only thing that you can really change at this point, besides the OS, however the OS hasn't been an issue on any Pis I've used or heard of other users I've known.

Theoi-Meteoroi, 2018-04-20:
The raspberry pi power supply must be a reliable 5.0v.  If you see a blinking RED light - the power quality indicator is telling you there is a problem.

As it turns out, the raspberry pi 3b+ needs a solid 5.1v to be stable. They changed the switching power supply design significantly. I have power supplies that worked fine with previous models that will not work reliably with a Pi3B+.  Notably some of these are USB power banks of 10k ma/hrs.   

Power supply is important. Dodgy circuit design can mimick power problems. This is probably (certainly?) not a Mycodo code problem.

kizniche, 2018-05-02:
That's good to have this resolved. Are you still having issues with your ADC, or was this the last of your issues?