kizniche/Mycodo #445: Arduino

magichammer, 2018-04-09:
good morning, Kiznike are excited about Mycodo, I wanted to ask you if my idea to connect an arduino directly to the raspberry in order to use the analog ports, and extend the number of digital ports, what can you tell me about?
Thanks again for everything
Fulvio Paganoni

kizniche, 2018-04-10:
I've seen people do this. You would essentially set up communication, and the easiest way I can think to get the measurements into Mycodo would be to add a Linux Command Input that will execute a script and store the returned value. You would then create a script to communicate with your Arduino and return a value for Mycodo. From there, the measurements can be used throughout Mycodo.

magichammer, 2018-04-10:
Grazie, per la risposta , provero questa via.
Magic

magichammer, 2018-04-10:
sorry i m write in italian

closed 2018-04-10 by magichammer

princeavalanche10, 2021-11-16:
> I've seen people do this. You would essentially set up communication, and the easiest way I can think to get the measurements into Mycodo would be to add a Linux Command Input that will execute a [...]

Hey Kyle, can you please explain how I can integrate the returned values from Arduino to Mycodo?

VhoMulamu, 2024-10-11:
Here's a revised version of your message for clarity:

---

Hey there! I can help you with this, but I assume you might have already found a workaround.

I managed to make it work by following these steps:

1. You need an MQTT broker server to send the data from your sensors.
2. Use a tool to test your data.
3. I assume you have both pH and temperature sensors, as the two classes in the code below demonstrate.
4. I used C++ to program the sensors.
5. I used C++ to read the sensor data from the serial port and send it to a broker, which worked well.
6. After that, you can decide what to do with the data, although I'm struggling to pull the JSON data into Mycodo.

Here's the code for the sensors:

```cpp
#include <Arduino.h>
#include <DFRobot_PH.h>
#include <ArduinoJson.h>
#include <OneWire.h>
#include <DallasTemperature.h>

#define PH_PIN A1
#define ONE_WIRE_BUS 6

float voltage, phValue;
float temperature; // Declare without initialization
DFRobot_PH ph;

// Setup a OneWire instance to communicate with any OneWire devices
OneWire oneWire(ONE_WIRE_BUS);
// Pass our oneWire reference to the Dallas Temperature sensor
DallasTemperature sensors(&oneWire);

void setup() {
    Serial.begin(115200); // Start serial communication
    ph.begin();           // Initialize pH sensor
    sensors.begin();      // Initialize temperature sensor
}

void loop() {
    static unsigned long timepoint = millis();

    // Read temperature and pH every second
    if (millis() - timepoint > 1000U) {
        timepoint = millis();

        // Request temperature from sensors
        sensors.requestTemperatures();
        temperature = sensors.getTempCByIndex(0); // Get the temperature from the first sensor

        // Check if the temperature reading is valid
        if (temperature != DEVICE_DISCONNECTED_C) {
            // Read voltage and convert to pH
            voltage = analogRead(PH_PIN) / 1024.0 * 5000; // Read the voltage
            phValue = ph.readPH(voltage, temperature); // Convert voltage to pH

            // Create JSON object
            StaticJsonDocument<200> doc;
            doc["temperature"] = temperature;
            doc["pH"] = phValue;

            // Serialize JSON to string
            char jsonBuffer[200];
            size_t n = serializeJson(doc, jsonBuffer);

            // Print JSON to Serial
            Serial.println(jsonBuffer);
[... 112 lines cut ...]
    options.c_cflag &= ~CSIZE; // Clear the current character size setting
    options.c_cflag |= CS8; // 8 data bits
    options.c_iflag &= ~(IXON | IXOFF | IXANY); // Disable software flow control
    options.c_lflag &= ~(ICANON | ECHO | ECHOE | ISIG); // Raw input
    options.c_oflag &= ~OPOST; // Raw output
    tcsetattr(fd, TCSANOW, &options); // Apply the settings

    // Read data from the serial port and publish to MQTT
    while (1) {
        int n = read(fd, buffer, sizeof(buffer) - 1); // Read up to sizeof(buffer)-1 bytes
        if (n > 0) { // Check if read was successful
            buffer[n] = '\0'; // Null-terminate the string
            publish_temp_and_ph(buffer);
        } else if (n < 0) {
            perror("Read error");
            break; // Exit loop on read error
        }
    }

    close(fd);
    return 0;
}
```

> Hey Kyle, can you please explain how I can integrate the returned values from Arduino to Mycodo?