kizniche/Mycodo #447: Error on upgrade to master branch

drgrumpy, 2018-04-12:
Version  - 5.7.2
 
I was trying to upgrade to the master branch in order to test the new sunset conditional to set pid method...

So this clearly a low priority...
I followed the wiki to set
FORCE_UPGRADE_MASTER = True
restarted frontend

result:

> [2018-04-12 12:38:17] #### Beginning Upgrade: Stage 1 of 2 ####
> [2018-04-12 12:38:52] Stopping the Mycodo daemon...Done.
> [2018-04-12 12:42:13] Downloading latest Mycodo version to /home/pi/Mycodo-master.tar.gz...Done.
> [2018-04-12 12:42:13] Creating /tmp/Mycodo-5.7.2...Done.
> [2018-04-12 12:42:16] Extracting /home/pi/Mycodo-master.tar.gz to /tmp/Mycodo-5.7.2...Failed: Error while trying to extract files from /home/pi/Mycodo-master.tar.gz to /tmp/Mycodo-5.7.2.

To check I ran:
`sudo tar xvzf Mycodo-master.tar.gz -C /tmp/Mycodo-5.7.2/`

And got:

> ....
> Mycodo-master/mycodo/mycodo_flask/static/css/bootstrap-theme.min.css.map
> tar: Skipping to next header
> 
> gzip: stdin: invalid compressed data--crc error
> 
> gzip: stdin: invalid compressed data--length error
> tar: Child returned status 1
> tar: Error is not recoverable: exiting now

kizniche, 2018-04-12:
It looks like there was a corruption of the tar.gz that was downloaded.

kizniche, 2018-04-12:
I'd recommend attempting the upgrade again.

drgrumpy, 2018-04-13:
Attempted again, exactly the same result.....

kizniche, 2018-04-13:
Strange

drgrumpy, 2018-04-13:
Yep peculiar, I then did a manual wget it downloads ok, the file extracts ok
the manually downloaded version and the Mycodo downloaded version appear to have identical file size BUT the md5sums are different, so clearly something weird....
So I have now run the upgrade again and it worked...
except the upgrade_post.sh hung at Updating ./env/src/tsl2591 clone
I killed it and all seems ok... just setting up the sunrise/sunset changing method....

closed 2018-04-19 by kizniche