kizniche/Mycodo #448: Use Output PWM as a condition

drgrumpy, 2018-04-12:
The problem:
Controlled env. cabinet.
Cooling is via fans with PWM control with value set by a PID
The fans are typical 12V PC fans, that means that when duty cycle is zero (i.e. no PWM input) they run at 100% (this is by design)
My solution is to have 3 conditionals:
turn the fan relay on when cooling duty cycle <-10%
off when >-10
off when no value

I also want to use those same fans to control (reduce/lower) humidty....
easy I thought, just set up a humidty PID and have it also control the PWM output to the fans....  
but they will never turn on... because even if I set more conditionals to turn relay on in similar way to above, they will conflict with the cooling conditionals and never run...

Effectively at present with multiple conditionals controlling the same output, they act together like as if joined by an 'OR' operator, so what I need is to have either the conditionals being able to use the actual output pwm value for the conditon (instead of the PID output duty cycle) or to be able to have multiple conditions joined by an 'AND' operator...

Make sense ?

kizniche, 2018-04-12:
The PWM outputs have the option to invert the PWM output. So, if the PID instructs the output duty cycle to 100%, it will actually be set to 0%. This takes care of your first issue without having to resort to workarounds like Conditionals.

I'm still trying to comprehend your entire setup, but I think I may be a bit confused.

drgrumpy, 2018-04-13:
Yes, it's confusing, so I'll try again....
I'm not sure I understand your comment about inverting the PWM output... but, if I do, I think it is not relevant, it will not obviate the need to control the relay...
When the PID output duty cycle is at 100% we want the PWM to be 100% (in this case 25 kHz) - this works very well. 
The fans are designed so that if they receive no PWM signal they run at full speed, so the fans also have a power relay that switches them on/off (via conditionals) when the PID duty cycle is greater than/less than 10% or does not exist - this works very well (since I figured out that the duty cycle is negative for cooling) with 3 conditionals to control the fan relay.
So I have been using the fans just for cooling without issue....

It would also be useful to use the same fans to reduce the humidty, I could use them just for humidity control in the same way...with PID control and PWM output to modulate the speed, and conditional to switch on the relay.

At least to start with the issue for me is not that the PIDs are in conflict (they may be, it may happen, I can't tell as the relay will switch on then immediately off, but this can be resolved with) but the 3 conditionals that are needed to switch on/off the relay would conflict with the 3 cooling conditionals because they can only be conditioned on the PID duty cycle (ie. if cooling PID = 0 the relay will be switched off, then if humdity PID = 20 the relay will be switched on, but the cooling PID will still be 0 so the cooling conditional will immediately switch off the relay).... if the condition could be based on the value of the PWM value the problem disappears (i.e. if it is possible to also be able to select a PWM output as the condition, hence the subject) as it then does not matter which pid is driving the pwm value
i.e.
if fan pwm output > 10% (or 2.5kHz) then switch on relay
if fan pwm output < 10% (or 2.5kHz) then switch off relay
if fan pwm == null then switch off relay

Alternatively if the condition could have two inputs with an AND operator, or two conditions can be linked with an AND operator, i.e.
if cooling duty cycle OR humidity duty cycle > x them switch on relay (this can be done now)
if cooling duty cycle AND humidity duty cycle < x then switch off relay  (not currently possible)
if cooling duty cycle AND humidity duty cycle == null then switch off relay (not currently possible)

I realise also now as you say that the two PIDs could conflict e.g. cooling may output 20%, but humidity may be say 50%, I guess we would want the value to be the max of the two. But mostly at present cooling is not providing an output so in this case cooling output = null would not conflict with humidity output = 50% ?

kizniche, 2018-04-13:
Why not just have the output as a simple On/Off output, and have both PIDs control that output? It appears that should work just fine.

kizniche, 2018-04-13:
A PWM conditional is useful though. Perhaps also a way to set up a dummy output to use with PIDs that shouldn't be directly controlling anything (if merely used for Conditionals). Though, this is an atypical use of a PID.

drgrumpy, 2018-04-13:
If it was just on/off the fans would then be always be running at full speed, giving big fluctuations and the relay would likely soon burn out ?  I think the fans have be under pwm control.

kizniche, 2018-04-13:
Didn't you describe the relays as turning on and off?

drgrumpy, 2018-04-13:
I should perhaps further explain connections to the fan(s):
GND - GND
VCC - 12V supply via mechanical relay controlled by conditional
Tacho - not connected
PWM - direct from GPIO 18 pwm output on pi zero

I also realise it should be 'easy' to avoid conflicts with the PIDs - the humdity control PID is only activated when the cooling PID output is giving 'no measurement'

kizniche, 2018-04-15:
I added a new conditional that when the duty cycle of a PWM output changes, it will check if it's above, below, or equal to a set value, then trigger the conditional actions. I'm working on getting 6.0 ready for release, which may or may not happen this weekend. When it's released, you can test. If you want to test now, you will have to install from scratch so the database schema is correct. If you do install fresh, you'll need to clone the github repository, then checkout the ```add_unique_id``` branch before running the install script.

kizniche, 2018-04-15:
Just another note, you cannot use the ```Upgrade to Master``` option because the 6.0 code isn't on the master branch, it's on the ```add_unique_id``` branch, so checking out the correct branch is crucial before running the install script.