kizniche/Mycodo #449: PID Setpoint doesn't change as expected

frodus17, 2018-04-17:
## Mycodo Issue Report:
  - Specific Mycodo Version:
 5.7.2

#### Problem Description
Cannot change PID setpoint in Dashboard as expected.  When want to change the setpoint, I expand the widget. I enter in a value, then hit "Set Setpoint". It does nothing that I can see. Then I  hit save, and then it changes the setpoint. I'm not sure what the "Set Setpoint" button does in this use case.

Also, like the other PID buttons, this would be more usable if it were accessible without having to hit the "+" to expand the widget.

kizniche, 2018-04-18:
Strange. I just tested in 6.0 and I could change it. I did notice I don't have error-handling if a settings is attempted to be changed (or held, paused, resumed) when the PID is inactive. I'll fix that. As for your issue, wait for 6.0 to be released, and if the issue persists, I'll investigate again, but it appears I must have fixed it some time in the path to 6.0 already.

frodus17, 2018-04-18:
Thanks. It was weird. It only works when "saved".
Is it possible to have it visible in the collapsed widget view?

kizniche, 2018-04-18:
I'll add an option to show it when collapsed or expanded.

closed 2018-04-19 by kizniche