kizniche/Mycodo #450: tsl2561 light sensor saturation and filling logs

drgrumpy, 2018-04-18:
Version 5.7.2

Suggestion/Enhancement

When we occasionally have a sunny day in the UK, my TSL2561 sensor gets saturated, it results in approx 12 lines in the logs, see below

>2018-04-06 15:00:26,834 - mycodo.input_7 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
> 2018-04-06 14:43:26,807 - mycodo.inputs.tsl2561 - ERROR - TSL2561Sensor raised an exception when taking a reading: Sensor is saturated
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 61, in read
>     self._lux = self.get_measurement()
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 51, in get_measurement
>     return self.tsl.lux()
>   File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py", line 312, in lux
>     return self._calculate_lux(broadband, ir)
>   File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py", line 240, in _calculate_lux
>     raise Exception('Sensor is saturated')
> Exception: Sensor is saturated

These 12 line logs are produced at every attempted measurement so on a sunny day the log apparently gets filled with this same stuff, repeated

1. I think that if the sensor is saturated there is no need for the log to be so verbose, i.e.
if exception = 'Sensor is saturated' do't bother with the traceback

2. I wonder also that if the sensor is saturated, perhaps we could automatically reduce the integration time? But I am also thinking that perhaps this should be done in the sensor library code (/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py) rather than in Mycodo, as there is already and autogain function and presumably a auto integration function could be modelled on the same lines ?

kizniche, 2018-04-18:
I can handle that exception with a smaller log line. I woldn't really know how to prevent the issue from occurring other than possibly putting a diffuser of some kind over the sensor to reduce the amount of light hitting the sensor.

drgrumpy, 2018-04-19:
Yes adding a diffuser is one option... but I was looking at the tsl2561.py library on github
and it seems there is a setting for integration time:
```
        if self.integration_time == TSL2561_INTEGRATIONTIME_402MS:
            self.delay_time = TSL2561_DELAY_INTTIME_402MS
        elif self.integration_time == TSL2561_INTEGRATIONTIME_101MS:
            self.delay_time = TSL2561_DELAY_INTTIME_101MS
        elif self.integration_time == TSL2561_INTEGRATIONTIME_13MS:
            self.delay_time = TSL2561_DELAY_INTTIME_13MS
self._begin()
```
and defaults to the longest:
```
def __init__(self, address=None, busnum=None,
                 integration_time=TSL2561_INTEGRATIONTIME_402MS,
gain=TSL2561_GAIN_1X, autogain=False, debug=False):
```
So there is perhaps potential to capture valid readings by reducing the integration time, and if I userstand the mycodo tsl2561.py code correctly it only sets the i2c address and the bus, so in mycodo we could do something like:
```
if e = 'Sensor is saturated'
    self.integration_time = TSL2561_INTEGRATIONTIME_101MS
...retry measurement
```
Forgive my lack of python knowledge but are those constants in the library available in the mycodo code ?

kizniche, 2018-04-19:
Ah. That shouldn't be too difficult, considering there's a function to change the integration time after the class is initialized:

https://github.com/sim0nx/tsl2561/blob/c76bf859c880fd1c60bb7f559169daf01021d720/tsl2561/tsl2561.py#L127-L140

My only concern is if someone didn't want this to occur and actually cause an exception if the sensor is saturated.

kizniche, 2018-04-19:
Here is some code to test, that will incrementally try to lower the integration time and attempt another measurement. It will try this twice, lowering it each time, before giving up and returning an error. Change this get_measurement() function:

https://github.com/kizniche/Mycodo/blob/b5a4d2d0bff9fc130a91f3ed2aea94aab5ab2a10/mycodo/inputs/tsl2561.py#L48-L51

To this:

```python
    def get_measurement(self):
        """ Gets the TSL2561's lux """
        self._lux = None
        saturated = False
        try:
            lux = self.tsl.lux()
            return lux
        except Exception as err:
            if err == 'Sensor is saturated':
                saturated = True

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_101MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_101MS)
            saturated = False
            try:
                lux = self.tsl.lux()
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    saturated = True

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_13MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_13MS)
            try:
                lux = self.tsl.lux()
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    self.logger.error("Could not obtain measurement: Sensor is saturated.")
```

If you want to have an indication if the code is working by notifying that the saturation event occurred, you can add logging lines, so the code is like this:

```python
    def get_measurement(self):
        """ Gets the TSL2561's lux """
        self._lux = None
        saturated = False
        try:
            lux = self.tsl.lux()
            return lux
        except Exception as err:
            if err == 'Sensor is saturated':
                self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                  "Setting integration time to 101 ms and trying again")
                saturated = True

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_101MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_101MS)
            saturated = False
            try:
                lux = self.tsl.lux()
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                      "Setting integration time to 13 ms and trying again")
                    saturated = True

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_13MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_13MS)
            try:
                lux = self.tsl.lux()
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    self.logger.error("Could not obtain measurement: Sensor is saturated.")
```

drgrumpy, 2018-04-19:
Testing underway...

drgrumpy, 2018-04-19:
I implemented the self.logger.error lines, but am getting no information (even if debug mode)
just getting:

> 2018-04-19 15:08:45,371 - mycodo.input_7 - ERROR - Error while attempting to read input: unsupported format string passed to NoneType.__format__
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/controller_input.py", line 744, in update_measure
>     measurements = self.measure_input.next()
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 39, in next
>     return dict(lux=float('{0:.2f}'.format(self._lux)))
> TypeError: unsupported format string passed to NoneType.__format__

so I cant tell if it is trying to change integration time or not...

kizniche, 2018-04-20:
Okay. There's a missing part to read() that the other input modules have. It should look like:

```python
    def read(self):
        """
        Takes a reading from the TSL2561 and updates the self._lux value

        :returns: None on success or 1 on error
        """
        try:
            self._lux = self.get_measurement()
            if self._lux is not None:
                return  # success - no errors
        except Exception as e:
            self.logger.exception(
                "{cls} raised an exception when taking a reading: "
                "{err}".format(cls=type(self).__name__, err=e))
        return 1
```

As for why it's not logging any exceptions, that's another issue. The only way it would return ```None``` is if it was saturated for all three integration times, however it should still have made log lines.

kizniche, 2018-04-20:
Here's also another get_measurement() with more logging lines:

```python
    def get_measurement(self):
        """ Gets the TSL2561's lux """
        self._lux = None
        saturated = False
        try:
            lux = self.tsl.lux()
            return lux
        except Exception as err:
            if err == 'Sensor is saturated':
                self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                  "Setting integration time to 101 ms and trying again")
                saturated = True
            else:
                self.logger.error("Error 0: {}".format(err))

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_101MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_101MS)
            saturated = False
            try:
                lux = self.tsl.lux()
                self.logger.info("Success with 101 ms")
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                      "Setting integration time to 13 ms and trying again")
                    saturated = True
                else:
                    self.logger.error("Error 1: {}".format(err))

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_13MS
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_13MS)
            try:
                lux = self.tsl.lux()
                self.logger.info("Success with 13 ms")
                return lux
            except Exception as err:
                if err == 'Sensor is saturated':
                    self.logger.error("Could not obtain measurement: Sensor is saturated.")
                else:
                    self.logger.error("Error 2: {}".format(err))

        self.logger.error("We reached the end!")
```

Theoi-Meteoroi, 2018-04-20:
I have been meaning to look into the driver used in Mycodo for this device - I bought a bunch of these for light level sensing.

There is a bunch of work I have seen in open source working on exposing the native dynamic range of this sensor. I don’t think that
exists in the current driver but I have to do a code review to see what might be improved.  Saturation level reports are common with less sophisticated drivers for this device.

I hope this helps!

Dave

drgrumpy, 2018-04-20:
Ok have added the extra logging and and the if in the read...
will report back later if/when the sun comes out...

drgrumpy, 2018-04-20:
the driver has an auto gain setting...
my understanding is that integration time is analogous to shutter speed and gain analogous to film speed/iso setting

drgrumpy, 2018-04-20:
So now we have an error which I think explains the lack of info:

> 2018-04-20 14:14:06,322 - mycodo.inputs.tsl2561 - ERROR - TSL2561Sensor raised an exception when taking a reading: 'TSL2561Sensor' object has no attribute 'logger'
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 53, in get_measurement
>     lux = self.tsl.lux()
>   File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py", line 312, in lux
>     return self._calculate_lux(broadband, ir)
>   File "/var/mycodo-root/env/lib/python3.5/site-packages/tsl2561/tsl2561.py", line 240, in _calculate_lux
>     raise Exception('Sensor is saturated')
> Exception: Sensor is saturated
> 
> During handling of the above exception, another exception occurred:
> 
> Traceback (most recent call last):
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 99, in read
>     self._lux = self.get_measurement()
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 61, in get_measurement
>     self.logger.error("Error 0: {}".format(err))
> AttributeError: 'TSL2561Sensor' object has no attribute 'logger'

my guess is that in tsl2561.py
it currently has:
`logger = logging.getLogger("mycodo.inputs.tsl2561")`
on ~line 6 before the class
but it needs moving within the class:
`self.logger = logging.getLogger("mycodo.inputs.tsl2561")`

drgrumpy, 2018-04-20:
Yep that solved it, I now get:
2018-04-20 15:23:54,565 - mycodo.inputs.tsl2561 - ERROR - Error 0: Sensor is saturated
2018-04-20 15:23:54,571 - mycodo.inputs.tsl2561 - ERROR - We reached the end!
so it would seem that even with the shortest integration time it is still too bright....

kizniche, 2018-04-20:
I forgot I was working with code from the 6.0 branch, which included the self.logger defined. Sorry about that. Now that we have logging, what you reported indicates the sensor saturation exception isn't being captured by my code. I've amended the code to attempt to fix this issue, below:

```python
    def get_measurement(self):
        """ Gets the TSL2561's lux """
        self._lux = None
        saturated = False
        try:
            lux = self.tsl.lux()
            return lux
        except Exception as err:
            if 'saturated' in err:
                self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                  "Setting integration time to 101 ms and trying again")
                saturated = True
            else:
                self.logger.error("Error 0: {}".format(err))

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_101MS
            self.logger.info("Setting 101 ms integration time")
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_101MS)
            saturated = False
            try:
                lux = self.tsl.lux()
                self.logger.info("Success with 101 ms")
                return lux
            except Exception as err:
                if 'saturated' in err:
                    self.logger.error("Could not obtain measurement: Sensor is saturated. "
                                      "Setting integration time to 13 ms and trying again")
                    saturated = True
                else:
                    self.logger.error("Error 1: {}".format(err))

        if saturated:
            from tsl2561.constants import TSL2561_INTEGRATIONTIME_13MS
            self.logger.info("Setting 13 ms integration time")
            self.tsl.set_integration_time(TSL2561_INTEGRATIONTIME_13MS)
            try:
                lux = self.tsl.lux()
                self.logger.info("Success with 13 ms")
                return lux
            except Exception as err:
                if 'saturated' in err:
                    self.logger.error("Could not obtain measurement: Sensor is saturated.")
                else:
                    self.logger.error("Error 2: {}".format(err))

        self.logger.error("We reached the end!")
```

drgrumpy, 2018-04-20:
Now I get a series of errors, but I think the key is:
>   File "/var/mycodo-root/mycodo/inputs/tsl2561.py", line 57, in get_measurement
>     if 'saturated' in err:
> TypeError: argument of type 'Exception' is not iterable

kizniche, 2018-04-20:
Try changing the 3 instances to ```in str(err)```

drgrumpy, 2018-04-20:
I was just trying to find how to force err into a string.... but struggle to see why it isn't already...

kizniche, 2018-04-20:
I've never had to catch a general exception like this. It's rather odd for the developer to raise an exception in this way.

kizniche, 2018-04-20:
I should have made a simple script to test, but I thought this would be easy. We should ha e this now.

drgrumpy, 2018-04-20:
still getting the same following the update:

>     if 'saturated' in str(err):
> TypeError: argument of type 'Exception' is not iterable

... and now the sun is going down so won't be able to test further until tomorrow....

But maybe that is the key.... raising 'Sensor is saturated' as an Exception is not a good way... better to just output the max possible or unfeasible value or like 65556 which would be easy to capture in Mycodo

drgrumpy, 2018-04-20:
I found this:
[https://stackoverflow.com/questions/4308182/getting-the-exception-value-in-python](url)

I think we may need to use:
`repr(err)`
instead of err

kizniche, 2018-04-21:
I was just able to run a test, and casting it as a string worked for me:

```python
try:
    raise Exception("TEST")
except Exception as err:
    print("TEST00: {}".format(err))
    if 'TEST' in str(err):
        print("TEST02")
```

Output:

```
pi@delta:~/Mycodo $ ./env/bin/python ./test_except.py 
TEST00: TEST
TEST02
```

drgrumpy, 2018-04-21:
Yes but my guess (without knowing anything about the intricacies of python) is that in your test above you are specifically defining the result of the exception as a string, whereas the result from the tsl2516.py is not, we are perhaps dealing with an object ?
Anyway I can confirm that using:
`if 'saturated' in repr(err): `
works and that the rest of the code is now so far doing what intended:

> 2018-04-21 13:15:14,298 - mycodo.inputs.tsl2561 - ERROR - Could not obtain measurement: Sensor is saturated. Setting integration time to 101 ms and trying again
> 2018-04-21 13:15:14,301 - mycodo.inputs.tsl2561 - INFO - Setting 101 ms integration time
> 2018-04-21 13:15:14,810 - mycodo.inputs.tsl2561 - INFO - Success with 101 ms

drgrumpy, 2018-04-21:
Later:

> 2018-04-21 13:31:14,842 - mycodo.inputs.tsl2561 - ERROR - Could not obtain measurement: Sensor is saturated. Setting integration time to 101 ms and trying again
> 2018-04-21 13:31:14,845 - mycodo.inputs.tsl2561 - INFO - Setting 13 ms integration time
> 2018-04-21 13:31:15,335 - mycodo.inputs.tsl2561 - ERROR - Could not obtain measurement: Sensor is saturated.
> 2018-04-21 13:31:15,337 - mycodo.inputs.tsl2561 - ERROR - We reached the end!
>

kizniche, 2018-04-21:
Okay, let's use repr(). So, would you like it to register 1 greater than the highest value (65536) if it reaches the end?

drgrumpy, 2018-04-22:
I have been doing a bit more research on the tsl2561 and of course because the lux value reported is a calculated from the ratio of visible + ir to ir, it is not so straightforward, 65536 is the max value that can be recorded by each separate sensor but it seems the realistic max value is something more like 30,000 according to this paper:
[(https://ac.els-cdn.com/S1474667015373596/1-s2.0-S1474667015373596-main.pdf?_tid=0308cdf4-576c-47ff-9568-09369b0ba9e1&acdnat=1524398398_9fd40163d906b8a4745cf7fa129305c6)](url)
and it seems that the max value is reduced with a shorter integration time, with perhaps little/no benefit of dropping below the 101ms, see this link (esp. the graph):
[http:box.blogspot.co.uk/2015/06/tsl2561-light-sensor-ranges.html](url)

I think also it is not going to be very accurate above a lux value of about 32k anyway...
so for my purposes censoring the value to say 32,000 would be the way to go (if we censor at ~65k, and the highest actual value reported is only ever 30k then the graphs end up being scaled to double to fit in the saturated readings)

kizniche, 2018-04-23:
Your second link did not work. 65536 seems logical to me to be able to identify a saturation event with certainty. Are you able to test an increasing light intensity to determine at what value saturation occurs? As for graph scaling, you can set your graph min/max y-axis to restrict the range.

drgrumpy, 2018-04-23:
Oops sorry missed out first part of link:
[http://ideasandbox.blogspot.co.uk/2015/06/tsl2561-light-sensor-ranges.html](url)
There is a graph down the page, you need to double click to see the whole thing....
I fully appreciate the thinking for 65536... what at first I also thought would be a good value...
But due to the scaling in the driver and the scale limits at different integration times and the unlikelihood of ir being zero when visible + ir is 65535, it will never get there...
Fort testing I can only wait til the sun comes out and see what is the max value returned before there are no values...
So far the highest value I have seen with the 402 ms integration time is about 18,000 lux
Since we added the auto selection of integration times I have seen values of up to 29,000. Also in line with suggestions from the link, I can't see any occasions where there is success with the 13ms integration time...

For the graphs scaling yes, I am thinking I want to see a flat max value....

drgrumpy, 2018-04-23:
Update: just had a value of ~40,000 lux

closed 2018-10-13 by kizniche