kizniche/Mycodo #451: Overheating ? Others experience ?

drgrumpy, 2018-04-18:
Version 5.7.2

More of a query of others experience to explain my results ?

We had our first warm day of the year today and as luck would have it I was out all day....and returned to find my pi stopped recording for the hottest part of the day
So my pi got fried, in the image below you can see that once the pi reached ~56°C (the read line) all readings stopped (blue and green), they then resumed when the pi got down to about 65°C (it got to a max of 72 C). I think I read somewhere the pi was ok up to ~70, so why did it stop reading at 56 and why did the pi carry on recording (this must mean that the pi was still running, and so was mycodo in order to record the data ? But no outputs or inputs were recorded. (The blue/green lines are temperature and stop/resume at value of around 25/24 C.

[image]

The only thing in the logs that starts at the same time as the measurements stop is:

> 2018-04-18 13:47:21,202 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
> 2018-04-18 13:47:21,209 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
> 2018-04-18 13:47:36,530 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
> 2018-04-18 13:47:36,533 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
> 2018-04-18 13:47:52,659 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
> 2018-04-18 13:47:52,663 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.
> 2018-04-18 13:48:07,915 - mycodo.lcd_1 - ERROR - Count not initialize LCD. Error: 121
> 2018-04-18 13:48:07,919 - mycodo.lcd_1 - ERROR - IOError: Unable to output to LCD.

(there's also lots of tsl2561 exceptions, but these started well before the measurements stopped.

Theoi-Meteoroi, 2018-05-02:
Which Pi model are you using?  The most recent rev (3B+) uses a heat spreader.  I've also found that the 3B needs a heat sink when you start rocking all 4 cores consistently, so usage patterns matter.  It could have been just the LCD component, most are not designed for use in direct sunlight.  Are you using any buffering ( mux? ) for the i2c connection?   Most commercial temp range components are specified to operate over 0 to 70 degrees C. Mil-spec and industrial are -40 to 85 degrees C.  The Pi is commercial and would need to be sorted for assemblies that might work at higher temps, but I'd expect the onboard power supply to be the first to let out the magic smoke.  Unless you can somehow moderate the temperature range to stay within the spec limits, a dead-man timer becomes a requirement if you don't want to fiddle with the equipment constantly.

drgrumpy, 2018-05-02:
Thanks for your comments @Theoi-Meteoroi. I am using a Pi zero wireless, with a heatsink.....
Since I added the shading to the box, there has not been a repeat of the issue (and there has been more sun and greater ambient temps), so everything points to the lcd which was in direct sun, but is no longer with the shading.

closed 2018-05-02 by drgrumpy