kizniche/Mycodo #452: Upgrade to 6.0.0 will fail, must perform a fresh install.

kizniche, 2018-04-21:
After upgrading to 5.7.3, the upgrade page will say there is an upgrade available and present a button to upgrade. This will not work. Instead, you must install Mycodo again, following the instructions in the README. No settings may be imported from a pre-6.0 version of Mycodo.

kizniche, 2018-04-21:
Here's the simplest way:

```sudo service mycodo stop```
```sudo service mycodoflask stop```
```sudo service nginx stop```
```mv ~/Mycodo ~/Mycodo-backup```

Then follow the install instructions to install again. This stops all the services and moves the current installation into a backup location, freeing up the new install directory name.

frodus17, 2018-04-21:
Removal and install went smoothly! Thanks. 

When I set the default page to dashboard, without having done anything to dashboard, there's an error. 

When I tried to add a DS18B20, I get this, and cannot get back to add input (even after reboot): 

Error 500: Internal Server Error
Something bad happened but it's probably not your fault. Letting the developers know about these issues is crucial to supporting Mycodo. Please submit a new issue on GitHub with the following error traceback (copy the entire traceback):

Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 1199, in page_data
    w1thermsensor_sensors.append(each_input.unique_id)
AttributeError: 'W1ThermSensor' object has no attribute 'unique_id'

kizniche, 2018-04-21:
I just fixed the landing page issue (commit coming soon). If you didn't have it installed already, were you prompted to install the DS18B20 library when you attempted to install the DS18B20 input?

kizniche, 2018-04-21:
I found the issue. Commit and release incoming.

frodus17, 2018-04-21:
I removed old Mycodo, and re-installed new. So the libraries should have been there.

I did this:
sudo service mycodo stop
sudo service mycodoflask stop
sudo service nginx stop
mv ~/Mycodo ~/Mycodo-backup

Then this:
sudo apt-get install -y jq
cd ~
curl -s https://api.github.com/repos/kizniche/Mycodo/releases/latest | \
jq -r '.tarball_url' | wget -i - -O mycodo-latest.tar.gz
mkdir Mycodo
tar xzf mycodo-latest.tar.gz -C Mycodo --strip-components=1
rm -f mycodo-latest.tar.gz
cd Mycodo/install
sudo /bin/bash ./setup.sh

kizniche, 2018-04-21:
The install creates a new virtualenv, so if you selected Full install or Custom and selected the w1thermsensor library to be installed, then you had it and it wouldn't prompt you to install it. If you selected Minimal install, then it should prompt you to install the w1thermsensor module.

kizniche, 2018-04-21:
Just created v6.0.1 release. Let me know if it fixes your issue.

kizniche, 2018-04-21:
There's an upgrade issue. stand by.

frodus17, 2018-04-21:
Failed again, rebooting to retry.

kizniche, 2018-04-21:
I just fixed the upgrade issue. A fresh install will fix it (v6.0.3).

frodus17, 2018-04-21:
Sorry for vagueness. It failed same place as 6.0.1. Rebooted, and Mycodo server is unreachable. Pi is at the IP address I programmed, can VNC into it. Mycodo daemon is running, but mycodoflask had errors.

kizniche, 2018-04-21:
I've fixed all the issues on my end. Landing page change works, DS18B20 should work. Upgrade system should work.

I don't know where it failed for you w 6.0.1, you never mentioned.

frodus17, 2018-04-21:
I thought I screenshotted it, but didn't. Sorry. It failed right after running the upgrade. I forgot the wording. it asked me to retry after reboot, but then I couldn't access Mycodo server anymore. 

Fresh install now. Will screenshot/copy the messages this time.

frodus17, 2018-04-21:
Install worked. Still having issues with adding the DS18B20.

[image]

kizniche, 2018-04-21:
Okay. I figured out the issue and released v6.0.4. I'll be away for a few hours, but that should do it. Let me know if there are any issues and I'll check it out when I get back.

frodus17, 2018-04-21:
Damn you're fast. 

DS18B20 now works and reads. Great work Kyle.

closed 2018-05-10 by kizniche