kizniche/Mycodo #453: pigpiod issue?

riverfreeloader, 2018-04-21:
Trying to install version 6 on a Raspberry Pi 3 B+. When performing a full install, the installation freezes when it gets to installing the pigpiod it gets as far as the lines below then freezes.
gcc -O3 -Wall -pthread -fpic -c -o pigpio.o pigpio.c
gcc -O3 -Wall -pthread -fpic -c -o command.o command.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if.o pigpiod_if.c
gcc -O3 -Wall -pthread -fpic -c -o pigpiod_if2.o pigpiod_if2.c

Any thoughts on what is happening?

Thanks

kizniche, 2018-04-21:
Nothing comes to mind why this would be happening. Could you try a new SD card with a fresh install of Raspbian before attempting the install?

kizniche, 2018-04-21:
There have been at least one successful 3B+ Mycodo installs thus far, so I can tell you it is possible. What the exact issue with your installation is, I don't know, but since we don't have an error to work with, I'd start with the easiest things to perform first, such as the SD swap recommendation.

riverfreeloader, 2018-04-30:
All good. Disconnected the DHT22 attached to GPIO4 and pigpiod installed.

closed 2018-04-30 by riverfreeloader