kizniche/Mycodo #455: Unable to upgrade

cbaymac, 2018-04-24:
## Mycodo Issue Report:
  - 5.5.15
 
#### Problem Description
Please list:

  - attempt upgrade
 
### Errors
  - Error (Full Traceback):

Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_admin.py", line 283, in admin_upgrade
    is_internet=is_internet)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 134, in render_template
    context, ctx.app)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/templating.py", line 116, in _render
    rv = template.render(context)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 1008, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/environment.py", line 780, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/jinja2/_compat.py", line 37, in reraise
    raise value.with_traceback(tb)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 2, in top-level template code
    {% set active_page = "upgrade" %}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/layout.html", line 210, in top-level template code
    {%- block body %}{% endblock -%}
  File "/home/pi/Mycodo/mycodo/mycodo_flask/templates/admin/upgrade.html", line 109, in block "body"
    You are {{ releases_behind }} release{% if releases_behind > 1 %}s{% endif %} behind the latest version.
TypeError: unorderable types: NoneType() > int()
 
### Additional Notes
I,did remove .upgrad file in Mycodo folder.

cbaymac, 2018-04-24:
Upgraded to 6.0.6 and installed a MH Z19b. It uses GPIO 14 TX and GPIO 14
RX. I do not see these values to input into the data sheet. Followed the
instructions for UART but they did not seem to apply to Mycodo?

Gordon McPheter
757-790-9042

kizniche, 2018-04-24:
Without more information about your setup, it is impossible to determine the issue. Review the documentation to determine if you're using the correct device: https://www.raspberrypi.org/documentation/configuration/uart.md