kizniche/Mycodo #456: Trying to get in contact

paulthomson-me, 2018-04-24:
Hello Kyle,

I can't seem to contact you via your website contact form or comments section so trying to reach you on here hope you don't mind...

Firstly, congratulations on your Mycodo project and many thanks for making it open-sourced as I'm sure many hundreds of people, maybe even thousands have benefitted from your work.

I'm currently in the process of starting up a social enterprise over in the United Kingdom where I'll be growing gourmet mushrooms and via partnerships with other community farms, will be teaching others to do so also.

I was wondering whether you'd be willing to build me a Mycodo control system and post it over to the UK? I'd be happy to pay you to do so, just I have no expertise in this field, or the time to learn currently.

If you are unable to do so, I'll look for someone local who might be willing to take on the project. Regardless of whether you can or can't, if we decide to adopt your system, I'd like to contribute a percentage of our business profits to help support you in the development of the Mycodo project, or even just to show our gratitude for the work you've done to date, which I'm sure is many, many hours.

Keep up the great work and I hope to hear from you soon,

Kind Regards,
Paul

paul [at] paulthomson.me

frodus17, 2018-04-24:
Let me know if you don't want to Kyle. I might be interested in helping.

samsixtysix, 2018-04-27:
I would also be able interested in helping, depending a bit on where you are based.

paulthomson-me, 2018-04-28:
Hello all,

Many thanks for your responses.

Just to clarify, I'm based in Bristol, United Kingdom.

I'd be looking for someone to build a mycodo unit and then post it over the pond to me.

Perhaps a better avenue to communicate on would be through email? Travis and @samsixtysix perhaps you could email me and we can talk more?

Kyle, could you also please drop me a quick email with your PayPal address so I have it on record for donations/support later down the line. We probably wont be commercially set for another 2-3 months, but after which I'd like to support your project going forward.

Thanks to you all and have a nice weekend,
Paul

paulthomson-me, 2018-04-28:
my email address: paul [at] paulthomson.me

Theoi-Meteoroi, 2018-04-29:
I've been thinking about a HAT design that would implement some of the basic sensor infrastructure and have better electrical characteristics of soldered components as well as protection for the Pi interface.   I have some ideas ( 2x i2c mux with jumper selectable bus voltages, MCP3008 a/d,  grove-type header options for some connections, etc) that would make putting a fairly complex system physically together easier.   

I've built several prototypes with various configurations on this board:  [https://www.digikey.com/product-detail/en/twin-industries/3.14-2/438-1148-ND/5214232](url)

Its a high quality board for a pretty good price. You just have to choose the soldered header style that you need ( long pins for stacking or short pins to be the last board ).

If there is enough interest, we might be able to crowd fund a manufactured interface board that makes the project more accessible to less electronically adept users and provide lower-noise performance for devices like the MCP3008 that really need a proper ground plane and filtering.

closed 2018-05-10 by kizniche