kizniche/Mycodo #457: Add ORP to config.py

lovelytwo, 2018-04-25:
Hey, I hope this is an easy request. Could you add the measurement ORP (Oxidation-Reduction Potential) to config.py's "MEASUREMENT_UNITS". Eg the following.

`'orp': {
'name': 'orp',
'meas': 'orp', 
'unit': 'mV'},`

Thanks.

kizniche, 2018-04-26:
What would be the benefit of adding measurements that no supported sensor measure?

kizniche, 2018-04-27:
This is indeed a bug. I managed to fix the main issue (committed moments ago), that should resolve all the downstream issues like graphs/gauges/etc.

Thanks for the bug report. The fix will be available in the next release. When it's released in a few days you'll be able to upgrade your system (Config -> Upgrade) and use it.

kizniche, 2018-04-27:
I'm glad you found a solution, by editing config.py. I'm sure that took a bit of experimenting. The config.py shouldn't need to be edited unless a new sensor is being added. The expected behavior is to use the user set measurement/unit for dashboard elements.

closed 2018-04-27 by kizniche