kizniche/Mycodo #459: AM2315 not working anymore after installing 6.0.7

wflost, 2018-04-27:
## Mycodo Issue Report:
Mycodo Version: 6.0.7
 
#### Problem Description
AM2315 not working anymore after installing 6.0.7
The sensor was working correctly under 5.7.2 and 5.7.3
 
### Errors
Below is the error log from the daemon log

2018-04-27 01:44:29,453 - mycodo.inputs.am2315_1 - ERROR - AM2315Sensor raised an exception when taking a reading: 'NoneType' object has no attribute 'is_on'
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/inputs/am2315.py", line 182, in read
    self._temperature) = self.get_measurement()
  File "/var/mycodo-root/mycodo/inputs/am2315.py", line 123, in get_measurement
    Output, unique_id=self.power_output_id).is_on()
AttributeError: 'NoneType' object has no attribute 'is_on'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. activate the AM2315 sensor

### Additional Notes
The sensor is detected under the system information on i2c bus 5

kizniche, 2018-04-27:
I just pushed a fix and release v6.0.8. Thanks for the bug report.

closed 2018-04-27 by kizniche

wflost, 2018-04-28:
Thanks tested and working as before

Your work is amazing, keep it up!!

Regards
Wayne