kizniche/Mycodo #460: dashboard error

magichammer, 2018-04-27:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.08
 
#### Problem Description
Please list: 

  - when i modify the dashboard and add in a graph two value: temp setpoint, temp from ds18b20
  
 
### Errors
  - List any errors you encountered.
  - Traceback (most recent call last):
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1982, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1614, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1517, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/_compat.py", line 33, in reraise
    raise value
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1612, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask/app.py", line 1598, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/Mycodo/env/lib/python3.5/site-packages/flask_login/utils.py", line 261, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/routes_page.py", line 400, in page_dashboard
    y_axes = utils_dashboard.graph_y_axes(dict_measurements)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 469, in graph_y_axes
    input_dev)
  File "/home/pi/Mycodo/mycodo/mycodo_flask/utils/utils_dashboard.py", line 538, in check_func
    elif each_device.cmd_measurement and each_device.cmd_measurement != '':
AttributeError: 'PID' object has no attribute 'cmd_measurement'

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1.  create data for ds18b20
 2.  create function Pid
 3.  add to graph : setpoint pid, ds18b20 temp

kizniche, 2018-04-27:
I should have tested the latest release for this. I'll push a fix and make a new release tonight.

magichammer, 2018-04-27:
Thank you for your commitment, for this wonderful project

kizniche, 2018-04-28:
Just released v6.0.9 with the fix for this. Thanks for the bug report.

closed 2018-04-28 by kizniche