kizniche/Mycodo #462: Dashboard Graph Unit Type from Math data Incorrect

mrplow, 2018-04-28:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.0.9
 
#### Problem Description
Please list: 

  - this is a minor display issue
  - I'm converting pressure input in PA to kPa via a simple Math input equation
  - Display on the dashboard is showing the unit type as PA even though it is showing the proper value in kPa
 
### Errors
  - none, display issue

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Create math data input, set unit type to kPa, equation is x/100, use pressure input from your sensor
 2. Create Graph in dashboard with math input

kizniche, 2018-04-28:
You're probably using the measurement "pressure", which will use the default unit "Pa". Use a unique measurement name, such as "kilopascals", to have it display your custom unit.

mrplow, 2018-04-28:
You nailed it. I'll go ahead and close this issue.

I had to create a new graph but I kind of expected that.

Thanks as always.

closed 2018-04-28 by mrplow

kizniche, 2018-04-28:
Welcome. If you change the measurement in the future, you should be able to just save the graph/gauge/etc. settings again and it will pick up the new measurement name, so you don't have to delete it.