kizniche/Mycodo #467: support for the ccs811 I2c sensor

wflost, 2018-05-08:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.0
 
#### Problem Description
Please list: support for the ccs811 I2c sensor
  
### Errors
  New feature

### Steps to Reproduce the issue:
  New feature

### Additional Notes
Sensor is a bit cheaper than a K30

kizniche, 2018-05-08:
Sure. Do you have this sensor to test?

wflost, 2018-05-08:
Hi Kyle,

No I will order one to stated development

Thanks
Wayne

wflost, 2018-05-10:
Hi Kyle,

Ordered an additional multi sensor which includes the following CCS811 HDC1080 BMP280 sensors one a single die.

kizniche, 2018-05-10:
Nice. I just added an untested HDC1000/HDC1080 input module. Let me know when you test and I'll try to help diagnose any issues that arise.

wflost, 2018-05-13:
> CCS811 HDC1080 BMP280 sensors one a single die
> Wflost please provide a Name or Product-link.

Hi RynFlush

I only have the ebay link

https://www.ebay.com.au/itm/CCS811-HDC1080-BMP280-Carbon-Dioxide-CO2-Temperature-Humidity-Air-Quality-Sensor/192522549844?ssPageName=STRK%3AMEBIDX%3AIT&_trksid=p2057872.m2749.l2649

kizniche, 2018-05-19:
I must not have released v6.1.1 when I meant to a few weeks ago. I think I wanted to test more and it slipped away from me. I'm making a release now with the new inputs.

wflost, 2018-05-19:
Hey Kyle

No worries,

the HDC1080 sensor works as expected

I am getting the following error on the ccs811 sensor

kizniche, 2018-05-19:
There wasn't an error in your last message. Could you try to send it again?

wflost, 2018-05-19:
I am getting the following error on the ccs811 sensor 

Error: Could not activate Input controller with ID 2936ab12-599a-4736-bd99-c47246034b38: Device ID returned is not correct! Please check your wiring. 
Under the system information it picks up correctly on bus 1 as 5a 
Neither of the sensors are proving any data

wflost, 2018-05-19:
Deactivating and reactivating the HDC1080 provides the following error

Error: Could not activate Input controller with ID 1c9cc20c-5438-4940-94c0-6d981f2ca7dd: 'HDC1000' is not a valid device type.

kizniche, 2018-05-19:
I fixed the HDC1000 initialization issue, but I'm not sure why the [Adafruit_CCS811](https://github.com/adafruit/Adafruit_CCS811_python/) driver is returning the error "Device ID returned is not correct! Please check your wiring."

kizniche, 2018-05-19:
Were you able to verify the CCS811 sensor is working with any other code?

wflost, 2018-05-19:
I will have to do some research on accessing it and what the error codes are

wflost, 2018-05-19:
Hi Kyle,

I changed the i2c speed and that removed the hardware id error, I need to check what the start up error is that I am getting now, it is still reporting a wiring error

dtparam=i2c_baudrate=10000

wflost, 2018-05-19:
Kyle,

Looks like the sensor is the issue I am only getting an error 2 trying to start it with the sparksfun Arduino base Reading

I have already contacted the one supplier because the sensor was not picking up on the RPI

wflost, 2018-05-27:
Kyle,

I have a SparksFun sensor working on Arduino. When I had it working with the RPI and Mycodo, I was getting an incorrect temperature and zero ppm values. I tried multiple times restarting Arduino and the first three reading are always zero ppm.

Are we able to change the reading for the sensor to only read after X seconds or to keep the sensor alive the whole time and only reading it every X seconds.

The three field shows the results from the BasicReadings.ino, downloaded from the following url 
https://github.com/sparkfun/SparkFun_CCS811_Arduino_Library/archive/master.zip

The third field shows the time the sensor has been active, are able to use this

[image]

I am not sure how the other sensors have been implemented, this seems different to the other sensors I have worked with.

The sensor is still connected to the  Arduino to do the 48 hour burn in

Thanks

kizniche, 2018-06-05:
Does the example code work on the Pi?

https://github.com/adafruit/Adafruit_CCS811_python/blob/master/examples/CCS811_example.py

wflost, 2018-06-06:
> Does the example code work on the Pi?
>
> https://github.com/adafruit/Adafruit_CCS811_python/blob/
> master/examples/CCS811_example.py

Hi Kyle,

The example works with the Adafruit library, I am still getting 0 ppm in
Mycodo

Thanks
Wayne

kizniche, 2018-06-06:
That's at least a good sign, because the Mycodo input module uses the adafruit library. I'll put together some test code for you to try soon.

wflost, 2018-06-07:
Thanks Kyle

kizniche, 2018-06-07:
I just pushed a potential fix. Could you make the changes in the commit, above, restart the backend, and see if that fixes the measurement issues?

kizniche, 2018-06-07:
Whoops, now the changes have been committed.

wflost, 2018-06-08:
Great, we are getting reading,

I had one spike to 65k which is making the other readings difficult,

will track the reading 1100 ppm seems a bit high

Thanks
Wayne

kizniche, 2018-06-11:
I don't see an attachment. Email replies aren't able to attach documents to your reply. You must use the web interface for that.

kizniche, 2018-06-11:
Human exhaled breath is around 300,000 ppmv, so if you've been breathing around the sensor, it may register a high value. I'm not sure what technology this sensor uses, but NDIR CO2 sensors produce false positive measurements from high humidity. I'd check if there are any variables that may confound the measurement with this particular sensor.

wflost, 2018-06-11:
Hi Kyle 

Attached the stability test

[image]

wflost, 2018-06-12:
> It appears there's an issue. You shouldn't be maxing out at 8000. Basal
> CO2 should be around 400. What event caused it to jump after the normal
> measurements at the beginning of the graph?

Hi Kyle,

I made a hole in a 2 liter juice container and inserted the sensor in the
container and sealed the hole the wires come in. I assumed there would be
something in the container that raised the count to 7992, when I opened and
removed the sensor from the container it dropped back to around 400.

Thanks
Wayne

wflost, 2018-06-20:
Hi Kyle,

Do you want to close this issue?

Thanks
Wayne

closed 2018-06-20 by wflost

kizniche, 2018-06-20:
Thanks. I meant to close it after the most recent release.

stevenvdr, 2018-10-23:
Is there a guide somewhere to get this one running properly? Even when lowering the clock speed and using the Adafruit library (both the old and the circuit python lib) I'm unable to get a reading.

kizniche, 2018-10-23:
Also, there are other issues where this sensor is discussed. Do a search to find them, and you may come across pertinent information.

wflost, 2018-10-24:
Hi Kyle and  stevenvdr 

I found the Adafruit and Sparksfun senors to work easily with RPI and arduino with the standard libraries. The issues I encountered was with generic CSS811 sensors, where I could see them on I2C but there were no data being returned or they did not even show on I2C. Which was due to faulty sensors. 

Swapping sensors to test if the sensor hardware is working correctly. I would recommend getting an Adafruit and SparksFun sensor to ensure your config is working correctly and then try using other generic CSS811 sensors.