kizniche/Mycodo #468: Access Mycodo via Port Number

cbaymac, 2018-05-09:
## Mycodo Issue Report:
  - 6.10
#### Problem Description
Please list: External access via internet to mycodo

  - I currently have two control boxes running mycodo to control humidity, temp, CO2 in my colonization room and fruiting chamber. I can access these boxes inside my network with their IP address, however when accessing remotely I can only access via https and my IP address without the port number added. I would like to access remotely with external IP:8888 and external IP:8889 for example. I think I need to change something in the NGINX config file, but not sure how to script or which file to edit.
Thank you,

kizniche, 2018-05-09:
You should be able to perform port forwarding with your router. You will need to set up rules as the following, replacing the internal IP addresses with the IP addresses of your systems:
Rule 1: external port 8888 -> 192.168.0.1:443
Rule 2: external port 8889 -> 192.168.0.2:443

cbaymac, 2018-05-09:
Kyle,
  Thank you. That works great and took my router knowledge to an entire new
level. Keep up the good work. You have an excellent product.

Gordon McPheter
757-790-9042

kizniche, 2018-05-10:
Good to hear! Thanks for the support.

closed 2018-05-10 by kizniche