kizniche/Mycodo #469: Command (PWM) outputs support

lovelytwo, 2018-05-13:
Version: 6.1.0
 
Problem Description
  - PID controller don't seem to support command_pwm outputs. 
https://github.com/kizniche/Mycodo/blob/2a2baf917e4fb4845a804df8be188ea55a00dc6a/mycodo/controller_pid.py#L576

Suspect there is also an issues with the frontend, continues refereeing to "duration" instead of duty cycle.

  - Conditional measurement with "Output (Duty Cycle)" doesn't list command_pwm outputs.

Thanks.

closed 2018-05-19 by kizniche

lovelytwo, 2018-05-20:
Hope you had a good holiday, and that your move isn't too painful.

The UI doesn't detect it's a "pwm" output probably due to the following:
https://github.com/kizniche/Mycodo/blob/968050eafa73be17bb2c0a8b939bbfec7c4760a5/mycodo/mycodo_flask/utils/utils_pid.py#L75

kizniche, 2018-05-20:
You can see all the edits I made in the referenced commits, above. They will be included in the next release.

kizniche, 2018-05-20:
I'll check out utils_pid.py

lovelytwo, 2018-05-20:
And maybe mycodo_flask/templates/pages/function_options/pid.html is worth a look.

lovelytwo, 2018-05-20:
I've made some screenshots, that illustrate the issues I'm seeing, as i'm not good at explaning in text.

1. Shouldn't this be min and max duty cycle?
[image]

2. duty cycle not duration.
[image]

3. I think this widget is unhappy, with the output being null, because it's looking for duration not duty cycle.
[image]

kizniche, 2018-05-20:
It may be from the issues displaying the proper PID options that are causing the PID widget to fail to show a value (e.g. if the min and max duty cycle were set to 0). I just fixed several references to command pwm outputs.