kizniche/Mycodo #471: 6.1.0: Send Test Mail broken?

fry42, 2018-05-15:
So my installation has been acting up lately, so I tried to set up some alerts.

After configuring all the smtp parameters I tried to send a test mail:
Entered my eMail Address, clicked "Send Test eMail" and got the following error:
Error: Modify Alert Settings: name 'message' is not defined

Either my installation has some serious issues, or mycodo is quite buggy right now...

kizniche, 2018-05-19:
I haven't been able to replicate this. I just tested sending a test email with gmail, and it worked.

CriticalFlaw, 2018-05-19:
I have the same issue, so perhaps myself and @fry42 jsut have something configured incorrectly?

[image]

kizniche, 2018-05-19:
Can you try applying the change in the commit I just pushed, then restart the frontend (Config->Restart Frontend) and see if there's a different error message?

CriticalFlaw, 2018-05-19:
Updated and restarted, getting the same error message.

kizniche, 2018-05-19:
If it's the ``exact`` same message, then you're not running the latest code.

kizniche, 2018-05-19:
Just to clarify, the commit referenced in this thread with the fix for the exception-handling is not included in the 6.1.1 release. You will have to manually edit the file, save, then restart the frontend for the changes to take effect.

closed 2018-06-20 by kizniche