kizniche/Mycodo #474: Recommended Temp Probe

cbaymac, 2018-05-17:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.0
 
#### Problem Description
Please list: What temperature probe type would you recommend for a 55 gallon drum sterilizer?  I would like to PID control the temp at 200F. Simpler is better and close is good enough. I plan to mount the probe through the side of the barrel directly into the heated water at the bottom of the barrel. I may actually just use a timer function once I get a feel of the system.  I plan to run at 200 F for 22 hours Thank you.

kizniche, 2018-05-19:
Atlas Scientific makes some nice PT-1000 probes that fit a number of different length thermowells: https://www.atlas-scientific.com/temperature.html

Otherwise, you could source your own thermowells and use any of the thermocouples supported by the MAX-Type chips that Mycodo supports (https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#max31850k)

cbaymac, 2018-08-30:
Kyle,
 I recently moved several Wordpress websites from a hosting provider to my
local raspberry 3+ running NGINX and SSL.  It is working great with DNS
support from the hosting provider as I have several domain names. So far so
good. How can I integrate Mycodo?  Now that my locally hoisted websites
also use port 443 I now can no longer access mycodo from outside my
network.
1. Can I change the 443 port to 442 for example? Or
2 Associate a domain name to mycodo? I plan on running two RPIs with mycodo
so please consider that.
Thank you,

kizniche, 2018-08-30:
The easiest solution I've found is to set up a VPN server on your home network (I use one of my Pis for this with http://www.pivpn.io/), which will allow you to connect from a VPN client outside your network and then access all of your internal IP addresses to pull up your Mycodo web interfaces.

Alternatively, you can attempt to modify [Line 3 of ~/Mycodo/intstall/mycodoflask_nginx.conf](https://github.com/kizniche/Mycodo/blob/e39a4e264d12040cc339f22929ae5f95a7890d9f/install/mycodoflask_nginx.conf#L3) and change the port 443 to another port. You could also in your router port forward port 443 of any of the IP addresses from any of your Pis with Mycodo on them to another external port, such as 8800, then connect to your domain.com:8800, which will route all traffic to/from your Pi's port 443 to the external 8800. Though, there may be other network configurations you will need to tweak to get this to work.

cbaymac, 2018-09-12:
Kyle. Great.  Thank you.

cbaymac, 2018-09-20:
Kyle,
   I just tried th PiVPN. Very cool. Thank you. That was indeed the
solution I was looking for. I’m still enjoying controlling my grow room
with your solution Mycodo. I’ve recent completed a week-long workshop near
Eugene, OR with Fungi for the People and just last weekend I receive my
certification from the state of Michigan for wild mushroom identification.
Hopefully this cooler weather will mean an increase in production as this
summer’s production was a bust due to the heat in my garage (grow room).
Thanks,
Gordon

kizniche, 2018-09-21:
That's great. Glad to hear everything is accessible now. Yeah, I've had issues regulating temperatures where I can't cool... It's always easier in the winter when a heater can be used.

cbaymac, 2018-09-22:
This is my first build. It has six individually controlled outlets, a USB
port for a remote camera, and an Ethernet outlet leading to a terminal
block. I have DHT-22 for temperature and humidity readings and a MH-Z19 for
CO2 readings.

Outlets control a mister for humidity, fan to control CO2 and a timer to
control lighting. During winter months I add a heater to my grow room, but
us the built in thermostat to control the temperature.

This project was inspired by your video build. Thanks,
Gordon

closed 2018-10-13 by kizniche