kizniche/Mycodo #476: Feat req: kPa output float

mrplow, 2018-05-19:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.0.11
 
#### Problem Description
Please list: Pressure reading in kPa is output as int, rounding out the decimal values in live and graph displays

### Additional Notes
This is just a feature request as using Pa instead works fine

kizniche, 2018-05-20:
All measurements should be float. If your pressure is still being recorded as int, then you'll need to delete your influxdb in order to fix the type from int to float. This is a remnant of pre-6.0 use of int pressure measurements.

mrplow, 2018-05-20:
Just noticed that the dashboard graph actually shows 97.000 but must have had it's decimal values rounded off. The live display shows 97 kPa.

Grabbing the values from the terminal with a random python script shows 97.451 with the decimal values

mrplow, 2018-05-20:
@kizniche what would be the best way to delete the influxdb? Reimaging the SD card and starting from scratch?

Sorry I didn't see your response until just now.

kizniche, 2018-05-20:
Check out https://github.com/kizniche/Mycodo/wiki/Deleting-and-Recreating-the-InfluxDB

mrplow, 2018-05-20:
Thanks for that info but it didn't solve my issue. Although the measurements haven't shown the pressure to deviate from 97, while checking manually from the terminal shows decimal values, it does seem odd to be perpetually stuck at 97. I'll check the hardware tomorrow.

[image]

kizniche, 2018-05-20:
Yes, if you have no measurement change, there is most likely a hardware issue.

mrplow, 2018-05-21:
Yeah I'm pretty sure the pressure sensor bit of my bme280 is toast. Wiring is fine but the measurement hasn't deviated beyond 97.x talking manual measurements in python, and showing strait 97.00 all day in mycodo.

That little chip sure didn't last long.

mrplow, 2018-05-24:
Upon further testing I believe this is an actual issue.

I setup my other pi fresh with mycodo 6.1.1 and setup my bme280 as Pa, which shows a nice graph slope.

I then deactivated that data input and created a new one with the pressure set to kPa. It is indeed cutting off the decimal values as shown in the attached image, jumping from 96 kPa to 97 kPa.
[image]

kizniche, 2018-05-24:
Okay. Probably an issue with the conversion function. I'll check it out later. Thanks for testing.

mrplow, 2018-05-25:
Double checked and I did make the right edit

`pi@raspberrypi:~ $ grep pascals_to_kilopascals  Mycodo/mycodo/config.py
    'pascals_to_kilopascals': 'x/1000.0',`

mrplow, 2018-05-25:
No problem.

Thanks

mrplow, 2018-05-29:
Using the math equation still works fine.

Also I found an equation to convert my pressure reading (in Pa) to sea level pressure (in kPa):
x*.001*(1-(.0065*315)/(12+.0065*315+273.15))**-5.257
315 is my elavation (in m) and 12 is my current temperature (in C, it also doesn't change the outcome tremendously)
Ref: https://keisan.casio.com/exec/system/1224575267

It would be cool to be able to calculate that all dynamically, and not have to fuzz the temperature, since the BME280 reads all the required measurements but I don't see a way to calculate with multiple inputs.

All this is sperate and not really relevant to this issue. But I figured that since the math calculations are working properly without dropping the decimal values I figured I'd share my current solution.

kizniche, 2018-06-05:
It's strange that my kPa seems to work fine.
[image]

mrplow, 2018-06-05:
Yeah it is strange. Are you doing strait kPa output or is there a math output involved?

I wouldn't worry about it as I have my work around.

kizniche, 2018-06-05:
I'm using the integrated conversion option for the input. This is my development system, so it's possible there have been some experimental stuff going on that fixed the issue. I need to test on a system that has had a fresh install, which I am currently in the middle of setting up to test the I2C mode of my K30 CO2 sensor. I'll need to then hook up a pressure sensor to perform the test. I need to get the CO2 testing out of the way before tackling the pressure issue. At least the Math workaround is a decent fix until I can figure out what's going on.

kizniche, 2018-06-06:
I just found the issue, and it's a forced-integer type in the BME280 input module. Pushing a fix and it'll be available in the next release.

mrplow, 2018-06-08:
I just grabbed the updated bme280.py and I can confirm 0e0ff8f fixes my issue.

kizniche, 2018-06-08:
Great! Thanks for confirming the fix.

closed 2018-06-20 by kizniche