kizniche/Mycodo #477: Edge detection issue mycodo 6.1.1

reymart081678, 2018-05-20:
## Mycodo Issue Report:
  - Specific Mycodo Version:6.1.1
 
#### Problem Description
Please list:
 Adding switch for edge detection
[image]

 
### Errors
2018-05-20 00:23:54,403 - mycodo.daemon - INFO - Mycodo daemon v6.1.1 started in 22.461 seconds
2018-05-20 00:23:54,451 - mycodo.daemon - INFO - 37.28 MB RAM in use
2018-05-20 08:22:52,799 - mycodo.input_15110ba3 - INFO - Activated in 3413.6 ms
2018-05-20 08:23:18,089 - mycodo.input_a1d3dc99 - INFO - Activated in 3549.5 ms
2018-05-20 08:23:19,188 - mycodo.input_a1d3dc99 - ERROR - Error: Failed to add edge detection
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 377, in run
    bouncetime=self.switch_bouncetime)
RuntimeError: Failed to add edge detection
2018-05-20 08:31:59,169 - mycodo.input_a1d3dc99 - INFO - Activated in 1831.2 ms
2018-05-20 08:32:00,258 - mycodo.input_a1d3dc99 - ERROR - Error: Failed to add edge detection
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 377, in run
    bouncetime=self.switch_bouncetime)
RuntimeError: Failed to add edge detection
2018-05-20 08:35:34,458 - mycodo.input_a1d3dc99 - INFO - Activated in 5350.6 ms

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-05-20:
Is anything else using pin 23? Have you tried other pins? I just tested and could replicate the issue only if I used an invalid pin.

reymart081678, 2018-05-20:
Hi, I tried to change the pin to 4 and still the same but when i use the manual test script and it was able to detect it.

[image]

2018-05-20 07:07:17,635 - mycodo.input_5efdfb81 - ERROR - StopIteration raised. Possibly could not read input. Ensure it's connected properly and detected.
2018-05-20 07:07:17,638 - mycodo.input_326b66b8 - ERROR - Error: Failed to add edge detection
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 374, in run
    bouncetime=self.switch_bouncetime)
RuntimeError: Failed to add edge detection

pi@raspberrypi:~/Mycodo/mycodo/tests/manual_tests $ ./test_gpio_read_state.py
Off: 0, On: 1
Off: 0, On: 2
Off: 0, On: 3
Off: 0, On: 4
Off: 0, On: 5
Off: 0, On: 6
Off: 0, On: 7
Off: 0, On: 8
Off: 0, On: 9
Off: 0, On: 10
Off: 0, On: 11
Off: 0, On: 12
Off: 0, On: 13
Off: 0, On: 14
Off: 0, On: 15
Off: 0, On: 16
Off: 0, On: 17
Off: 0, On: 18
Off: 0, On: 19
Off: 0, On: 20
Off: 0, On: 21
Off: 0, On: 22
Off: 0, On: 23
Off: 0, On: 24
Off: 0, On: 25
Off: 0, On: 26
Off: 0, On: 27
Off: 0, On: 28
Off: 0, On: 29
Off: 0, On: 30
Off: 0, On: 31
Off: 0, On: 32
Off: 0, On: 33
Off: 0, On: 34
Off: 0, On: 35
Off: 0, On: 36
Off: 0, On: 37
Off: 0, On: 38
Off: 0, On: 39
Off: 0, On: 40
Off: 0, On: 41
Off: 0, On: 42
Off: 0, On: 43
Off: 0, On: 44
Off: 0, On: 45
Off: 0, On: 46
Off: 0, On: 47
Off: 0, On: 48
Off: 0, On: 49
Off: 0, On: 50
Off: 0, On: 51
Off: 0, On: 52
Off: 0, On: 53
Off: 0, On: 54
Off: 0, On: 55
Off: 0, On: 56
Off: 0, On: 57
Off: 0, On: 58
Off: 0, On: 59
Off: 0, On: 60
Off: 0, On: 61
Off: 0, On: 62
Off: 0, On: 63
Off: 0, On: 64
Off: 0, On: 65
Off: 0, On: 66
Off: 0, On: 67
Off: 0, On: 68
Off: 0, On: 69
Off: 0, On: 70
Off: 0, On: 71
Off: 0, On: 72
Off: 1, On: 72
Off: 2, On: 72
Off: 3, On: 72
Off: 4, On: 72
Off: 5, On: 72
Off: 6, On: 72
Off: 7, On: 72
Off: 8, On: 72
Off: 9, On: 72
Off: 10, On: 72
Off: 11, On: 72
Off: 12, On: 72
Off: 13, On: 72
Off: 14, On: 72
Off: 15, On: 72
Off: 16, On: 72
Off: 17, On: 72
Off: 18, On: 72
Off: 18, On: 73
Off: 18, On: 74
Off: 18, On: 75
Off: 18, On: 76
Off: 18, On: 77
Off: 18, On: 78
Off: 18, On: 79
Off: 18, On: 80
Off: 18, On: 81
Off: 18, On: 82
Off: 18, On: 83
Off: 18, On: 84
Off: 18, On: 85
Off: 18, On: 86
Off: 18, On: 87
Off: 18, On: 88
Off: 18, On: 89
Off: 18, On: 90
Off: 18, On: 91
Off: 18, On: 92
Off: 18, On: 93
Off: 18, On: 94
^CTraceback (most recent call last):
  File "./test_gpio_read_state.py", line 29, in <module>
    time.sleep(0.1)
KeyboardInterrupt

kizniche, 2018-05-20:
Have you attempted to use the default settings when configuring the input? Both your screen shots have non-default settings.

reymart081678, 2018-05-21:
I am using the shield below DFR0311 

[image]

I use the RPI GPIO pins G4,G17,G18 and G21 for rocker switch, not sure it the mapping is correct that is causing invalid edge. 
I tried both GPIO state and EDGE state for pin 4 but edge same error message

reymart081678, 2018-05-21:
Seems the problem is resolve by setting the overclock from turbo mode in raspi-config to none.
[image]

kizniche, 2018-05-21:
Good to hear. What version Pi is this with?

reymart081678, 2018-05-21:
Pi Model B+ V1.2

closed 2018-05-23 by kizniche