kizniche/Mycodo #482: Sensors give no output

rowkois, 2018-05-21:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.1
 
#### Problem Description
Please list:

-Since the update from 6.7.2 to 6.0.1 needed an upgrade, i created a new image.
 In the new image which then was updated from 6.0.1 to 6.1.1 my DHT22 and my ADS1X15
is not able to be read anymore.
-A roll back to the image with 6.7.2 is working.
-Settings and wiring stayed the same

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 515, in read_adc
    measurements = self.adc.next()
  File "/var/mycodo-root/mycodo/devices/ads1x15.py", line 48, in next
    return dict(voltage=float('{0:.4f}'.format(self._voltage)))
TypeError: unsupported format string passed to NoneType.__format__
2018-05-21 21:01:39,461 - mycodo.input_e5d4506e - ERROR - Error while attempting to read adc: unsupported format string passed to NoneType.__format__
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 515, in read_adc
    measurements = self.adc.next()
  File "/var/mycodo-root/mycodo/devices/ads1x15.py", line 48, in next
    return dict(voltage=float('{0:.4f}'.format(self._voltage)))
TypeError: unsupported format string passed to NoneType.__format__
2018-05-21 21:01:42,030 - mycodo.inputs.dht22_5 - ERROR - Could not connect to pigpiod.Ensure it is running and try again.
2018-05-21 21:01:47,498 - mycodo.input_e5d4506e - ERROR - Error while attempting to read adc: unsupported format string passed to NoneType.__format__
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/controller_input.py", line 515, in read_adc
    measurements = self.adc.next()
  File "/var/mycodo-root/mycodo/devices/ads1x15.py", line 48, in next
    return dict(voltage=float('{0:.4f}'.format(self._voltage)))
TypeError: unsupported format string passed to NoneType.__format__
2018-05-21 21:01:55,448 - mycodo.input_e5d4506e - ERROR - Error while attempting to read adc: unsupported format string passed to NoneType.__format__

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-05-21:
There is no such version 6.7.2

kizniche, 2018-05-21:
I was able to find an issue with the ADS1x15 module, so I pushed a commit that should fix it, which will be available in the next release.

As for your DHT22, this error indicates you don't have pigpiod running:

```
2018-05-21 21:01:42,030 - mycodo.inputs.dht22_5 - ERROR - Could not connect to pigpiod.Ensure it is running and try again.
```

rowkois, 2018-05-21:
you are right, i had a typo, my last stable version was 5.7.2, any idea about the pigpiod, since this
is a fresh install and I didn't had this issue before.
btw, thanks for the fast response.

kizniche, 2018-05-21:
Not sure. Try disabling/uninstalling and reinstalling from the Config -> Configuration -> Raspberry Pi page

rowkois, 2018-05-23:
reinstallation via Raspberry Pi page is not possible, no reinstallation possible.
Instead I tried a new installation with the new update, sensors are working fine again.
Thanks for your work.

closed 2018-05-23 by rowkois