kizniche/Mycodo #483: MiFlora Sensor doesn't get reed by USB Bluetooth interface

rowkois, 2018-05-21:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.1
 
#### Problem Description
-Please list: MiFlora Sensor is not geting read USB bluetooth adapter.
-Pi3B internal BLE reads the device, Pi2B with usb bluetooth adapter does not show measuring results.
-bluetooth interface is logilink bt0015
-MiFlora Sensor is found via comandline bluetooth tools, usb interface is capable of BLE functions

  - what were you trying to do

hciconfig hci0 features
hci0:   Type: Primary  Bus: USB
        BD Address: M:A:C  ACL MTU: 310:10  SCO MTU: 64:8
        Features page 0: 0xff 0xff 0x8f 0xfe 0xdb 0xff 0x5b 0x87
                <3-slot packets> <5-slot packets> <encryption> <slot offset>
                <timing accuracy> <role switch> <hold mode> <sniff mode>
                <park state> <RSSI> <channel quality> <SCO link> <HV2 packets>
                <HV3 packets> <u-law log> <A-law log> <CVSD> <paging scheme>
                <power control> <transparent SCO> <broadcast encrypt>
                <EDR ACL 2 Mbps> <EDR ACL 3 Mbps> <enhanced iscan>
                <interlaced iscan> <interlaced pscan> <inquiry with RSSI>
                <extended SCO> <EV4 packets> <EV5 packets> <AFH cap. slave>
                <AFH class. slave> <LE support> <3-slot EDR ACL>
                <5-slot EDR ACL> <sniff subrating> <pause encryption>
                <AFH cap. master> <AFH class. master> <EDR eSCO 2 Mbps>
                <EDR eSCO 3 Mbps> <3-slot EDR eSCO> <extended inquiry>
                <LE and BR/EDR> <simple pairing> <encapsulated PDU>
                <non-flush flag> <LSTO> <inquiry TX power> <EPC>
                <extended features>
        Features page 1: 0x03 0x00 0x00 0x00 0x00 0x00 0x00 0x00

sudo hcitool leinfo MiFlora:Mac:Adress
Requesting information ...
        Handle: 71 (0x0047)
        LMP Version: 4.0 (0x6) LMP Subversion: 0x706
        Manufacturer: RivieraWaves S.A.S (96)
        Features: 0x01 0x00 0x00 0x00 0x00 0x00 0x00 0x00

 sudo hcitool lescan
LE Scan ...
M:A:C (Flower care)
M:A:C (Flower care)

  - specific setup details that are involved
 
### Errors
  - List any errors you encountered.
  - Copy and pasting crash logs, or link to any specific
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

kizniche, 2018-05-21:
I added the ability to set the bluetooth adapter. It defaults to 'hci0' and you'll need to determine what the new USB BT adapter is.

rowkois, 2018-05-23:
After Today's reinstallation this is now also working, thanks for the new option

closed 2018-05-23 by rowkois