kizniche/Mycodo #484: PID Max on time and min off time not working

frodus17, 2018-05-22:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.1
 
#### Problem Description
Please list: PID Max on and min off time no longer work. Compressor is staying on, regardless.

New install (because of sensors not reading right after upgrade). GPIO17 wired to SSR that controls compressor and fan. DS18B20 sensor. Created new PID with default values. Kp gain of 1 didn't work well, but 5 or higher started to work. Unfortunately I don't want to have the compressor on for more than 10 minutes or restart unless 2 minutes has elapsed. 
 
### Errors
No errors, but graphs show that the compressor runs all the time.

kizniche, 2018-05-22:
Please provide all the pertinent settings in order to replicate the issue and I'll run some tests soon.

kizniche, 2018-05-22:
Your PID period is 30 seconds. With a minimum on time of 30 seconds, the output will need to be at least 30 seconds to actually tuen on. This will necessitate the output being on 100% of the time or 0% of the time.

frodus17, 2018-05-22:
I can get it to turn on. My issue is it will not turn off after 10 min. It stays running. Also if I do get it to turn off somehow it won't wait 120 seconds to turn on again. It's unstable.

kizniche, 2018-05-23:
If the period is set to 30 seconds, the maximum time the output can be turned on is 30 seconds. If you set the minimum time it can turn on to 30 seconds, this makes 30 seconds the only amount of time that the relay can turn on for. Every 30 seconds, the output can either turn on for 30 seconds (if happening continually, this appears as it's always on, however this is just many 30 second intervals back-to-back), or not turn on if the duration is to be less than 30 seconds.

For your configuration to operate properly, you will need to increase your period. I would start conservatively, to perhaps 90 seconds and see how the system operates.

For an incubator of mine, I chose not to use a PID, as it would be overkill and difficult to use. Instead, I created a Measurement Conditional, that was tied to a temperature sensor, and would turn the compressor on for x seconds every time the temperature reached y. Depending on the target temperature, I needed a different duration to achieve my desired temperature drop (2 degrees C). Since my outside air temperature doesn't change, this works well. If my outside air did change, then the duration would be variable and PID may be the way to go.

Here are two conditionals I alternate between that keeps the temperature at wither 17-19C or 12-14C:

[image]

frodus17, 2018-05-24:
Yeah, I need something that can actively control. It's in my garage that swings wildly. It could be 80 some days, and others like 60 in the summer.

Thanks for the tip. I tried some things, it's closer now.... still having a hard time getting the temp to go down. Lots of space inside, so it'l take a while.

Gossen1, 2018-06-04:
This is interesting. I use the max on duration for a airco unit. I stripped the thermostat and use the pi with mycodo and a temp sensor as a replacement. For preventing a freeze over inside the airco unit I use the max on and min off. But my pid period is also 30 sec. Max on is 3000 seconds. See attachment. Should I change the pid period to 3000 seconds? 
[image]

kizniche, 2018-06-05:
The current PID controller does not work exceptionally well for controlling air conditioners/heaters. I've been wanting to build a dedicated AC/Heating Function that fixes some of the issues that the general PID falls short when attempting to control an AC.

The thing that should be understood about the PID is that the P, I, D values will be calculated every Period (30 seconds in this case), and a new On duration will be sent to the Output Controller (if it's instructed to be on for a duration). If your period is 30 seconds, then the maximum on duration sent will be 30 seconds. This doesn't mean the device cannot be on for a sustained amount of time longer than 30 seconds. For instance, if the P, I, D values are so large that it continually turns on the device for 30 seconds every period, and the durations overlap, it will appear as the device is continually staying on. It operates under the assumption that if a new duration is sent to turn the output on when it's already operating under a duration, it will merely keep it on and update the duration. If you want your Max On Duration to have a real effect, you will need to set your period larger than your Max On Duration.

Gossen1, 2018-06-05:
Thank you for this very clear explanation.

frodus17, 2018-06-05:
I got mine to work better. 

I set PID period to 1860s
I then set max on time to 1800s
Then I set the min off time to 60s

This way, the max it can be on in a period is 1800s, which is less than the PID period, so I always get at least 60s of off time. 

The bummer: My chamber needs a recharge. It won't get down past 60 easily.

kizniche, 2018-06-05:
Do you mean it won't get down easily if you open a door, or just in general?

If it's the issue of breaching the system, then you could set up a switch that detects the door opening, that will disable/pause the PID, start the timer, then turn the device on for an extended period. After the extended period on to recover the system, the timer can deactivate the conditional activated by the switch and activate/resume the PID.

frodus17, 2018-06-05:
It won't get down easily because it needs Refrigerant and I haven't gotten around to it. 

I keep it closed, so I don't think that's it. I don't care about fluctuations form the door opening, it'll get cooled back down quickly. That's a cool idea though. I do have a door switch I could install.

closed 2018-06-18 by frodus17