kizniche/Mycodo #488: Conditional timer:Daily Point / Daily Span won't save times

eyeconoklast, 2018-06-03:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.12
 
#### Problem Description
Please list: When creating a Conditional Timer: Daily point or Conditional Timer:Daily span mycodo will not save the created timer.  Continuously gives error when setting time, every attempt to save produces error.  

Error: Mod Conditional: Start Time (HH:MM) must be a valid HH:MM time format

Time has been properly formatted as requested.  I have tried typing in the time manually and selecting the time from the pop up clock overlay.  Neither works.  It even renders the error when I create a brand new conditional and try to save it with the default times.  

  - what were you trying to do
  - specific setup details that are involved
 
### Errors
  
Error: Mod Conditional: Start Time (HH:MM) must be a valid HH:MM time format

cohe4ko, 2018-06-05:
I got the same issue:
Error: Mod Conditional: Start Time (HH:MM) must be a valid HH:MM time format

Gossen1, 2018-06-05:
Same here. Fresh install 6.1.2
Happens with Timer (Daily point) and Timer (Daily Span)

kizniche, 2018-06-05:
I have fixed this twice, once by reverting a previous fix. Now it seems the original fix fixes it. Something strange is going on. I'm going to completely redesign this part of the code.

kizniche, 2018-06-05:
I found the issue. Pushing a fix that will be incorporated into the next release.

closed 2018-06-20 by kizniche