kizniche/Mycodo #489: All measurements returned failed CRC

Gossen1, 2018-06-05:
## Mycodo Issue Report:
  - 6.1.2
 
#### Problem Description
"All measurements returned failed CRC"  Shows up in Daemon log for all 3 sensors. Wanna know why. No problem per se. Sensor readings are ok for now. (3 x AM2315 sensors)

  - I use fairly long cables (12 meter is the longest) I started with unshielded UTP cat5 with pullup resistors 10k. These gave me errors after some time (NO DATA) After some googling I changed the setup to s-ftp (shielded and foil) CAT6 to counter EMI and 4.7k pullup to decrease the raise time to counter bus capacitance.
 
### Errors
2018-06-05 14:41:18,879 - mycodo.inputs.am2315_2 - ERROR - All measurements returned failed CRC
2018-06-05 14:46:14,199 - mycodo.inputs.am2315_1 - ERROR - All measurements returned failed CRC
2018-06-05 14:48:18,784 - mycodo.inputs.am2315_2 - ERROR - All measurements returned failed CRC
2018-06-05 15:04:59,157 - mycodo.inputs.am2315_1 - ERROR - All measurements returned failed CRC
2018-06-05 15:05:59,183 - mycodo.inputs.am2315_1 - ERROR - All measurements returned failed CRC
2018-06-05 15:21:25,631 - mycodo.inputs.am2315_3 - ERROR - All measurements returned failed CRC
2018-06-05 15:27:48,884 - mycodo.inputs.am2315_2 - ERROR - All measurements returned failed CRC
2018-06-05 15:30:48,821 - mycodo.inputs.am2315_2 - ERROR - All measurements returned failed CRC

kizniche, 2018-06-05:
Check out #315 for a solution to increasing the reliability of the AM2315 by utilizing software I2C instead of the hardware/BCM2835 I2C.

Gossen1, 2018-06-05:
Thanks for the quick reply. I use a I2C multiplexer to connect the 3 sensors (tca9548a) and this multiplexer is showing up on Bus 1 at 0x5c. I'm a bit confused what to do. Is it always good practise to use the i2c_gpio overlay? Or only with certain hardware?

kizniche, 2018-06-05:
I'm not exactly sure how to set up a multiplexer using the SW I2C. I've only used the multiplexer overlay. I would first test with creating the SW I2C and connecting one AM2315 to see if it fixes the issue before delving into attempting to get the multiplexer on the SW I2C.

Gossen1, 2018-06-05:
I will test that in the future on my test setup. For now, the "All measurements returned failed CRC" message sometimes creeps up (5-10 minutes between them), but the readings are ok. Thanks for the support. Will keep you posted.

Theoi-Meteoroi, 2018-06-08:
This is probably an indication that the I2C stream is dropping bits.  The sensor turns itself off 3 seconds after a reading. The first thing it needs to see is a wakeup command sequence.   It would seem circuit conditions are not stable.   12 meter long cable?  At that length, it probably functions as an antenna and well, I'd say is wildly optimistic for robust operation.   For lengths like that you really should use RS-232 or something like it to drive long cables.   I2C wasn't really designed for running over cables.  I could see where you could damage devices with standing wave reflections.  Back when RS-232 was ubiquitous - even those devices would fail from such impulses when cables were detacted and re-attached on cables over 50'.  Terminal server designs all had MOV transient suppressors to keep the equiment from being damaged.

closed 2018-10-13 by kizniche