kizniche/Mycodo #490: Feature: Dedicated AC/Heating Function

kizniche, 2018-06-05:
This thread is for the development of a dedicated AC/Heating Function that incorporates the benefits of PID control with the features required for operating an efficient AC/Heating system. Additionally, features that would enable low temperatures with a standard wall AC system can be integrated ([coolbot](https://www.storeitcold.com/) clone).

Ref: #484 #346

drgrumpy, 2019-02-16:
Maybe I am being dense, but how is this different from a PID with just the P set and then having saying long sampling intervals, long min on/off values? Or a couple of conditionals for on/off ? Or am I missing something completely?

Routout, 2019-02-17:
I've been thinking this also, but I'm not sure if it's stable process or do it change over the time.
Now I'm using different software, but as soon when this issue is solved I'm happily jumping in.

Routout, 2019-04-18:
@kizniche Do you still have this on your roadmap?

kizniche, 2019-04-18:
What features would everyone like to see in something like this?

Routout, 2019-04-18:
Well for myself I can answer that after this I could use Mycodo with fridge compressor to make air dried meat. Normal setup is an old fridge, temp and humidity controller. 
I could imagine that Mycodo have very good possibilities with brewing guys also after this.

kizniche, 2019-04-19:
I have some thoughts on how an AC/Heat Function could be constructed.

1. Incorporate two PID controllers into one AC/Heat Function; one for the AC side (lowering temperature) and the other for the heating side (raising temperature). Each PID will have its own PID settings, as each will have different dynamics in the same system (depending on the particular AC or heating hardware).

2. Allow the use of a Band, enabling a setpoint range (e.g. 75 °C ± 1 °C). This would allow both PID controllers to operate for an intermediary period before one turns off and the other turns on. If the setpoint had no band, as soon as the temperature rose over the setpoint, the AC would turn on and the heater would turn off. If a band existed, the heater could stabilize within the band (for example 75 °C ± 1 °C) as it rose past 74 °C, the integrator would start curbing the heater output to keep it at the lower bound of the band (74 °C). If the ambient temperature began to rise to cause the measured temperature to rise above 76 °C, then the AC PID would begin to operate. And, complimentary to the heater, as the temperature decreased to 76 °C, the AC integrator would start to curb the AC output (either by modulating the compressor or a heat bypass valve).

3. Add an AC output option for controlling the compressor and an option for controlling the heat bypass valve. Since AC systems are variable, there may exist a heat bypass valve used to fine-tune temperature control. There should exist the option to select an output that controls this electronic valve. This output, just like the current PID outputs, should have a Min On, Max On, and Min Off to prevent switching that could damage the system.

4. Add an option to select an Input for a temperature probe attached to the AC heat exchanger to detect if it's become frozen (<= 0 °C) and to turn off the compressor or heat bypass and allow thawing. This prevents damage to the AC system and makes the system more efficient (as a frozen heat exchanger that condenses water vapor will prevent air flow, which will decrease the ability to cool the room and stress the fan due to air flow restriction, among other issues). This is also the main methodology of how the CoolBot operates.

Let me know what you think and if these sound like good or bad ideas.

Routout, 2019-04-19:
Sounds pretty complicated vs simple turning output on or off depending temperature.
Maybe different issue than #497

kizniche, 2019-04-19:
This is all theoretical at this point, and PID can be one of several options.

What options do you think would be appropriate for a simple thermostat that uses both AC and heat?

drgrumpy, 2019-04-19:
As I see it, 1 and 2 can already be done, without any changes (this is what I do heating is set to say 16 and cooling to 21). And can't 3 and 4 be done with conditionals ? Or another PID ?  So to me it is just adding an extra layer (of complexity) when it may be better to focus on improving/ensuring the reliance/reliablity of the existing system/code ?

kizniche, 2019-04-19:
> 1 and 2 can already be done, without any changes

Yes, but not within one function.

> can't 3 and 4 be done with conditionals ?

Theoretically, since conditionals are Python code, you can do anything with them, including shoving a whole PID or multiple PIDs within one. Furthermore, there's currently no support for operating both the compressor ***and*** heat bypass valve for a PID. 

> So to me it is just adding an extra layer (of complexity)

I think you're missing the bigger picture, that what this will provide is an all-in-one function that will be plug-and-play for an AC/Heater. No need to configure several controllers across mycodo and no need to work through trial and error getting the often-confusing and user-unfriendly conditionals to work.

> it may be better to focus on improving/ensuring the reliance/reliablity of the existing system/code

As of right now, I don't see any major issues that are open. There's also nothing to say issues and features cannot be concurrently worked on. If you see a reliability issue, please open an issue.

Keep in mind what I suggested is for the most basic implementation for this function to operate. The interesting part comes once we add accessories that improve functionality. With what I suggested already, it would be very difficult to implement using what is currently available.

Potential additional options that build upon the base options:

1. Incubator mode(s): Continuous cooling with heater used to maintain specific temperature, for very accurate regulation.

2. Setpoint tracking, which is a common feature of PIDs, but would be difficult or impossible to implement properly if one were using two PIDs separately.

3. Fan speed control, for changing air turnover rate or efficiency of heat exhange from cooler/heater.

Routout, 2019-04-20:
> This is all theoretical at this point, and PID can be one of several options.
> 
> What options do you think would be appropriate for a simple thermostat that uses both AC and heat?

For my use it's enough that fridge turns on when upper temperature limit is reached, and off on lower limit. Same for humidifier. No need for PID.

Magnum-Pl, 2019-04-29:
I’ve been thinking about using Mycodo control a garage furnace with a bad controller board. Having the ability to control heating and cooling in one function would be nice. 

I’m currently using Mycodo to control a small AC unit and the option to have a temp sensor to detect if the heat exchanger has become frozen would be great. I’d like to see that as an option aside from the heating/cooling function discussed here. Possibly a stand-alone AC function, or would I be able to use the heating/cooling function without using the heating PID?

Another option that would be nice is temp sensor to turn the compressor off if it gets too hot. I’m using Mycodo to get a small chest freezer colder than possible with the stock thermostat. I had an issue where something fell and covered the grill over the compressor so it got super hot and ended up turning itself off from the safety temp sensor on the compressor itself.  Even though most compressors have a temp sensor as a safety feature, mine got much too hot for my comfort level before turning itself off. Even then, it turned itself back on again after cooling down, only to overheat again. Not sure how many of these cycles happened before I fixed it.

Is there any way we could use a knob or buttons to raise/ lower temps without having to use the web interface?

kizniche, 2019-05-01:
> Is there any way we could use a knob or buttons to raise/ lower temps without having to use the web interface?

This is fairly straightforward. You would wire simple push buttons and set up Edge detection in Mycodo for the pins connected to the buttons. Then add actions for these Edge detections to change the setpoint.

kizniche, 2019-05-01:
Some notes on the cool-bot clone feature:

1. Small heating element to be placed in contact with the AC thermometer that has been removed from the AC heat exchanger. Heats the AC thermometer to simulate it being hotter than it really is so the AC will turn on. Heater turns off and allows thermometer to return to ambient temperature when the AC should turn off.
2. Thermometer placed on the heat exchanger fins to detect if freezing has occurred. If frozen, turn heater off to induce AC compressor to stop and allow thawing.
3. Thermometer placed in the ambient space to detect the temperature of the room that the controller will use to regulate temperature with.

Magnum-Pl, 2019-05-10:
> This is fairly straightforward. You would wire simple push buttons and set up Edge detection in Mycodo for the pins connected to the buttons. Then add actions for these Edge detections to change the setpoint.

I tried this but couldn’t get the results I was looking for. In “Data”, I added a Rising/Falling Edge input. In “Function”, I added a Trigger:Edge with an action to change the PID Setpoint to a certain value. Can you add an action that will allow me to raise/lower the setpoint by one unit? Is there another way to do this?

kizniche, 2019-05-10:
Thinking about it, it's not **that** straightforward. Your first part about setting up the trigger is good. Then you would need an Action that executes a script. The command would be:

```/home/pi/Mycodo/env/bin/python /home/pi/script_raise_one.py```

For instance, the script to raise the setpoint one unit would look like this:

```python
import sys
import os
sys.path.append('/home/pi/Mycodo')
from mycodo.mycodo_client import DaemonControl
control = DaemonControl()
current_setpoint = control.pid_get('0000-0000-0000-0000', 'setpoint')
new_setpoint = current_setpoint + 1
control.pid_set('0000-0000-0000-0000', 'setpoint', new_setpoint)
```

Replace '0000-0000-0000-0000' with your PID controller's ID. You can create another script to lower by replacing "+1" with "- 1". I haven't tested the script, but theoretically it should work. Start a new Issue (I want to keep this one on topic) if it doesn't and we can sort it out.

kizniche, 2019-11-13:
I know this is an old topic, but since the addition of the custom controller functionality, development of controllers got a lot easier, and I wanted to take a shot at making a CoolBot clone controller as the first real custom controller. I just whipped together a controller that mimics the CoolBot (as far as I understand what it does).

Briefly, there is an air conditioner (AC), typically a window unit, that has the temperature sensor that monitors the condenser for freezing removed from the condenser and has a small heater attached to it (Output: Heater). A temperature sensor is attached to the condenser to monitor condenser freezing (Input: Condenser Temperature). An output controls power to the AC (Output: AC). A temperature sensor is hung in the room, away from the AC airflow, to monitor the room temperature (Input: Room Temperature). The AC is turned to as low as it can go, typically around 60 F. The heater attached to the AC's temperature sensor is used to "trick" the AC into thinking it's warmer than it actually is, and run the AC as the room temperature drops below the temperature set on the AC. While the room temperature is greater than the set temperature (e.g. 40 F), and the condenser temperature is greater than freezing, the small heater is turned on and the AC runs. If the room temperature is equal or less than the set temperature or the condenser is below freezing, the small heater is turned off and the AC is permitted to stop cooling and defrost its condenser.

I just added an example custom controller than can be imported into Mycodo, [custom_controller_coolbot_clone.py](https://github.com/kizniche/Mycodo/blob/4ad7c9083127c4e63ebd9227f15aaaae97d2b713/mycodo/controllers/custom_controllers/examples/custom_controller_coolbot_clone.py), that does what I just described. It's a very basic implementation, and there are a few more options (such as period and max age of inputs), but it should work very similarly to the CoolBot.

If you have any suggestions for this controller's improvement, or if this can be used to jumpstart the development of any of the other controllers we discussed, there's now a whole custom controller system in Mycodo that permits rapid and [relatively] easy development of new controllers. At the very least, it makes the creation of single-file controller modules possible, so you don't need to edit anything beyond the custom controller file in order to build a controller. There's also an other example controller in the same directory demoing all the potential options that can be used to make options that display on the Web UI and can be used to set values, text, dropdown selection boxes, other controller selection boxes, checkboxes, etc.

eap904, 2020-02-12:
Kyle, I don't know if you're still tinkering with this. 

A while back there was an arduino-based window AC controller called coldsnap, and the author sent me one he had made. I discovered that I had to set the fins temp to somewhere in the range of -5 to -8 C in order to get my fridge to go below 40F. With mushrooms, the colder the better. 

This depends on the size of the walk in relative to the capacity of the unit, in general YMMV, but it would be a good variable to have easily changed. 

Having this functionality built-in to mycodo would be fantastic. I suppose mycodo could run the heater just as a resistor across a GPIO pin.  I am currently using said coldsnap, and it works great, I'll try and find out the size of the resistor, which is wrapped in some heat shrink.

kizniche, 2020-02-13:
The controller file I linked to in my last post is able to be imported through the Configure -> Controller page. Check out the file and notice the custom options and how they will present to the user when the controller is added from the Setup -> Function page. Custom Inputs and Custom Controllers was developed as an easy way for users to develop new inputs and controllers, with all required info for the input/controller contained within a single file. The manual has more detail.

If you've checked out the linked custom controller, let me know if you have any suggestions for improving it or what should be changed.

berserktron3k, 2020-05-04:
Hey, sorry to be ask a tangential question regarding this, but I've been trying to figure out how to make/buy a compact AC that I can route in to a fruiting chamber to more effectively control the temperature. Does any one have any recommendations/guides to follow?

eap904, 2020-05-05:
> Hey, sorry to be ask a tangential question regarding this, but I've been trying to figure out how to make/buy a compact AC that I can route in to a fruiting chamber to more effectively control the temperature. Does any one have any recommendations/guides to follow?

 There are many ways to do this, if you want to chat at length, send me a message.. wait is there a messaging feature in Github?

In short, people make some silly looking ducts out of corrugated plastic (political yard signs) and foam-board and tape, that work just fine for a window unit. If you want something a bit less janky, you can build something like a 4x4 cube from foam board, install the window unit in there, and duct air from that pre-conditioning chamber to the fruiting chamber, which also gives you the ability to add humidity before the air hits the room. You'd want to control the AC from the air in the grow room. 

Another option is to modify a window AC into a water chiller - lots of beer brewers do this - and run that water through a radiator into the grow room..

Kyle, have you made an official discord channel? Here's a link to a server I made if you want to chat about the AC 

https://discord.gg/NFZRX4S

closed 2020-07-23 by kizniche