kizniche/Mycodo #491: Custom colors Graph dashboard doesn't work

Gossen1, 2018-06-07:
## Mycodo Issue Report:
  - Specific Mycodo Version: 6.1.2
  - Chromium Version 60.0.3112.89 (Developer Build) Built on Ubuntu 14.04, running on Raspbian 9.4 (32-bit)
 
#### Problem Description
Please list: When enabling custom colors in Graph dash the color pickers won't show. Can't select any colors.

 
### Errors
  - No errors

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. Make graph -> save
 2. collapse graph -> tick custom colors -> save
 3. collapse graph -> no colorpicker or input fields
 
Not a biggie but I thought I bring it to your attention.

kizniche, 2018-06-07:
I haven't been able to replicate this issue. Could you upload a screenshot? Also, can you test with just one input on the graph?

[image]

Gossen1, 2018-06-07:
Tested it with 1 input and no change. 
[image]

kizniche, 2018-06-07:
Is your browser blocking Javascript?

kizniche, 2018-06-07:
https://stackoverflow.com/questions/13020530/chrome-blocking-javascript-on-localhost

Gossen1, 2018-06-08:
No. Is set to Allow. Double checked it on: https://www.adspeed.com/Knowledges/262/Problems-Account/howto-test-browser-supports-JavaScript.html

kizniche, 2018-06-10:
I installed chromium on my desktop but still couldn't reproduce your issue. All of my Pis are headless, so I don't have immediate access to test on Raspbian. I see you have addons installed. Can you try running a tab in incognito mode and see if you can see the color selectors?

kizniche, 2018-06-24:
Can you verify this is your browser and not Mycodo, by trying a different browser?

mrplow, 2018-06-25:
I'm seeing this issue too. Happens in Chrome and Firefox.

I noticed the colour selection disapeared after I entered my 2nd or 3rd graph. My logs show nothing odd.

[image]

kizniche, 2018-06-28:
I was able to replicate this issue. I'm investigating.

kizniche, 2018-06-28:
I just pushed a fix that will be incorporated in the next release.

closed 2018-06-28 by kizniche

mrplow, 2018-06-29:
This didn't fix the issue for me, I ran the upgrade and rebooted my pi. The html source looks the same as before.

Will I need to recreate the InfluxDB @kizniche?

Thanks!

kizniche, 2018-06-29:
What are all the dashboard elements you have on your dashboard?

Have you tested with only a graph on your dashboard?

There's probably one element causing the issue with the graphs, you just need to test to find what it is. It's not an issue with your influxdb database.

mrplow, 2018-07-01:
I took screenshots of all my dashboard elements and deleted them one at a time.

When I deleted my 2nd to last element, which was the PID Control, my now last element, which was the kPa history now showed the color chooser.

I readded all the elements back and the color chooser remained. Once I was done I realized none of the other elements had the color chooser available. I wasn't sure at which point, if ever, their option was even available.

I then started disabling each graph item from within each element one at a time.

Removing the PID Control entirely fixed the color chooser for all remaining elements, it was all working.

I then readded the PID Control and everything was still working. I then readded each element item one at a time and by the end it was all still working.

So, to make a long story short its all working, with the same setup as before, but I don't know why.
Sorry I couldn't nail down on the exact conditions that messed it up.

### Dashboard:
**1. Graph: kPa history**
1 math output: equation on BME280 Pa output of: x*.001*(1-(.0065315)/(12+.0065315+273.15))**-5.257

**2. PID Control: Room A Humidity down**

**3. Graph: Room B**
2 Inputs: BME280 Temp, and Humidity

**4. Graph: Pi**
4 Inputs: Pi CPU Load1/2/3, CPU Temp

**5. Graph: Room A**
2 Inputs: DHT22 Temp, and Humidity

**6. Graph: Room A PID Info**
1 Input: DHT22 Humidity
1 Output: GPI PWM Duty Cycle (for 12V 4-wire CPU fan)
1 PID: Setpoint

kizniche, 2018-07-01:
Weird. It must be something about how new PID Control elements are created compared to how they used to be created (settings-wise). So it looks like the fix works but old PID elements may need to be deleted and recreated for it to work properly.

neongrau81, 2020-05-01:
Is it possible that this bug is back? 🤔  
I experience exactly the same behavior as mentioned above.
Two times the exactly same graph. One shows the dialog for the custom colors, the other (newer one) doesn't

kizniche, 2020-05-01:
Check the [CHangelog](https://github.com/kizniche/Mycodo/blob/master/CHANGELOG.md), this has yet to be released.

neongrau81, 2020-05-01:
ooops. That would probably have been the better place to look for it. 
I only checked the release notes 😇