kizniche/Mycodo #492: 5v Relay Module Not switching

HendoJ, 2018-06-23:
## Mycodo Issue Report:
  - Specific Mycodo Version:

 6.13

#### Problem Description
Please list:

Hi Kyle. 
I'm extremely new to these type of projects and my understanding is very limited at this stage and I apologize preemptively if my problem is a rookie one (I'm learning and trying to be better).
 
I've managed to install Mycodo(ver 6.1.3 on a Rpi 3B) and get everything working. I've connected a DHT22 sensor and have easily managed to setup graphs and readings ect.
-
Now I want to connect a simple 5v relay to control a fan. I have a "JQC-3FF-S-Z" 5v relay connected and setup in Mycodo as per instructions;
As soon as I setup the GPIO/OUTPUT in Mycodo the relay's state LED turns on, which indicates to me that the GPIO pin is indeed set to "Output" and initially set to "LOW" as the relay is activated with a LOW signal. The system info window in Mycodo does also confirm that the selected GPIO pin (BCM no. 4) is set to OUTPUT.  
 
PROBLEM is:  I can not for the life of me get the relay to switch it's state (Get Mycodo to write a HIGH state to the relay's GPIO pin). When i switch the relay ON through Mycodo- it "Activates" the relay but the rely doesn't physically change state (and the state LED doesn't change either) I've tested the relay with a arduino and it works just fine. 

I know that the GPIO pins only produce a 3.3V signal voltage- Could that be the problem?? I've unplugged and plugged the signal cable and it seems to activate as soon as it is connected (Which makes me think the 3.3V is actually enough to switch the relay)..

Is there additional steps that I am missing?? I realy hope I've described my (stupid) problem and that the solution is a simple one. 

Kind regards
HendoJ

kizniche, 2018-06-23:
Welcome. Don't worry about whether your questions are obvious or not. Ask anything you're unsure of.

I'll take a look at the specific relay you're using when I get home and get back to you.

HendoJ, 2018-06-23:
Thank you Kyle.

I've hooked up a single relay with another RPi to try and test it with basic gpio commands through terminal.
I hooked up the relay (VCC -- 5V; GND -- gnd; IN -- gpio 4 (BCM)). With a LED on the same pin (To see if the LED switches On and Off as I switch the Write state)

After a bit of "learning" and realizing the default raspian uses wPi numbering. I managed to set the pin state to output: "gpio mode 7 output" (wich set BCM 4 to Output pin). And successfully blinked the LED with "gpio write 7 1; gpio write 7 0". 

I then connected the relay to the same pin and tried the same commands (Which, in my mind, should work to switch the state of the relay).  The State LED on the relay goes On, indicating to me that it is receiving a LOW signal and does absolutely nothing when I attempt to switch the signal (with "gpio write 7 1).

Thus I THINK that the 3.3V on the GPIO pin is not strong enough to switch the relay....?

Am I right to conclude this....?

Thank you for coming back to me I really appreciate it.

kizniche, 2018-06-23:
Here is the datasheet for this relay: https://www.robotshop.com/media/files/pdf/datasheet-v1.0-dfr0017.pdf

The pickup voltage for your should be 3.8 volts, which is above the 3.5 volts the GPIOs output. So, you're going to have to either increase the voltage or find a relay that can operate with 3.5 volts. Does yoru arduino by chance operate at 5 volts?

HendoJ, 2018-06-23:
Awesome thank you. 

I intend to get a different relay tomorrow so that I don't have to mess around with alternative circuit elements and power supplies ect. Is there a relay module that you would suggest?

Yes the arduino supplies a 5V signal. (Read a few threads explaining the issue/difference with the RPi and Arduino and how the 3.3V COULD be the issue but I wasn't sure...until now (:.

Thank you for this it is an AMAZING project! And thank you for the help as well!!

kizniche, 2018-06-23:
You're welcome. It seems that is most likely the issue.

For smaller loads (up to 2 amps), I use these, which come in 4 and 8 relay modules: https://www.sainsmart.com/products/4-channel-5v-solid-state-relay-module

For larger loads, I usually go with Crydom SSRs, such as this: http://www.crydom.com/en/products/panel-mount/

HendoJ, 2018-06-23:
Thanks a Mill Kyle! Really appreciate it! Lets hope I get it right tomorrow. :)

closed 2018-10-13 by kizniche