kizniche/Mycodo #494: Access Mycodo

peterabdo, 2018-06-25:
## Mycodo Issue Report:
  - Specific Mycodo Version: latest version
 
#### Problem Description
Please list:
Hi Kyle,
We at UTS have used the mycodo software last year during November and December, in fact Mr. Vahik Avakian was in touch with you in relation to mycodo.
I was trying to use the software and tried to connect through https://138.25.51.246 but i was getting a message that the website is unavailable; and that it might be switched off or moved to a new address. Would you please help resolve this issue.
Thanks,
Peter Abdo 

  - what were you trying to do access mycodo
  - specific setup details that are involved https://138.25.51.246
 
### Errors
  - List any errors you encountered.  message that the website is unavailable
  - Copy and pasting crash logs, or link to any specific 
code lines you've isolated (please use GitHub permalinks for this)

### Steps to Reproduce the issue:
How can this issue be reproduced?
 1. step 1
 2. step 2...
 3. etc 
 
### Additional Notes
Is there anything that should be added to make it easier
to address this issue?

peterabdo, 2018-07-01:
Hi Kyle.
Should https://138.25.51.246 work? we dont think that the port is the issue as we did not change any. This was working before until we upgrade to the latest version.
Your help is really appreciate to sort this issue. Thanks.
Peter

peterabdo, 2018-07-01:
> It sounds like a routing issue. Make sure the router/modem the Pi is connected to has the correct port forwarded (443 for HTTPS). Make sure Mycodo is accessible from within your network (connected the same router). If you can access Mycodo from within the network but not from outside, then it's likely an issue with port forwarding.
>
>
> UTS CRICOS Provider Code: 00099F DISCLAIMER: This email message and any accompanying attachments may contain confidential information. If you are not the intended recipient, do not read, use, disseminate, distribute or copy this message or attachments. If you have received this message in error, please notify the sender immediately and delete this message. Any views expressed in this message are those of the individual sender, except where the sender expressly, and with authority, states them to be the views of the University of Technology Sydney. Before opening any attachments, please check them for viruses and defects. Think. Green. Do. Please consider the environment before printing this email.

Hi Kyle.
Should https://138.25.51.246 work? we don’t think that the port is the issue as we did not change any. This was working before until we upgrade to the latest version.
Your help is really appreciate to sort this issue. Thanks.

Regards

Peter Abdo
PhD Candidate
Faculty of Engineering and Information Technology | University of Technology Sydney
Email: peter.abdo@uts.edu.au<mailto:peter.abdo@student.uts.edu.au>

kizniche, 2018-07-01:
Are you able to access Mycodo from your internal network?

peterabdo, 2018-07-02:
Hi Kyle,
No we are not able to access Mycodo from our internal network.
I was trying to use the software and tried to connect through https://138.25.51.246 but i was getting a message that the website is unavailable; and that it might be switched off or moved to a new address.
This happened after we upgraded to the latest version.

kizniche, 2018-07-02:
Are there any errors in the upgrade log?

/var/log/mycodo/mycodoupgrade.log

peterabdo, 2018-07-05:
Hi Kyle,

we have upgraded to version 6.1.4.
Now we are able to log in, however we have a different problem and that is that we cannot configure the BME280 sensors.  Previously we used to introduce the information of the multiplexer along with the related channel while configuring the sensors, however now we are not able to.

Please see attached photos.

Would you please help resolve this issue and advise on how to configure the sensors.
[image]
[image]
[image]
[image]
[image]
[image]

Thanks.

peterabdo, 2018-07-05:
Hi Kyle,

we have upgraded to version 6.1.4.
Now we are able to log in, however we have a different problem and that is that we cannot configure the BME280 sensors.  Previously we used to introduce the information of the multiplexer along with the related channel while configuring the sensors, however now we are not able to.

Please see attached photos.

Would you please help resolve this issue and advise on how to configure the sensors.

Thanks.

Regards

Peter Abdo
PhD Candidate
Faculty of Engineering and Information Technology | University of Technology Sydney
Email: peter.abdo@uts.edu.au<mailto:peter.abdo@student.uts.edu.au>

kizniche, 2018-07-05:
With version 6.x, the integrated multiplexer functionality has been removed from Mycodo. This doesn't mean you can't use a multiplexer, but you will have to configure a driver so the multiplexer works at the kernel-level. This is the preferred and more robust way to use the multiplexer. The instructions can be found [here](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#i2c-multiplexers). Essentially, the driver creates a new I2C bus for each channel of the multiplexer, so if you have 8 channels, you will have 8 new buses to choose from (there is an I2C bus option for every I2C input). Once the multiplexer is set up, these buses can be viewed on the System Information page (click the version number in the top-left, or select Config -> System Information). From there, you can view all the buses and the devices connected to each.

peterabdo, 2018-07-30:
Hi Kyle,
Would you kindly provide a step by step procedure to install the multiplexer driver?
I try to install the driver for : TCA9548A/PCA9548A: I2C Multiplexer link (I2C), and opened the link, But I could not find where to install the driver, or a link which says install.

I understand that after installation I have to enable the driver for the TCA9548A/PCA9548A, (visit GPIO-pca9548 to get the code and latest install instructions. If successfully set up, there will be 8 new I2C busses on the Config -> System Information page.)

Appreciate your help. Sorry I could not get back to you earlier since I was overseas.
Thanks.
Peter.

kizniche, 2018-07-30:
From the 0Multiplexer section of the manual](https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#i2c-multiplexers), we'll install the driver from the [Theoi-Meteoroi/GPIO-pca9548](https://github.com/Theoi-Meteoroi/GPIO-pca9548) repository. Here are the commands to enter into your Pi's terminal.

```bash
cd ~
wget https://github.com/Theoi-Meteoroi/GPIO-pca9548/blob/master/i2c_gpio-pca9548.dts
sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo ./i2c_gpio-pca9548.dts
```

Now, edit ```/boot/config.txt``` with the following command:

```
vi /boot/config.txt
```

Go to the end of the file and add the following two lines:

```
dtoverlay=i2c-gpio,i2c_gpio_sda=23,i2c_gpio_scl=24,i2c_gpio_delay_us=4
dtoverlay=i2c_gpio-pca9548
```

Save your edits in vi, exit vi, then reboot. You will need to have your multiplexer connected with SDA on pin 23 and SCL on pin 24. If you have it properly connected and powered, after rebooting you will have several new I2C buses available for use, one for each channel of the multiplexer. You can verify the buses are available by going to the Config -> System Information page of Mycodo and viewing the I2C buses. If there's only one, there is an issue. If you see 9 buses (1 from the original bus + 8 new buses from the multiplexer), then everything is working and you can begin connecting devices to your multiplexer channels.

Theoi-Meteoroi, 2018-07-30:
I noticed this ... I shall add to the depot page as well to improve the docs.  
I am open to feedback to make using this overlay as simple as possible.

Thanks to Kyle for stepping in to provide this helpful narrative.
Kyle, you rock. As always.

peterabdo, 2018-08-07:
Hi Kyle,
Sorry to be the pain but I still have issues with the installation.
Once I put the command 
sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo ./i2c_gpio-pca9548.dts
I get a fatal error: unable to parse input tree ; please see attached photo.

also would you please advise in relation to how to edit the file and save in vi???I used the following command in the terminal : vi /boot/config.txt
went to the end of file and added the two lines, but I could not find how to save, or I must have not accessed the file!!
[image]
[image]

Theoi-Meteoroi, 2018-08-08:
Howdy @peterabdo 

The first error is a syntax error. I can't see the end of the command line on your screen but it sure looks to me like an invocation error (wrong input).  The dtc command can't parse the tree because it can't find the file.

 It sorta looks like the file you downloaded  has a .1 at the end. (i2c_gpio-pca9548.dts.1) You need to change the filename to:  i2c_gpio-pca9548.dts

Check the command line carefully to match:

sudo dtc -I dts -O dtb -@ -o /boot/overlays/i2c_gpio-pca9548.dtbo i2c_gpio-pca9548.dts

Methinks a install script is the correct (forever) answer to this problem.

I hope this helps!

peterabdo, 2018-08-08:
Hi Dave.
Thanks heaps.
I am not sure about the file downloaded. Would it be possible please to send me a snap shot showing where to download it from , and how to make sure it has the correct name.

Appreciate your help.
Thanks.
Peter

kizniche, 2018-08-08:
The ```wget``` command I provided, above, has the direct link to the file.

kizniche, 2018-08-09:
What version Raspberry Pi are you performing this on?

kizniche, 2018-08-09:
Also, is your system up-to-date? i.e. have you performed ```sudo apt-get update && sudo apt-get upgrade```

Theoi-Meteoroi, 2018-08-23:
After reviewing the instructions while working on the install script - I found that the wget command URL is not correct.  I have fixed the readme with the correct wget command:

wget https://raw.githubusercontent.com/Theoi-Meteoroi/GPIO-pca9548/master/i2c_gpio-pca9548.dts

That gets the file content properly into a file that will compile.  Sorry about that @peterabdo and @kizniche    @peterabdo was correct at pointing out a bug.

peterabdo, 2018-08-23:
Thank you both Kyle and Theoi for your feedback.
I will try the new command soon and let you know.
@Theoi-Meteoroi ; is the install script ready to use? 
Please let me know as I am not familiar with the installation using command etc....
Thanks again Dave and Kyle.
Cheers
Peter A

Theoi-Meteoroi, 2018-08-23:
Here is a simple install script that performs all the steps in the readme with the defaults.

[mux-install.txt](https://github.com/kizniche/Mycodo/files/2312996/mux-install.txt)

download or copy the file and rename to mux-install.sh and run with:

/bin/sh  mux-install.sh  

Only run it only *once* or there will be multiple entries in /boot/config.txt.

Theoi-Meteoroi, 2018-09-02:
I have substantially re-written the README for the overlay and added the install script to the repository.  I have also added some other SMbus information and  a debug resource.  

The mux-install.sh script will perform all the actions covered in the manual installation section.

You can now just grab the install script and run it and you are done.

kizniche, 2018-09-03:
Your install script should prevent a lot of headaches.

kizniche, 2018-09-11:
Good to hear! The theme for each use can be configured under Configure -> Users.

peterabdo, 2018-09-12:
Thanks Kyle, Changed the theme as advised.
This is the second day in a row where the system shuts down by itself. This was not happening when we used it before. Have we missed something during set up configuration? I would need this to run continuously. I feel that it is shutting up sometime during the night at Sydney!

Please let me know if we can overcome this from the software or could this be a hardware issue?

Thanks Kyle.
Peter

kizniche, 2018-09-13:
Your entire Pi shuts down? That doesn't sound typical.

kizniche, 2018-10-13:
If you can provide information about the question I asked or any additional information to diagnose your issue, feel free to reopen this issue.

closed 2018-10-13 by kizniche

peterabdo, 2018-10-13:
Hi Kyle,
Thanks for following up.
I have order some sensors to connect to the system and therfore did not operate the set up for a while. These should arrive in 1 or 2 weeks.
The problem still happens periodically and I will check for it again tomorrow and let you know. Yes it seemed that the entire Pi shuts down! Could that be a setting issue?
Thanks again.
Peter

peterabdo, 2018-11-12:
Hi Kyle,
I hope you are doing fine.
Yes this issue still occurs. The pi shuts down entirely and I noticed that this happens about midnight every night. Is this a setting issue with Pi? Appreciate your help with this and advise to diagnose this problem.
Thanks.
Peter

kizniche, 2018-11-12:
I've never experienced this issue before, and I haven't heard of anyone else experiencing it. It sounds as though it's an issue with your Pi or power supply. Do you have another Pi and power supply you can test?

peterabdo, 2018-12-12:
Hi Kyle
I hope you are doing fine.
Mycodo have been working fine for me for the last week - 10 days. Unfortunately since yesterday I get the following message:
504 Gateway Timeout ; nginx/1.10.3
I attach two screen shots.
Please let me know your advise. I am not sure if this is caused from your end or from my end. 
Thank you

[image]
[image]

peterabdo, 2018-12-12:
Mycodo 6.2.3

attached screen shot showing the version and how i see mycodo when i log in. do i need to upgrade?
with upgrades will i be able to directly able to use my set up or is it necessary to configure everything all over again?

thanks, peter

[image]

peterabdo, 2018-12-12:
Hi Kyle
after the upgrade to 7.0.1 do I need to install drivers of multiplexer etc...
Thanks. Peter

kizniche, 2018-12-12:
Not if it's already installed and working.

peterabdo, 2018-12-12:
Hi Kyle,

which upgrade menu you mean? I tried to upgrade by clicking upgrade (red shown in the previous attachment) but I got the same error message as before (504 Gateway Timeout ; nginx/1.10.3).
is this the way to upgrade? or is there any other upgrade menu?

kizniche, 2018-12-12:
Probably best to install from scratch then.

peterabdo, 2018-12-12:
Sorry being the pain.
would you please provide the procedure how to install from scratch? I tried looking into your site and could not locate where etc... frankly I am not good doing this.
Thanks.
Peter

kizniche, 2018-12-12:
Enter these commands at the command line:

```bash
sudo service mycodo stop && sudo service mycodoflask stop
sudo mv ~/Mycodo ~/Mycodo-bcakup
curl -L https://raw.githubusercontent.com/kizniche/Mycodo/master/install/install | bash
```

peterabdo, 2018-12-12:
Thanks Kyle.
would you please send me instructions for how to install the multiplexer?

peterabdo, 2018-12-13:
Hi Kyle,
Thanks for your above instructions. I was able to install 7.0.1 and then I was able to install the driver for multiplexer and now my BME sensors are working. I had to install BME280 dependencies earlier.
I still need to configure my ADS1115 (Adafruit - analog to digital converter) with 4 channels. From the list of input devices there is ADS1X15 (Texas instruments) which i try to add. but i get an error that there are unmet dependencies, refer to attached photo. There is also a note to reset the install state, which I tried but i also got an error (Permission denied).
Please let me know how to add the Adafruit ADS1115. 
Thanks Kyle.

As a separate note while trying to install the driver for the multiplexer following the instructions on the website did not work completely. Then I looked into this thread from earlier this year around September and I used the following (/bin/sh mux-install.sh) which completed the installation. On the website it says a different command which did not work for me as follows: /bin/bash mux-install.sh

[image]
[image]

kizniche, 2018-12-13:
Please paste the last lines of the dependency log. Your screenshot ends before showing these lines. Deleting files from the linux commandline is performed with the ```rm``` command. You will need root permission to do this, therefore ```sudo rm``` will be necessary.

peterabdo, 2018-12-14:
HI Kyle,
using the sudo rm command you provided I was able to delete dependencies and then install the ADC which now I use to connect 4 0-5 volts output air velocity sensors. Thank you Kyle.
Once setting these sensors I tried to use the option of converting the unit but I could not, see attached.
 I want to change the Unit Volts to m/s if possible.
Thanks.
[image]

kizniche, 2018-12-14:
It will not be possible to convert to another unit until you have scaled the voltage to an initial unit. Therefore, set the resale options and select a measurement unit. You will most likely not need to convert to another unit after this.

peterabdo, 2018-12-15:
Hi Kyle, Thanks.
I rescaled my 0-5 volts to 0 -25.4 . 
For the rescaled measurement I selected Electrical Potentiel Volts. I could not find any measurement with a unit m/s in the options of Rescaled measurement. Is there a default one for me to choose?

kizniche, 2018-12-15:
If the unit doesn't exist, you can go to [Gear Icon] -> Configure -> Measurements and create a new unit. This can then be selected for the ADC channel.

peterabdo, 2018-12-15:
Thanks Dear Kyle, done , velocity m/s

peterabdo, 2018-12-15:
Hi Kyle,
since yesterday I get a message (Failed Netwrok error ) when I try to download XLS or CSV files. I can download the charts PNG Image.
[image]

peterabdo, 2018-12-17:
Hi Kyle,
It seems that using Chrome the downloads fails. But now I use firefox and all good.
Thanks.
Peter

peterabdo, 2019-01-14:
Hi Kyle,
Happy New Year!
I hope this finds you well.
Mycodo has been running smoothly since more than a month.
Today I tried to upgrade to the latest version. but then I got a message that there was an error and that I should follow one of three steps.(reboot or then 2. a command or 3. another command).

Upon reboot I got a message that something bad happened and that I should contact Mycodo.
Note I did run the step 2 and step 3 of the initial steps to fix, and after step 3 I also got the error that I should contact Mycodo.

Please let me know how to fix if possible.

Thanks. Peter

[image]

[image]

kizniche, 2019-01-14:
I'm not sure what could cause that. Your issue seems to be an issue with your particular system. I know of at least one system that successfully upgraded from 7.0.3 to 7.0.5. My only recommendation would be to rename the current Mycodo directory and install again anew.

peterabdo, 2019-01-21:
Thanks Kyle, I installed again and now all is good.
Cheers, Peter