kizniche/Mycodo #495: Add Input: Cozir GC-0020 Temp/RH/CO2 sensor

kizniche, 2018-06-25:


kizniche, 2018-06-28:
I just released v6.1.4 for testing.

emuehlstein, 2018-07-01:
I added the following to controller_input.py:
```
 elif self.device == 'COZIR_CO2':
            from mycodo.inputs.cozir_co2 import COZIRSensor
            self.measure_input = COZIRSensor(self.input_dev)
```

The device is failing to activate with the following:
```
2018-07-01 23:35:33,792 - mycodo.daemon - ERROR - Could not activate Input controller with ID d5e489d0-28dd-4d30-9865-f8a038dfcf01: 
Traceback (most recent call last):
  File "/var/mycodo-root/mycodo/mycodo_daemon.py", line 480, in controller_activate
    ready, cont_id)
  File "/var/mycodo-root/mycodo/controller_input.py", line 302, in __init__
    self.measure_input = COZIRSensor(self.input_dev)
  File "/var/mycodo-root/mycodo/inputs/cozir_co2.py", line 30, in __init__
    self.sensor = Cozir(self.device_loc)
  File "/home/mycodo/Mycodo/env/src/cozir/cozir/cozir.py", line 48, in __init__
    self.set_mode(OpModes.polling)
  File "/home/mycodo/Mycodo/env/src/cozir/cozir/cozir.py", line 113, in set_mode
    assert res == b'K %05d' % mode.value
AssertionError
```

kizniche, 2018-07-01:
Doh! I'll address this when I go through issues later tonight.

kizniche, 2018-07-02:
I fixed the activation issue, but that doesn't fix the reading issue you're having. Have you verified the sensor works, by reading data with other software on the Pi or with an arduino, or similar?

Did you enable UART in raspi-config?
You may need to follow the instructions fro setting up UART for the K30 to get the serial to work properly: https://github.com/kizniche/Mycodo/blob/master/mycodo-manual.rst#serial

kizniche, 2018-07-02:
Great! Thanks for testing. Let me know if there are any other issues or how the sensors perform after you've sufficiently tested.

closed 2018-07-02 by kizniche